luc-github/ESP3D #302: RGB led support

iLLiac4, 2019-02-16:
Would it be possible to add RGB led support with GCODE control like this plugin in octoprint on devices tlike WemosD1 mini?
https://plugins.octoprint.org/plugins/ledstripcontrol/

luc-github, 2019-02-16:
Esp3d allows pin control
Get/Set pin value
[ESP201]P<pin> V<value> [PULLUP=YES RAW=YES]
if no V<value> get P<pin> value
if V<value> 0/1 set INPUT_PULLUP value, but for GPIO16 INPUT_PULLDOWN_16
GPIO1 and GPIO3 cannot be used as they are used for serial
if PULLUP=YES set input pull up, if not set input
if RAW=YES do not set pinmode just read value

luc-github, 2019-02-16:
Does it fit your needs ?

luc-github, 2019-02-17:
I am currently implementing analog support for read and write should be ready soon

luc-github, 2019-02-17:
Please check here: https://github.com/luc-github/ESP3D/issues/291
Need a call for each color unlike octopi call so 3 ESP201 according pin color

closed 2019-02-17 by luc-github