luc-github/ESP3D #303: Error 503 :Communication locked!

Sanju-K, 2019-02-23:
Hello,

I loved your software and also appreciate the efforts you took in this project.
I uploaded the firmware (V1.0 )on ESP8266 and conected to Tevo Black Widow 3D printer.
Its working very well and all functions are working expect the actual printing, Whenever I upload file it uploads till 100% but nothing happens after, at same time in comments window it shows

>  "Error 503 :Communication locked!"

[image]

and then continue normal.....
the file is uploaded on the sd card in the printer, and also the timer for print is started but it stays ideal, not the temperature rise . Also the files in SD card is not displayed in window.

luc-github, 2019-02-23:
In 1.0 there is no good control of commands so because esp8266 cannot handle any commands when uploading every commands during upload is rejected
And because you have auto check temperature checked commands are rejected
So 503 connection locked is not an error just not really frendly way to reject command during upload

What printer fw do you use ?

luc-github, 2019-02-23:
Sorry just saw it in your comment. Reading from phone.
Please use branch 2.0 of ESP3D and web UI 2.1 they are improved version for Marlin

Sanju-K, 2019-03-02:
Sorry for the delay....... 
The Problem still persist but now only OK comes in the comments window and files are not shown on the SD Window

luc-github, 2019-03-02:
what is the output in command of M20 command ?
can you share the screenshot of info panel of ESP3D?

Sanju-K, 2019-03-02:
The Output for M20 is complete list of files in SD Card 
[image]
(Top one before uploading ...... Bottom on after uploading.....file name : butoon.gco)
and also the file which i uploaded is in list
But no output on SD CARD window
Info of ESP3D
[image]
[image]
[image]

luc-github, 2019-03-02:
this is settings window, I need the info one - the blue button on left of this window
also please give me the version of WebUI and ESP3D - they are in menu

Sanju-K, 2019-03-02:
This is given in tab 
[image]
[image]
 
version of WebUI and ESP3D
UI: 2.0c31 / FW: 1.0

I updated according to Instruction but FW is still 1.0

I tried to upload the bin file in file system and also in firmware update 
[image]

luc-github, 2019-03-02:
Well obviously no you used 1.0 code so FW is 1.0 
use : https://github.com/luc-github/ESP3D/tree/2.0
also please use WebUI 2.1 it is up to date for Marlin not listing Files with file Size

luc-github, 2019-03-02:
2.0 is different than 1.0 : https://github.com/luc-github/ESP3D/tree/2.0#installation

Sanju-K, 2019-03-02:
I am not understanding the actual procedure.
The 2.0 Instruction says to add libraries, so the libraries should added in example file as in (#include "LIB.h" ) or in ardunio library folder.
Also it mention to Upload the data content on ESP3D file system - so which file has to upload and where - ESP3D Filesystem or Upload from arduino IDE as code in ESP module 
and also where to upload the .bin file 

PS: A suggestion : Add a video for complete procedure for 2.0 and above as the instructions are not clear enough.

Sanju-K, 2019-03-02:
Sorry for the mistake my side. 
I updated the UI to 2.1 now the SD Windows shows file list. The printer is currently busy . Will update the print from WEB UI status.
Thank You For Your time and patience. 🥇🙌😊 
 I am Still not able to update the FW to 2.0

luc-github, 2019-03-02:
This process of adding library is not specific to this project it is same for any arduino project that is why it is not detailed. A little search answer all questions: https://www.arduino.cc/en/guide/libraries
Copy content of libraries directory :
[image]

which is :
[image]

to arduino libraries directory
[image]

which give you
[image]

then copy the complete ESP3D project as ESP3D is also a library to arduino library
[image]

which give :
[image]

then go to example and open basicesp3d.ino :
[image]

you can compile and flash
I cannot do more clear

Sanju-K, 2019-03-02:
Thank You luc
for Your patience 
Sorry for the trouble........
its all done it printing from web UI .

closed 2019-03-02 by Sanju-K

luc-github, 2019-03-02:
Happy printing then 😸