luc-github/ESP3D #304: Commands Dont seem to work with marlin 2.0 bugfix firmware

forum999, 2019-02-25:
Hi, 

Great device, i have purchased the fysetc preset up device only minor issue is when i type a command into the terminal on the page i get an error (Example):

g28
echo:Unknown command: "g28"

Not sure what i am doing wrong here can you help?

controls work fine and if i hit home it actually chosws the g28 command carried out in the terminal as it homes? no commands work i have tried g29 g1 z0 g28 etc... yet all seem to work in the controls at the top of the page

forum999, 2019-02-25:
also when setting up and i get to:

What access point ESP need to be connected to:

If you click scan you get the error::
Failed:406 Wrong data

luc-github, 2019-02-25:
>g28
>echo:Unknown command: "g28"

GCODE is case sensitive - Marlin FW wants G28 not g28

About second question, I do not know fysetc so I need more details : 
1- what ESP3D FW version do you use ?
2 -  What version of WebUI

forum999, 2019-02-25:
Excelent you solved the issue with commands, it needs to be in caps. 

fw 0.9.99
UI 0.9.60 LE

luc-github, 2019-02-25:
Ok Marlin needed several update from these versions
please update to ESP3D 2.0 and ESP3D-WEBUI 2.1

forum999, 2019-02-25:
Hi Luc, appologies im fairly new to this can you instruct on how to update?

forum999, 2019-02-25:
and thankyou for your quick and helpfull attitude its greatly appreciated mate

luc-github, 2019-02-25:
Compilation steps for branch 2.0 are here : https://github.com/luc-github/ESP3D/tree/2.0
For UI replace existing index.html.gz by 2.1 https://github.com/luc-github/ESP3D-WEBUI/tree/2.1 
the file is this one : https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz

forum999, 2019-02-25:
Its a bit beyond my capabilities Luc? i dont suppose you have team viewer and could do a quick update for me so i could watch and learn? im familiar with arduino ide etc with marlin but thats about myu limit

luc-github, 2019-02-25:
I do not do teamviewer sorry - the best I can give you is https://github.com/luc-github/ESP3D/wiki/Install-Instructions and https://github.com/luc-github/ESP3D/tree/2.0#installation
It use arduino ide so it is not difficult

forum999, 2019-02-25:
ok ill give it a try , the problem is identifying the board, would you have any idea which board this is?

http://www.aliexpress.com/item/ESP8266-WIFI-Extensible-Module-Remote-Control-ESP3D-For-3D-Printer-Board-Connect-AP-Mode-Client-Station/32863565514.html

luc-github, 2019-02-25:
the board is not the issue use generic one with CPU at 160MHz
[image]

but for the flash size - you need to check as it is not specified

forum999, 2019-02-25:
much appreciated, how do i check the flash size?

forum999, 2019-02-25:
Not sure if the above helps but if i click on the green folder in your gui it gives me the following?

Total: 934.88 KB  |  Used: 379.44 KB  40%

luc-github, 2019-02-25:
I may means you may have 1MB SPIFFS but not about total Flash
https://github.com/luc-github/ESP3D/wiki/Flash-Size#figuring-out-the-flash-size

luc-github, 2019-02-25:
esp-12-e has 4MB flash so you can choose same setting as I gave you in screen shot

forum999, 2019-02-25:
can i not just use the minimal flash size to be safe?

forum999, 2019-02-25:
and just to confirm i am uploading the latest version 1.0, do i still need to manually add the files you quoted earlier:

"Compilation steps for branch 2.0 are here : https://github.com/luc-github/ESP3D/tree/2.0
For UI replace existing index.html.gz by 2.1 https://github.com/luc-github/ESP3D-WEBUI/tree/2.1
the file is this one : https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz "

or are these already in the latest version?

luc-github, 2019-02-25:
Marlin changed  since 1.0 a lot so yes need to use ESP3D 2.0 and ESP3D-WEBUI 2.1

forum999, 2019-02-25:
ok ill give it a g,

just confirming i am using this version:
https://github.com/luc-github/ESP3D/archive/1.0.zip

and do i need to add the files you referenced earlier manually or will they already be in this version?

" Compilation steps for branch 2.0 are here : https://github.com/luc-github/ESP3D/tree/2.0
For UI replace existing index.html.gz by 2.1 https://github.com/luc-github/ESP3D-WEBUI/tree/2.1
the file is this one : https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz "

luc-github, 2019-02-25:
do not use https://github.com/luc-github/ESP3D/archive/1.0.zip
go directly to https://github.com/luc-github/ESP3D/tree/2.0 and download the zip

forum999, 2019-02-25:
ok so i downloaded 2.0 but the layout to the 1.0 folder is differend, when i open examples >basicespd.ino it doesnt have access to the other tabs such as config.h etc...

luc-github, 2019-02-25:
in theory you do not need to modify config.h, by default it is OK for common usage
sorry I need to boarding now - 14 hours flight + 3h train + 1h car so I won't connect for some time

forum999, 2019-02-25:
no problem thanks for your help :)

forum999, 2019-02-25:
Thanka for your help today luc, 

I have got as far as exporting as a binary,

I went to update on the edp8266 device and it completed.

When it restarted i could access the unit via wifi but it says there is no index.html or gz file, yet it is present in the sketch.

I also cant get the 2.0 version to compile at all? I dont think it works? Due to something im not doing

I would really appreciate your help mate if you could compile a bin file for me and post it here so i can upload it?

luc-github, 2019-02-28:
What compilation error do you have ?
Sorry I do not binary - if I start, everyone will ask for and I do not have bandwidth to serve everyone as there are too many boards and configurations possible.

luc-github, 2019-03-02:
@forum999 
Here the actions to be done with screen shots:
This process of adding library is not specific to this project it is same for any arduino project that is why it is not detailed. A little search answer all questions: https://www.arduino.cc/en/guide/libraries

Copy content of libraries directory :
[image]

which is :
[image]

to arduino libraries directory
[image]

which give you
[image]

then copy the complete ESP3D project as ESP3D is also a library to arduino library
[image]

which give :
[image]

then go to example and open basicesp3d.ino :
[image]

you can compile and flash
I cannot do more clear

luc-github, 2019-03-18:
I close issue if no update

closed 2019-03-18 by luc-github