luc-github/ESP3D #305: Q. Is there any way to upload a file to run without having an SD card?

neocogent, 2019-03-05:
I tried using the upload button but it seems to require and SD card board be installed. I'm using Marlin on a MKS Gen-L on my CNC build. I can enter commands and use the on screen controls. What I really need is a way to send a file to run.

I did a test using port 8888 via network. On linux used I did,

`cat myfile.gcode | nc 192.168.4.16 8888`

That sent the file but it has some problem perhaps with flow control as part way thru the machine just moves off on it's own. It may be getting corrupt command data or something.

I'm looking for the correct way to upload a file a sit seems like that must surely be a core functionality of a interface like this?

luc-github, 2019-03-05:
The SD card is on Main Board and controled by Marlin you mean your printer does not have SD card reader ?

neocogent, 2019-03-05:
My MKS Gen-L board does not have an SD card. If it's easy to wire one up then I could do that. I believe there was an SD card board for Ramps 1.4 that Gen-L is compatible with. Right now I have ESP8266 with ESP3D connected to Aux1 interface. It communicates via serial with Gen-L running Marlin but no SD card present. I had this setup on my 3D printer as well but SD card was provided by add-on TFT32 display board. Adding that, for me, would defeat the purpose of having ESP connected. I'm probably just missing something then as I think it should be cheap and easy to add an SD card to the Gen-L via one of the add-on connectors.

neocogent, 2019-03-05:
For more clarity, what I was wanting was to be able to upload file to ESp3D and it would buffer and send to Marlin (via serial as it does with screen controls). Maybe that would require too much RAM? I mean it kind of works when sending a file using netcat (nc) via port 888 but it doesn't seem to be managing flow quite right. Maybe it cannot due to TCP/IP protocol support in ESP8266. Not sure.

Any way, I was just googling and found there is some small SD card boards around for a few $ for ramps that have same pin out as Aux-3 on Gen-L board. So prob can plug one in and set config values for SDSUPPORT and get SD card on main board. Presumably then  ESP3D can upload into that card. Maybe that is best for now, though it would be kind of nice if one could use ESP3D as a cheap replacement for TFT32 or similar display controller (which is kind of what I wanted instead of using a raspberry pi and CNCjs, for example).

luc-github, 2019-03-05:
Well if no SD card the streaming is the only way but need flow control unlike the command you use,
you can use repetier host to stream your gcode or pronterface both support telnet connection

Having an SD is really safer than streaming GCODE but adding GCODE you need to reflash Marlin I guess to add the SD  support

ESP3D does not have host feature yet - being host is more than just flow control it is also managing transmission error and retry.
not using host over serial neither ethernet is really not safe

ESP3D 3.0 will such feature but it will need ESP32 to handle wifi + host as ESP8266 is too weak for handling both at once

neocogent, 2019-03-05:
Thx. Ya, that's what I figured. I'll add a $3 sd card module to my mainboard and then just use ESP3D to upload to that. No problem updating Marlin for SD support. I am right in thinking ESp3D will upload into SD card? I see the section on interface so assume that should work reliably.

luc-github, 2019-03-05:
Will work but transfert is slow like hell - but it is not ESP3D  problem - it is the Marlin upload protocol
 check FAQ : https://github.com/luc-github/ESP3D/issues/292

neocogent, 2019-03-05:
So I'm better to just put files on the sd card on my laptop, plug into mainboard and then select them via ESP3D interface to run. Not ideal but also not really a problem. This is on my CNC build so files aren't that big (compared to 3D printer).

luc-github, 2019-03-05:
if not big it may not take too ling to upload then

luc-github, 2019-03-05:
Ok I close issue

closed 2019-03-05 by luc-github