luc-github/ESP3D #306: Update Canceled After 100 %

sancardu, 2019-03-13:
I have a problem with the embedded uploader, when i try to upload the index.html in the serial monitor it shows "M117 UPDATE 100%" but in the browser it says Update Canceled !
[image]

[image]
[image]

sancardu, 2019-03-13:
Even I can upload the SPIFFS Image but when i open the server it doesnt show
[image]

Anybody knows what im doing wrong?

sancardu, 2019-03-13:
Im using a ESP8266 With 1M ang 512KB SPIFFS

luc-github, 2019-03-13:
Try to enable puya patch , you may have  this  flash issue  https://github.com/luc-github/ESP3D/issues/229

sancardu, 2019-03-13:
So i found the esp.h [here](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Esp.h) but do i add it to the arduino project where the ESP3D.ino is?

luc-github, 2019-03-13:
No you update the one already in instaled in core

sancardu, 2019-03-13:
Ok done ! But still having the same program! When i try to upload the index.html.gz it goes to 90% and says Update failed and with index.html it goes to 100% but says Uodate canceled!
[image]

luc-github, 2019-03-13:
Can you connect your ESP8266 to your PC and open Arduino Terminal and send the following commands ?
1 - [ESP710]FORMAT
2 - give me the output of [ESP420]plain

sancardu, 2019-03-13:
Sure!

Formating...Done
Chip ID: 5795032
CPU Frequency: 160Mhz
Free memory: 41.39 KB
SDK: 3.0.0-dev(c0f7b44)
Flash Size: 1.00 MB
Available Size for update: 151.67 KB(Not enough)
Available Size for SPIFFS: 471.36 KB
Baud rate: 115200
Sleep mode: Modem
Channel: 11
Phy Mode: 11g
Web port: 80
Data port: 8888
Active Mode: Access Point (DE:4F:22:58:6C:D8)
SSID: ESP3D
Visible: Yes
Authentication: WPA
Max Connections: 4
DHCP Server: Started
IP: 192.168.0.1
Gateway: 192.168.0.1
Mask: 255.255.255.0
Connected clients: 0
Disabled Mode: Station (DC:4F:22:58:6C:D8) is disabled
Captive portal: Enabled
SSDP: Enabled
NetBios: Enabled
mDNS: Enabled
Web Update: Enabled
Pin Recovery: Disabled
Authentication: Disabled
Target Firmware: ???
FW version: 1.0

luc-github, 2019-03-13:
Ok does Arduino SPIFFS uploader works after formating ?

sancardu, 2019-03-13:
I think it doesnt work! But i dont know if i am doing something wrong, first I press and hold the reset button then I press and hold the program button, then Irelease the reset button, the ESP finally I release the program button and upload the sketch and finally I go to tools and Click ESP8266 Sketch Data Upload
[image]

luc-github, 2019-03-13:
Upload process for buttons is same process as flash, 100% identical, also must not have arduino serial terminal open

sancardu, 2019-03-13:
My bad! It does work but it says the same:

Chip ID: 5795032
CPU Frequency: 160Mhz
Free memory: 41.39 KB
SDK: 3.0.0-dev(c0f7b44)
Flash Size: 1.00 MB
Available Size for update: 151.67 KB(Not enough)
Available Size for SPIFFS: 471.36 KB
Baud rate: 115200
Sleep mode: Modem
Channel: 11
Phy Mode: 11g
Web port: 80
Data port: 8888
Active Mode: Access Point (DE:4F:22:58:6C:D8)
SSID: ESP3D
Visible: Yes
Authentication: WPA
Max Connections: 4
DHCP Server: Started
IP: 192.168.0.1
Gateway: 192.168.0.1
Mask: 255.255.255.0
Connected clients: 0
Disabled Mode: Station (DC:4F:22:58:6C:D8) is disabled
Captive portal: Enabled
SSDP: Enabled
NetBios: Enabled
mDNS: Enabled
Web Update: Enabled
Pin Recovery: Disabled
Authentication: Disabled
Target Firmware: ???
FW version: 1.0
[image]

luc-github, 2019-03-13:
so now you can upload index.html.gz - so what is the issue ?

sancardu, 2019-03-13:
It says that it uploads but it actually doesnt ! When i connect to the ESP8266 I see this page :
[image]

And it shouldnt show that because it is suppose to be already loaded,

And when i try to Upload the firmware update what I get is an Update Canceled

[image]

luc-github, 2019-03-13:
Check your flash size to be sure you have 1M https://github.com/luc-github/ESP3D/wiki/Flash-Size

sancardu, 2019-03-13:
Yes I have 1M

[image]

luc-github, 2019-03-13:
Well I am dry about what is happenning, try flash using another partionning like 194K spiffs

sancardu, 2019-03-13:
THANK YOUUU! IT WORKED WITH 192K SPIFFS! You have been really helpful!

sancardu, 2019-03-13:
Do you know if I can connect the ESP3D to a J3 Connector in an Anet A8?

closed 2019-03-14 by sancardu

luc-github, 2019-03-14:
Good - looks like a bug in core of ESP8266
Happy you worked around

I do not know, sorry. I do not have any Anet printer