luc-github/ESP3D #307: DHT22 esp32

Carrotiam, 2019-03-16:
Hello, tell me how to enable DHT22 for esp32.
The sensor is exactly working, when checked with a test sketch for esp32, the readings are displayed.
In the preferences of the web interface, "Show DHT output" is included, but I see only static 20 ° C and 20%.
Perhaps the FW version is not the same, or I’m doing something wrong, but in “ESP3D Settings” I don’t see the DHT22 item.

SDK: v3.2-dev-39-gaaf12390
FW version: 2.0.0.c23 ESP32

I apologize for Google translator.

luc-github, 2019-03-17:
if DHT is not visible in settings it means it is not enable at compilation, by default it is commented in config.h
Uncomment this line here : https://github.com/luc-github/ESP3D/blob/2.0/src/config.h#L77    

and use proper pin here : https://github.com/luc-github/ESP3D/blob/2.0/src/config.h#L114

Carrotiam, 2019-03-17:
I did right away, but unfortunately that's not the point.
Here are the screenshots.
[image]
[image]
[image]

luc-github, 2019-03-17:
Hi thanks for reporting, issue is not related to DHT detection 
 when you compile the ESP3D the config.h is inside the library as on my side when I enable DHT in config.h the DHT appear in settings
[image]
but as you show you do not have it so - DHT is not enabled in code 

When you recompiled you modified another config.h not the library one  this is my guess

Additionnaly looking at code I have found an ugly bug I never noticed before, I have fixed it but it is not related to your issue

Carrotiam, 2019-03-17:
You're right. It worked, not quite as I expected.
Thank you for help.
The topic can be closed.

luc-github, 2019-03-17:
You DHT is not working as you expected ? What is wrong ?

Carrotiam, 2019-03-17:
Wrong spoke.
ArduinoIDe does not work exactly as I thought)
DHT22 is working now, after your hint.
[image]

luc-github, 2019-03-17:
Ok. Cool have fun ^_^

closed 2019-03-17 by luc-github