luc-github/ESP3D #308: Errors when I turn on ASYNCWEBSERVER in ESP3D 2.0

HansJakob, 2019-03-17:
When I try to compile ESP3D 2.0 with ASYNCWEBSERVER defined, I get the following errors i arduino ide:

```
libraries\ESP3D\asyncwebserver.cpp.o:(.literal._Z20SDFile_serial_uploadP21AsyncWebServerRequest6StringjPhjb+0x34): undefined reference to `sendLine2Serial(String&)'

libraries\ESP3D\asyncwebserver.cpp.o:(.literal._Z20SDFile_serial_uploadP21AsyncWebServerRequest6StringjPhjb+0x38): undefined reference to `CloseSerialUpload(bool, String&)'

libraries\ESP3D\asyncwebserver.cpp.o: In function `SDFile_serial_upload(AsyncWebServerRequest*, String, unsigned int, unsigned char*, unsigned int, bool)':

C:\Users\hansj\Documents\Arduino\libraries\ESP3D\src/asyncwebserver.cpp:1200: undefined reference to `sendLine2Serial(String&)'

C:\Users\hansj\Documents\Arduino\libraries\ESP3D\src/asyncwebserver.cpp:1242: undefined reference to `sendLine2Serial(String&)'

C:\Users\hansj\Documents\Arduino\libraries\ESP3D\src/asyncwebserver.cpp:1244: undefined reference to `CloseSerialUpload(bool, String&)'

C:\Users\hansj\Documents\Arduino\libraries\ESP3D\src/asyncwebserver.cpp:1250: undefined reference to `CloseSerialUpload(bool, String&)'
```

HansJakob, 2019-03-18:
I have made a small Add-on module for my SKR v1.1 controller board based on an ESP32. I want a simple but effective way to add wifi to this controller.

I have started a UI based on Angular and where I want to use WebSocket to communicate with ESP3D. Would it make sense to look at 3.0?

luc-github, 2019-03-18:
2.0 on ESP32 works well will ESP32 with sync webserver
async faster on ESP32 but currently bring some stability issues and it is not 100% support on developer side because of his small bandwidth  that is why support is not 100% ESP3D

3.0 is only sync webserver for the moment and I did not do the webUI yet so not yet even in pre alpha, currently websocket is working as feedback of printer web commands  
but if you build your own Webui It should work

HansJakob, 2019-03-18:
OK. Async vs. Sync is not a big problem for me at this time. I just need to be aware to assemble everything in one file, when I move the UI to the embedded web server. I just wanted to make sure it wasn't me who made a mistake in the config. :-)

However, I have another question regarding WebSockets. I have made a small test UI that opens a connection to ESP3D via websocket. I get a connection and it responds with "CURRENT_ID: 0" and "ACTIVE_ID: 0", but then the connection becomes completely silent. I can understand on your response above, that I can only receive data from ESP3D via the web socket, but I have to send via "/command?CommandText=", but I did not expect the connection to be completely silent. If I enable websocket on your UI and look at the developer terminal in my browser, it's the same story, it gets the responds "CURRENT_ID: 0" and "ACTIVE_ID: 0", and then silence. Is there anything I should enable in ESP3D, in addition to "WS_DATA_FEATURE", to make it work?

luc-github, 2019-03-18:
There are 2 channels in websocket
Text one: this one I use it to simulate the event source to get connection ID. Send DHT data etc...
The binary one it is the one which do the bridge from serial. You cannot see it in console.log as it is not seen as text

luc-github, 2019-03-19:
@HansJakob  did you handled the websocket  ?

HansJakob, 2019-03-19:
Yes, thanks for asking and thank you for your help.

What throw me off was that ESP3D only sents data on websocket when it is unsolicited. Websocket is silent when using "/command?CommandText=".

After an intimate evening with your code, where I learned which "buttons" I should press, I now have all the building blocks to begin my project. ;-)

As I said, it will be an Angular project as I need to brush off my skills. At the same time I love the idea of using an ESP32 as a web server/wifi addon for a 3d printer. It is simple and elegant.

luc-github, 2019-03-19:
Which FW do you use as base ? 2.0 or 3.0 ?

HansJakob, 2019-03-19:
So far I've used 2.0 since I've used your UI as a refferance, but I compiled 3.0 yesterday and expect to use that going forward.

luc-github, 2019-03-19:
Yes I suggest to use 3.0 as base because code is far more clean and easier to understand - I separated every feature in a file so it is easier to understand

luc-github, 2019-03-19:
Fixed - I did a release as now the development version is 3.0
I did a 2.1 just because of setup - ESP3D is no more a library

closed 2019-03-19 by luc-github