luc-github/ESP3D #309: When Oled support is enabled I get panic error

ArturNadolski, 2019-03-19:
```
19:53:23.733 -> M117 Connecting..  
19:53:24.823 -> M117 Connecting... 
19:53:27.952 -> 
19:53:27.952 -> Panic C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp:103 __yield
19:53:27.986 -> 
19:53:27.986 -> >>>stack>>>
19:53:27.986 -> 
19:53:27.986 -> ctx: sys
19:53:27.986 -> sp: 3fffeb20 end: 3fffffb0 offset: 01b0
19:53:27.986 -> 3fffecd0:  00000000 4022053f 00000001 40221219  
19:53:27.986 -> 3fffece0:  00000010 00000010 00000000 4021ddc9  
19:53:27.986 -> 3fffecf0:  00000000 00000002 00000020 00000000  
19:53:27.986 -> 3fffed00:  3fff07ac 00000384 000004a8 00000006  
19:53:27.986 -> 3fffed10:  0000000a 00000000 00000000 00000007  
19:53:27.986 -> 3fffed20:  00000000 00000000 0000000c 00000000  
19:53:28.019 -> 3fffed30:  00000030 00000020 00000043 402204fc  
19:53:28.019 -> 3fffed40:  00000010 3fffeda0 3ffef2cc 00000030  
19:53:28.019 -> 3fffed50:  0000000d 3fff07ac 3ffef2cc 4021e0c0  
19:53:28.019 -> 3fffed60:  00000000 00000000 00000000 402205d4  
19:53:28.019 -> 3fffed70:  00000000 0000000a 3fff07ac 4022060a  
19:53:28.019 -> 3fffed80:  3fff0764 00000000 3ffef2cc 00000030  
19:53:28.019 -> 3fffed90:  00000080 00000000 3ffef2cc 4020d6e6  
19:53:28.053 -> 3fffeda0:  3fff077c 0000000f 0000000a 00000190  
19:53:28.053 -> 3fffedb0:  4023b422 3ffe95dc 3fff0764 0000000f  
19:53:28.053 -> 3fffedc0:  0000000a 0000000d 3ffee604 4023eac5  
19:53:28.053 -> 3fffedd0:  00000000 00000036 3ffec100 00000018  
19:53:28.053 -> 3fffede0:  401044a1 00040000 7fffffff 3ffeed98  
19:53:28.053 -> 3fffedf0:  3ffe88a8 00000000 3fff074c 4020d720  
19:53:28.053 -> 3fffee00:  3ffeda98 401033be 39363535 4020dd00  
19:53:28.053 -> 3fffee10:  00000010 00000010 00000000 40101409  
19:53:28.087 -> 3fffee20:  3fffc278 4022053f 3fffc200 00000022  
19:53:28.087 -> 3fffee30:  3fffee40 00dc47e4 3fff2214 00000000  
19:53:28.087 -> 3fffee40:  40101af6 00000030 0000001c 3ffeed98  
19:53:28.087 -> 3fffee50:  3ffe88a8 3ffe88a8 3fff074c 4020dd39  
19:53:28.087 -> 3fffee60:  4025f010 00000009 3fffeeb0 3ffeed98  
19:53:28.087 -> 3fffee70:  4025f010 00000009 3fffeeb0 40220648  
19:53:28.087 -> 3fffee80:  3ffef498 4025f010 3fffeeb0 40220868  
19:53:28.121 -> 3fffee90:  00000000 00000009 00000009 3ffeed98  
19:53:28.121 -> 3fffeea0:  3ffef498 00000009 00000009 4020ddc3  
19:53:28.121 -> 3fffeeb0:  3fff074c 0000000f 00000009 00000000  
19:53:28.121 -> 3fffeec0:  00000000 00c0265a 40100aa8 00009d32  
19:53:28.121 -> 3fffeed0:  00000000 3ffe96e4 00000000 4020ddfa  
19:53:28.121 -> 3fffeee0:  00b5c21a 3fff06b4 00000000 4021106e  
19:53:28.121 -> 3fffeef0:  40105cea 00bfc778 60000600 60000600  
19:53:28.121 -> 3fffef00:  02660ada 3ffeebe0 3ffeebb8 4022d6e8  
19:53:28.155 -> 3fffef10:  40105cea 00bfc778 3fff092c 40222222  
19:53:28.155 -> 3fffef20:  4024e84c 3ffeebb8 3ffeebe0 40212377  
19:53:28.155 -> 3fffef30:  4024e872 3fffdab0 00000000 3ffeed98  
19:53:28.155 -> 3fffef40:  3fffdc10 00000000 3fff06b4 402528b8  
19:53:28.155 -> 3fffef50:  40000f49 3fffdab0 3fffdab0 40000f49  
19:53:28.155 -> 3fffef60:  40000e19 40001878 00000002 00ce15b6  
19:53:28.155 -> 3fffef70:  3fffff10 aa55aa55 000000ed 4010565c  
19:53:28.195 -> 3fffef80:  40105662 00000002 00ce15b6 13250f02  
19:53:28.195 -> 3fffef90:  4010000d fbe5c936 b3eadfd6 293bcb18  
19:53:28.195 -> 3fffefa0:  40101604 3fffef3c 401015b1 3ffffda8  
19:53:28.195 -> 3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
19:53:28.195 -> 3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
19:53:28.195 -> 3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
19:53:28.195 -> 3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
19:53:28.195 -> 3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
19:53:28.195 -> 3ffff000:  feefeffe feefeffe feefeffe feefeffe  
19:53:28.229 -> 3ffff010:  feefeffe feefeffe feefeffe feefeffe  
19:53:28.229 -> 3ffff020:  feefeffe feefeffe feefeffe feefeffe  
19:53:28.229 -> 3ffff030:  feefeffe feefeffe feefeffe feefeffe  
19:53:28.229 -> 3ffff040:  feefeffe feefeffe feefeffe feefeffe  
19:53:28.229 -> 3ffff050:  feefeffe feefeffe feefeffe feefeffe  
19:53:28.229 -> 3ffff060:  feefeffe feefeffe feefeffe feefeffe  
19:53:28.229 -> 3ffff070:  feefeffe feefeffe feefeffe feefeffe  
19:53:28.229 -> 3ffff080:  feefeffe feefeffe feefeffe feefeffe  
[... 230 lines cut ...]
19:53:29.284 -> 3ffffef0:  00100000 00001000 00002000 00000100  
19:53:29.284 -> 3fffff00:  00000200 00002fb0 40223630 00000000  
19:53:29.284 -> 3fffff10:  00000000 00000202 3fff09d8 000000b1  
19:53:29.318 -> 3fffff20:  00000001 00000022 00000000 00000001  
19:53:29.318 -> 3fffff30:  3ffef400 3ffeed40 3ffef3bc 000000b1  
19:53:29.318 -> 3fffff40:  3ffef494 3ffeed40 3ffef3bc 4020cdea  
19:53:29.318 -> 3fffff50:  feefeffe feefeffe feefeffe feefeffe  
19:53:29.318 -> 3fffff60:  feefeffe feefeffe feefeffe feefeffe  
19:53:29.318 -> 3fffff70:  00001f40 feefeffe feefeffe feefeffe  
19:53:29.318 -> 3fffff80:  feefeffe feefeffe feefeffe 3ffefaa4  
19:53:29.318 -> 3fffff90:  3fffdad0 00000000 3ffefa74 4020d054  
19:53:29.352 -> 3fffffa0:  feefeffe feefeffe feefeffe 402211d0  
19:53:29.352 -> <<<stack<<<
19:53:29.352 -> 
19:53:29.352 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
19:53:29.352 -> 
19:53:29.386 -> load 0x4010f000, len 1384, room 16 
19:53:29.386 -> tail 8
19:53:29.386 -> chksum 0x2d
19:53:29.386 -> csum 0x2d
19:53:29.386 -> v951aeffa
19:53:29.386 -> ~ld
```
When Oled support is disabled, everything working fine.
Tested on LoLin NodeMcu v3 (ESP-12E)

luc-github, 2019-03-19:
please decode the stack - this output is not really useful - only decoded stack could give some clue

what settings do you use in config.h and what pins for the screen ?

ArturNadolski, 2019-03-19:
#define OLED_PIN_SDA  5
#define OLED_PIN_SCL  4

`

Decoding stack results
0x4022053f: String::changeBuffer(unsigned int) at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 156
0x40221219: __yield() at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 105
0x4021ddc9: OLEDDisplay::drawStringInternal(short, short, char*, unsigned short, unsigned short) at C:\Users\Artur\Documents\Arduino\libraries\ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display\src\OLEDDisplay.cpp line 855
0x402204fc: String::invalidate() at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 141
0x4021e0c0: OLEDDisplay::drawString(short, short, String) at C:\Users\Artur\Documents\Arduino\libraries\ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display\src\OLEDDisplay.cpp line 469
0x402205d4: String::copy(char const*, unsigned int) at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 182
0x4022060a: String::String(char const*) at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 36
0x4020d6e6: OLED_DISPLAY::display_text(char const*, int, int, int) at C:\Users\Artur\AppData\Local\Temp\arduino_build_853032\sketch\esp_oled.cpp line 222
0x4020d720: OLED_DISPLAY::print(char const*) at C:\Users\Artur\AppData\Local\Temp\arduino_build_853032\sketch\esp_oled.cpp line 106
0x40234c6c: mem_malloc at core/mem.c line 210
0x4020dd00: ESPCOM::print(char const*, tpipe, ESPResponseStream*) at C:\Users\Artur\AppData\Local\Temp\arduino_build_853032\sketch\espcom.cpp line 288
0x40101409: realloc at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\umm_malloc\umm_malloc.c line 1504
0x4022053f: String::changeBuffer(unsigned int) at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 156
0x4020dd39: ESPCOM::print(char const*, tpipe, ESPResponseStream*) at C:\Users\Artur\AppData\Local\Temp\arduino_build_853032\sketch\espcom.cpp line 300
0x40220648: String::copy(__FlashStringHelper const*, unsigned int) at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 192
0x40220868: String::operator=(__FlashStringHelper const*) at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 252
0x4020ddc3: ESPCOM::print(__FlashStringHelper const*, tpipe, ESPResponseStream*) at C:\Users\Artur\AppData\Local\Temp\arduino_build_853032\sketch\espcom.cpp line 208
0x4022b6fc: dhcp_start at glue-esp/lwip-esp.c line 421
0x4020ddfa: ESPCOM::println(__FlashStringHelper const*, tpipe, ESPResponseStream*) at C:\Users\Artur\AppData\Local\Temp\arduino_build_853032\sketch\espcom.cpp line 314
0x4021106e: onWiFiEvent(WiFiEvent) at C:\Users\Artur\AppData\Local\Temp\arduino_build_853032\sketch\wificonf.cpp line 180
0x40222222: std::_Function_handler ::_M_invoke(const std::_Any_data &, _esp_event *) at c:\users\artur\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40212377: ESP8266WiFiGenericClass::_eventCallback(void*) at c:\users\artur\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/stl_list.h line 157
0x40229946: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x402253b4: _printf_i at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf_i.c line 194
0x40229946: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x4022112c: esp_yield() at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 91
0x4022053f: String::changeBuffer(unsigned int) at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 156
0x4022120a: __yield() at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 100
0x4021ddc9: OLEDDisplay::drawStringInternal(short, short, char*, unsigned short, unsigned short) at C:\Users\Artur\Documents\Arduino\libraries\ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display\src\OLEDDisplay.cpp line 855
0x40101128: _umm_free at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\umm_malloc\umm_malloc.c line 1300
0x402018f5: twi_write_byte at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_si2c.c line 240
0x40201ad1: twi_writeTo at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_si2c.c line 260
0x4021d810: TwoWire::endTransmission(unsigned char) at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\Wire\Wire.cpp line 153
0x4022053f: String::changeBuffer(unsigned int) at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 156
[... cut ...]
0x40222004: spiffs_hal_erase(unsigned int, unsigned int) at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\spiffs_hal.cpp line 168
0x4020cdea: Esp3D::begin(unsigned short, unsigned short) at C:\Users\Artur\AppData\Local\Temp\arduino_build_853032\sketch\esp3d.cpp line 191
0x4020d054: setup() at C:\Users\Artur\Documents\Arduino\ESP3D-2.1\esp3d/esp3d.ino line 30
0x402211d0: loop_wrapper() at C:\Users\Artur\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 122
`

luc-github, 2019-03-19:
I cannot reproduce your issue
What others settings did you changed in config.h ?
What is the version of the FW 

Do you use Git version of ESP8266 core ?

ArturNadolski, 2019-03-19:
ESP core is "gited"

You just point me on something...

esp_display.drawXbm(33, 10, ESP3D_Logo_width, ESP3D_Logo_height, ESP3D_Logo);
invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]

I forgot that I change const char  ESP3D_Logo[] to const uint8_t  ESP3D_Logo[] as an workaround of this error

ArturNadolski, 2019-03-20:
After reclonning ESP8266 core from git now invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive] error disappear. There is still a lot of warnings but compilation ends with success. Oled display still not working but there is no panic error.

luc-github, 2019-03-20:
Ok I think issue come from PROGMEM but I do not get 
```
const char  ESP3D_Logo[]
```
and 

```
// Draw a XBM
    void drawXbm(int16_t x, int16_t y, int16_t width, int16_t height, const char *xbm);
```
both use const char * why do you have a conct uint8_t * ?

luc-github, 2019-03-20:
Ok I see you do not use the on I provide (v3.2.7 )- you use another one V4 may be ?

ArturNadolski, 2019-03-20:
> Ok I see you do not use the one I provide (v3.2.7 ) - you use another one, V4 may be ?
Sorry but I don't get it. Version of what? Provided where?

luc-github, 2019-03-20:
the version of Oled library - I provide the libraries to be used in library directory

ArturNadolski, 2019-03-20:
Now I'm using provided lib. Only first compilation with enabled Oled was using external lib.
Currently project is compiling with success but display doesn't work. I will try to check signal on SDA SCL pins.

luc-github, 2019-03-20:
just do [ESP444]RESET to be sure OLED output is enabled

ArturNadolski, 2019-03-20:
OK,  I will try at home.
BTW when You plan release v3.0?

luc-github, 2019-03-20:
When done ^_^ - sorry no clear date, it is a lot of work to rewrite all - I am now on pure FW part - webUI is not yet ready, but will be this year hope before summer come
just started with oled and today working on tft + touch

ArturNadolski, 2019-03-20:
Finally I get Oled working. The root of a problem was Arduino environment. My fault - too much libs and too much modifications.

closed 2019-03-20 by ArturNadolski