luc-github/ESP3D #311: JSON via WebSocket.

HansJakob, 2019-03-22:
When/if you implement two-way communication via WebSocket, would you be so kind as to consider JSON.

Right now when the WebSocket is just one-way, it is not a problem. But it would be so much easier if all the communication was via the WebSocket and using JSON. :-)

luc-github, 2019-03-22:
The pure websocket communication is planned in 3.0
But I do no see the link with JSON

HansJakob, 2019-03-22:
It's really just a convenience thing. :-)

Right now, nothing else happens on WebSocket than ESP3D sending the serial output from the printer. But, for example when connecting to the ESP3D, it responds with:

currentID: 0
activeID: 0

It would be more convenient (and arguably more robust) if it was formatted as JSON.

It would also be great if the client could send in JSON format. For example, if the client was to update the configuration, the entire configuration object could be sent instead of having to send each parameter via [ESP401].

It's just a little thing, and there are probably problems with it, that I can't see right now.

luc-github, 2019-03-22:
I do not see how to pack every feedback from Printer in JSON format, how to consider a printer output is part of one json package vs another ?

about event output : EVENTID: EVENT data, it is adding extra work and bandwidth from ESP3D for an  identical result with an arguably benefit.

About atomic change using [ESP401] and using batch change using another command (JSON or other format)  is not related to JSON IMHO. I did it because of code simplification.
What happen if one parameter is not correct ? Cancel all ? Rollback to previous state for all already applied settings ?
I do not says it is bad idea but such change means a lot of code change and it is not so easy to capture all user cases.

And what you suggest is to remove the webserver and use pure websocket for communication ?

HansJakob, 2019-03-22:
I see your point. You are right that it is easier to handle these things on the client side. I humbly withdraw my proposal . :-D

As for  pure websocket for communication... It would be great for me. It would eliminate the need to keep track of two data sources. However, 3.0 has already come a long way. If I send a command via /command?cmd=, it resounds via WebSocket. This I don't think 2.0 did. This means I don't have to listen to http and merge it into my response event. But I don't know the consequences on the server side.

luc-github, 2019-03-22:
on 3.0 curently commands are sent by web commands and only ESP commands are sent back using web because it use same communication to answer.    
If you use telnet to send ESP command, the answer will go to telnet, same for serial, I use same 'pipe' for in and out.
When I will enable the input websocket, it will follow same process so you won't need to listen http anymore.
The reason I did not do yet,it  is because it need to handle also file upload to remove the need of http so I need more work on this.
Finally using pure socket also imply a  need of a dedicated client, because just connecting with phone / tablet won't work, every phone has web browser by default, but none has a dedicated websocket client,  some application like laser web already have such feature, but I do not know any printer application using pure websocket - that is why it was not a top priority for me to implement websocket.
Additionally using telnet protocol is faster than websocket protocol, that is why it is also the prefered protocol until now.

HansJakob, 2019-03-22:
Hmm. Then it might be a bug, that if I send a /command?cmd=M122 I don't get a response via http, but get it through my WebSocket connection.

I totally agree with you that the web server is needed, as a web application is the easiest and most widely used tool. What I was thinking about was the communication between the web application and the server should be possible to be purely WebSocket. But I can also hear that is the plan.

luc-github, 2019-03-22:
yes M122 is printer command not ESP command, ESP command are [ESPXXX]YYYYYYY

HansJakob, 2019-03-22:
Ahh, sorry. My mistake.

luc-github, 2019-03-28:
3.0 has now websocket bridge : https://github.com/luc-github/ESP3D/commit/6f5d87a92708c8f3ed177cee4ce9e154444c582d
configuration can be done with [ESP160] and [ESP161] default port is 8282
I have also wrote a simple WebSocket monito it is in tools directoryr as now Websocket can be used also for debug
[image]

I did not tested fully as bridge yet but basic tests seems ok
[image]

closed 2019-03-28 by luc-github

HansJakob, 2019-03-28:
That sounds cool. I will immediately implement it in my own interface. :blush:

[image]