luc-github/ESP3D #312: Always Loading....

mjozwiakk, 2019-03-29:
Hello,

I've uploaded GRBL for ESP32 to my ESP32 DevKit V1 board, everything looks fine - in serial port monitor i can see, that esp is connected to my netword, also i can control all the functions using every G-code sender. 
When I try go to WebUI there is only:
"Loading..."
on the screen, and nothing happens. Any ideas how to fix it?

luc-github, 2019-03-29:
I do not know may be the HTML is corrupted or filesystem table has been modified after you uploaded the index.html.gz
connect to your ESP by serial and send this command to format the file system
`[ESP710]FORMAT pwd=admin` 
then connect your ESP you should have the following UI to help to upload the index.html.gz present in data directory of GRBL_ESP32
[image]

mjozwiakk, 2019-03-29:
Thank you for the response.
I think that the problem was that i did some modifications in flash memory. 
Uploading factory firmware and then grbl_esp32 solved the problem.

luc-github, 2019-03-30:
Ok great, I  close issue then

closed 2019-03-30 by luc-github

mjozwiakk, 2019-03-30:
Interesting thing was that clearing flash using esptool (erashe_flash) did not work. Only after uploading the default firmware it helped.

luc-github, 2019-03-30:
Default FW ?

mjozwiakk, 2019-03-30:
From this site
https://www.espressif.com/en/support/download/at