luc-github/ESP3D #313: Printer configuration is not displayed

Roman-RV78, 2019-04-11:
Hello. 
On the printer tab, the printer configuration is not displayed. Everything else works.
Arduino Due, marlin 2.0, esp3d UI: 2.1b23 / FW: 2.1.0.c26.
When updating the printer configuration in the command window, I observe: Unknown command: "*"
In the program Repetier-Host configuration printer displays.
Translated using google.

luc-github, 2019-04-11:
What is the output of M503 in terminal ? Does it display Marlin configuration ?
Where you from Roman ?

Roman-RV78, 2019-04-11:
Yes, after entering the command M503 I see it:
M503
  G21    ; Units in mm (mm)
  M149 C ; Units in Celsius

  M200 D1.75
  M200 D0
  M92 X80.00 Y80.00 Z3200.00 E302.00
  M203 X150.00 Y150.00 Z20.00 E80.00
  M201 X600.00 Y600.00 Z50.00 E10000.00
  M204 P600.00 R2000.00 T600.00
  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
  M206 X0.00 Y0.00 Z0.00
  M420 S0 Z0.00
  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
  M301 P24.58 I3.27 D46.20
  M304 P168.41 I11.57 D613.00
  M851 Z-0.86
  M603 L10.00 U700.00

I'm from Russia

luc-github, 2019-04-11:
weird 
what version of Marlin 2.x ? latest Git version ?

Roman-RV78, 2019-04-11:
Yes, marlin Bugfix-2.0 latest version

luc-github, 2019-04-11:
I need to setup a system with marlin 2.x to try to reproduce - may something changed since I tested last time
can you do a screenshoot of the printer window - I want to see error
and also another one of ESP3D info window - where you see SDK version, free memory etc...
So I will have all informations to reproduce

Roman-RV78, 2019-04-11:
[image]
[image]
[image]

luc-github, 2019-04-11:
one comment not related to issue - in AP mode ESP can be slow because every request go to it, AP mode is more destined for initial configuration, you better use it connected to an access point

luc-github, 2019-04-11:
what printer tab display ?

Roman-RV78, 2019-04-11:
[image]
There is no information here.

luc-github, 2019-04-11:
ok thank you 
I will setup a system and try to reproduce

luc-github, 2019-04-11:
Ok ihave reproduced it  - I am checking now

luc-github, 2019-04-11:
ok 
* `*` is a typo issue no idea how it happen after command M503 - fixed
* output format changed again - so output is not recognized - working on it

luc-github, 2019-04-11:
@Roman-RV78 format changed from \t to 2 spaces it is updated
you just need to update webui in esp3d filesystem panel
https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz
thank you for the feedback

Roman-RV78, 2019-04-11:
[image]
Yes, everything works

luc-github, 2019-04-11:
Cool

closed 2019-04-11 by luc-github

ShadowOfTheDamn, 2019-10-03:
Hi @luc-github 
I am using marlin 2 current GIT version and Im having the same kind of problem. exactly like what roman-rv78 showed empty configuration page for 3d printer while the M503 works in the console menu. can you please check to see the bug still exist. maybe marlin changed sth.
thanks a lot

luc-github, 2019-10-03:
Please open new issue i need more informations as you can see issue was solved. So rpot cause may be different

ShadowOfTheDamn, 2019-10-03:
wow fast reply.
I will right now, ty.

ak10ng, 2020-02-01:
Please help for Marlin 1, using ESP32. No Printer configuration
[image]

ShadowOfTheDamn, 2020-02-01:
> Please help for Marlin 1, using ESP32. No Printer configuration

visit this page: 
https://github.com/luc-github/ESP3D-WEBUI

Please use the latest ESP3D firmware and then from ESP3D-WEBUI copy the index.html.gz file on root of SPIFFS, in theory ESP3D have a version of web-ui but it may not be the latest one.
Hope this helps
Regards
N. Nouri

ak10ng, 2020-02-03:
Solved. Upgrade FW using Marlin 2. Reflash ESP3D and upload index.html.gz from ESP3D-WEBUI. Board Anet 1.5. Thanks.