luc-github/ESP3D #315: I can not see the files which are in sd card and I can not upload gcode into sd card

PandaofPython, 2019-04-16:
Hello everyone! I have a Nodemcu amica and mks tft35 v1.0.  First of all, I tried to upload the esp3d into mks wifi tft v1.0(ESP-12S) which is already included in it. then I couldn't upload esp3d into mks wifi tft v1.0 . So I decided to use the Nodemcu amica for esp3d and then I was able to upload esp3d into it. I can see the web interface of esp3d which uploaded in nodemcu but I can't see the files of sd card which plugged into mks tft35 v1 and I can't see any information. Please help me how I fixed this problem or how I upload esp3d into mks tft wifi v1.0.

luc-github, 2019-04-16:
MKS tft is not a printer and use special protocol to communicate with ESP to enable bridge with printer
we are working on it https://github.com/luc-github/ESP3D/issues/249 but slowly

Also about how access SD connected to MKS TFT by ESP - I am not sure it is possible, the GCODE commands for query / access SD are for SD connected to printer in theory, may be using M998 1 but did not tested as we work on make ESP3D recognized by MKS TFT to enable wifi bridge first

PandaofPython, 2019-04-16:
thank you so much. But can I use any another module different from ESP modules? Would you like to advise me any method for mks local connection with it?

luc-github, 2019-04-16:
what you mean ? 
Your mks wifi tft v1.0 with original MKS FW (not ESP3D) should work to make your MKS TFT having WiFi

PandaofPython, 2019-04-16:
I want to use nodemcu instead mks wifi tft v1.0 . Can I communicate nodemcu with mks tft35 ?

luc-github, 2019-04-16:
Any ESP board will work with proper FW, but ESP3D currently does not work with MKS TFT , we are working on it, please read the thread I gave you to know the status

PandaofPython, 2019-04-16:
oh sorry, that was my bad. Thanks for answering. Have a good day.

luc-github, 2019-04-16:
NP feel free to participate on thread I gave you

closed 2019-04-16 by luc-github