luc-github/ESP3D #316: File upload to SD not working properly

MuellerDominik, 2019-04-20:
I'm using ESP3D 2.1 with ESP3D-WEBUI (index.html.gz from 2.1 branch) and I've found that the SD upload is not working properly.

My target is running Marlin 2.0 (bugfix-2.0.x branch).

### File
I've created a short test file and tried to upload it, using the web interface.

Contents of the file `test.gco` on the host:
```
G21
G0 Y10 F10000
G0 Y0 F10000
G4 S3
```

Contents of the file `test.gco` on the target (after manually executing `M29`, to resolve the error state):
```
G21
G0 Y10 F10000
G0 Y0 F10000
G4 S3
M29
```

As you can see the `M29` command (Stop SD write) is written to the file. That's why I had to manually execute the `M29` command, to resolve the error state (using the web interface console).

### Communication
I have monitored what ESP3D is sending and receiving.

TX (ESP3D):
```
N1 M29*25
N7 M29*31
M21
M110 N0
M117 Uploading...
N1 M28 /test.gco*68
N2 G21*24
N3 G0 Y10 F10000*5
N4 G0 Y0 F10000*51
N5 G4 S3*104
N6 M29*30
M117 SD upload done
M20
```

RX (ESP3D):
```
Error:Line Number is not Last Line Number+1, Last Line: 6
Resend: 7
ok
ok
echo:SD card ok
ok
ok
ok
echo:Now fresh file: /test.gco
Writing to file: test.gco
ok
ok
ok
ok
ok
ok
Error:No Checksum with line number, Last Line: 6
Resend: 7
ok
Error:No Checksum with line number, Last Line: 6
Resend: 7
ok
```

As far as I can tell, Marlin is behaving the way it should (correct me if I’m wrong).

Why is the `M28` and `M29` command prefixed by a line number? I think that's what's breaking things. I'm surprised it even created a new file at all (since the `M28` command is prefixed as well).

I'd expect the communication to be something like this:
```
M21
M110 N0
M117 Uploading...
M28 /test.gco
N1 G21*<checksum>
N2 G0 Y10 F10000*<checksum>
N3 G0 Y0 F10000*<checksum>
N4 G4 S3*<checksum>
M29
M117 SD upload done
M20
```

luc-github, 2019-04-20:
M29 need line number last time I have tested and it was not fixed I think : https://github.com/MarlinFirmware/Marlin/issues/11871

about M28 using Line number I do not see the problem, any command can use it.

and I tested it before it was working.

about M29 for some reason is not taken in account when it was before - Marlin always change behavior so I may missed the update but looking at issue it is not

luc-github, 2019-04-20:
and to clarify I would love not to have to handle this M29 command with line number/ checksum, it bring lot of code complexity when it should not : https://github.com/MarlinFirmware/Marlin/issues/11871#issuecomment-424094214

luc-github, 2019-04-20:
@MuellerDominik what baudrate are you using ? and what is your printer board that use Marlin 2.0 ?

MuellerDominik, 2019-04-20:
You're right, it should always be possible to use line numbers in conjunction with a checksum.

The problem seems to be, that Marlin is doing something weird when parsing the `M29` command. It appears, that the command is treated differently when used with line numbers and without.

What I've found so far is, that when the `M29` command is used without a line number it does what it should and stops writing to the file (just the way you would like it). When used with a line number, it writes the `G29` command to the file, which as you said should not happen.

This is funny because the way I understand the problems described in the Marlin issue 11871, it was exactly the other way around before. I'll try to find the error in the Marlin firmware.

Currently I'm not using an existing printer board. I'm using a STM32F103ZET6 development board. We're designing a new printer board as a school project. At the moment I'm using a baud rate of 250000 (default of Marlin). This should not really be an issue, since the commands are received and sent correctly. They're are just not interpreted properly as it seems.

luc-github, 2019-04-20:
Yes , I agree your baud rate should not be the root cause.

and yes it this issue is really confusing, looks like there was some update in Marlin which bring more mess on this issue.
If you find out what is happening please let me know - I have currently no bandwidth to investigate on Marlin code.

Good luck with your project

MuellerDominik, 2019-04-20:
Wow, I just found and fixed the issue when I saw this: https://github.com/MarlinFirmware/Marlin/pull/13013

The problem was fixed 5 days ago in the following commit (now, the `M29` command is accepted properly with and without a line number): https://github.com/MarlinFirmware/Marlin/commit/92ed9a2bb6d90a969f2ceaf0a73ebe3c0cdc4c16

My version of Marlin is about 9 days old...

I have one last question though, why is `ESP3D` sending the following before the `M21` command (as seen in my original post):
```
N1 M29*25
N7 M29*31
```

luc-github, 2019-04-21:
Ok looks like only for 2.0 not for 1.X
the initial M29 is a sanity check to close any open transfer, the first failed because not proper line number used but allow to get proper line number, the second succeed or do nothing if was not necessary

MuellerDominik, 2019-04-21:
Okay, nice solution to make sure there are no open transfers.

I do not currently have a running v1.1.x of Marlin, but as far as I can tell it shouldn't be an issue there:

1. It properly allows the `G29` command without the line number [see here](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L1094)
2. It closes the file after receiving a `G29` command (with or without a line number) [see here](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L15331)

The only problem I see there is, that they only check if `M29` is a substring of the current command (bullet point nr. 2). If in fact the command is `M290` (again, doesn't matter with or without a line number) it would find the substring `M29` and stop the SD transfer.

The solution applied 5 days ago to the bugfix-2.0.x branch involves an additional check, if the following character is numeric [see here](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/queue.cpp#L531).

Since the `M290` command is available since v1.1.7, this could lead to a problem. I'll open an issue an let them know.

luc-github, 2019-05-03:
I close issue as it. Is Marlin issue and it is solved

closed 2019-05-03 by luc-github