luc-github/ESP3D #319: MKS GEN V1.0 or 1.2 - LCD TFT and ESP8266

quichy3016, 2019-04-25:
Hello, good afternoon, sorry, but I use a translator.
 The subject is the following I have an MKS Gen L V1.0 in which I already made the communication using the ESP8266 module and the ESP3D firmware and it was a success. To do this, disconnect my MKS TFT32 from the AUX 1 terminal and at pins TX0 and RX0 connect the pins of the ESP8266. I would like to connect the MKS TFT32 to the AUX 1 and connect the ESP8266 module to the same pins for some resistors like this one and to the TX / RX pins of the AUX 1 with the USB (serial BUS type).
The query is what resistance values ​​should be used to make the connection.
Thank you.

luc-github, 2019-04-25:
This is not a good idea you will have 2 devices sharing same TX, and will send requests n same tme causing gabage on line due to mixed data flow

quichy3016, 2019-04-25:
It is not the same when you have connected the USB cable and also the TFT32 screen?

luc-github, 2019-04-25:
if you connect to same TX - yes but many boards block one of them when both are connected at least Arduino Mega do - now you can try if the esp and TFT do not send data in same time it should be ok, but TFT send permanent requests to get info

luc-github, 2019-04-25:
connection suggestions are in wiki : https://github.com/luc-github/ESP3D/wiki/ESP8266-01 and https://github.com/luc-github/ESP3D/wiki/ESP8266-01

quichy3016, 2019-04-25:
Así lo podríamos conectar directamente a los pines tx0 y rx0 recién el TFT conectado?

luc-github, 2019-04-25:
Depende de la configuración del FW Marlin / Repetier.
si el FW admite TFT en TX0 / RX0 y ESP en TX / RX, se puede

quichy3016, 2019-04-25:
En el pin tx no llevaría resistencia para que no le ll3gu3 una corriente cuando otro tx este activo??

luc-github, 2019-04-25:
No entiendo la pregunta, TX0 y TX1 son diferentes.
Como lo he escribo  antes, tener 2 dispositivos en un mismo TX no es correcto

luc-github, 2019-05-12:
ok if no update I close issue

closed 2019-05-12 by luc-github