luc-github/ESP3D #320: FR: CIFS server

iz3man, 2019-04-25:
Not an issue at all, but a feature request: Uploading files through the webinterface is nice, but **mounting the printer's file system as a network share drive** would be more convenient in many cases. Open the stl in the slicer, then just save it to the network drive (printer's SD card) directly is way faster.
CURL and webdav is fine, if you're on Linux (& Mac) and stuff, but not if you're using Windows as your main computer to design and slice.

luc-github, 2019-04-25:
that is what webdav do no ?
https://www2.le.ac.uk/offices/itservices/ithelp/my-computer/files-and-security/work-off-campus/webdav/webdav-on-windows-10

iz3man, 2019-04-25:
Hmmmm. Somehow. Maybe it's a way to do it :)
Used webdav under Linux for quite some time, but tried it long ago under Windows and was not convinced. Maybe i'll give it another try.

luc-github, 2019-04-25:
in olds windows it was not so easy I guess- but now it is Ok flashair use WebDav also to mount the card https://jp.toshiba-memory.com/ww/support/pdf/FlashAir_drive_(WebDAV).pdf

When I was doing research for low footprint code to mount a drive under windows  only webdav popup to me - unless you know another way 
doing smb server or even ftp server will be more complex I think

iz3man, 2019-04-25:
I bought a FlashAir just for that purpose but i never connected to my Ubiquity router. So i came to ESP3D. Flahed it yesterday to an ESP-12s and all worked fine. Only issue was 8.3 naming. So i bow look into MK4DUO. 
And being able to save gcode directly to the mounted SD card over wifi would be the icing on the cake. 
I'm no coder at all, so i have no idea how hard it is to implement smb/cifs code into 4Mb flash ;). I will try webdav when back from travel tomorrow and let you know how good it works. 
Thanks!

luc-github, 2019-04-25:
@iz3man be noted that serial file upload is slow like hell especially if you have big files (https://github.com/luc-github/ESP3D/issues/292), upload using the flashair is 100x faster 
I also got issue to configure webdav with flashair I was not conviced on their implementation and never used - but other people told me it was working

FYI : Soon Panucatt devices (http://www.panucatt.com/Default.asp) with sell some boards allowing to share SD card between ESP and Printer, depending on printer available connections, in same way as Azteeg WiFi do, it use custom version of ESP3D and upload are faster than serial upload.

iz3man, 2019-04-25:
I know serial transfer is really slow as it's limited to 250k serial speed. And .gcode files tend to be quite big. Several Mb. The transfer limit should be 1.2 mega byte per minute. But still better then running down to the basement, copying the file, then going back again ...
Maybe I'll think about a RPI for that purpose. Should speed things up a lot as well.
Does your Flashair implementation change the firmware of the card as well? I'm not sure where the problem with mine lies. It never worked in STA mode, only AP mode, and if so it way a PITA to use. It was SUPER slow. I'm afraid i received a fake card from China?! Do you know of an alternative firmware or anything to test the card?

luc-github, 2019-04-25:
I always use my flashair using latest FW from toshiba, it is easy to update, and in STA mode.
 No my code is just settings and html/pages no FW changes

iz3man, 2019-04-25:
Ok. Thanks. Well I have Version 03 card, with latest FW from Toshiba and it's just not working. Maybe it's broken. :(

luc-github, 2019-04-26:
ok I play with some sample code and got one standalone webdav remote dir working on ESP32
[image]

[image]

I see  3 issues
* WebDav port need to be 80 so need to see how to make working with existing  webserver using port 80, from another point of view, if webserver is not using port 80, need a separate server to support webdav on port 80
* WebDav space is wrong but it is a known issue : https://support.microsoft.com/fr-fr/help/2386902/webdav-mapped-drive-reports-incorrect-drive-capacity
* I need to dig more to see how to add authentication as currently no authentication

A13xund3r, 2019-05-06:
> I know serial transfer is really slow as it's limited to 250k serial speed. And .gcode files tend to be quite big. Several Mb. The transfer limit should be 1.2 mega byte per minute. But still better then running down to the basement, copying the file, then going back again ...

I was hoping this calculation will apply and tried 2Mbit. Serial line speed is not the limit here. Apart of data loss, there was still slow transfer about 15kbit, unfortunately 8bit atmega is too slow for this purpose.

> Maybe I'll think about a RPI for that purpose. Should speed things up a lot as well.

For some time I've used octoprint on SBC, was working fine but I was not happy using another computer when printer stand-alone is working very well.

I think this is good moment to lobby again in favor of hardware solution [https://github.com/luc-github/ESP3D/issues/296#issuecomment-460585858](url) but without webdav functionality which I see is not the best solution. IMHO currently implemented file upload feature can be used to send data directly to card FS instead of serial port. Of course there should be option for user in config which way it have to be done.

luc-github, 2019-05-11:
Boards are available Boards are now available https://www.panucatt.com/ProductDetails.asp?ProductCode=WB8266
```
Versions:
Viki LCD Wifi Backpack - For Viki2 LCD
RRD LCD Wifi Backpack - For RepRapDiscount Full graphic(128x64) and Smart Display Controller (20x4 LCD)
Wifi Puck - Stand alone module with built-in SD card slot (for machines without displays)``

iz3man, 2019-05-13:
@luc-github That sounds great. Could you please go into a bit more detail what the boards do? I don't see any description. Will it allow full speed access to the SD card? Unfortunately, as you said: serial communication is REALLY slow. Really too slow to be useful.
I have a rerap full graphics display which has 2x 10pin headers. This is where the Wifi Backpack Module connects to?

iz3man, 2019-05-13:
Another question: Does the module support FULL LONG filenames with Marlin? If so it'd be an instant buy.

luc-github, 2019-05-14:
@iz3man 
* Yes direct SD card access , SD card is shared between ESP and printer board - when one is using the other has no access, so faster than serial, Roy from Panucatt is still working on Documentation but he already took your feedback in account for description:
`Direct access to SD card by both ESP module and main controller`

* yes here the board for full GLCD
[image]

* about shipping cost, Roy is also working on it, he told me that he was able to get $6.75 for international shipping, TBC as I just transmit feedback and answers ^_^

* about long names in Marlin, sorry not yet, during my early tests I had some issues using long names with upload on SD with Marlin, Marlin use a specific SD library which is very peaky, so I left the limitation - and to be honest I forget it because workaround Marlin's limitation was not my top priority, allowing fast upload was more important. 
That said, I am currently working on new SD library implementation and will review this part to see if I can remove the limitation, thank you for reminding me

iz3man, 2019-05-14:
@luc-github Thanks for the long reply. That's very welcome. I guess I'll order one. 
Will the daughter board be upgradeable? I don't want to be stuck with the 8.3 limitation, as this is quite a pita to handle, as i use a lot of versioning and description in my filenames to identify them weeks later again, so 8 character will not do it. ;)
I will contact Roy directly, thanks for your support!

luc-github, 2019-05-14:
>Will the daughter board be upgradeable?

I guess you talk about FW?
yes FW upgrade will be possible of course - FW update can be done by webUI or automaticall by SD at start - just need to go to panucatt web side to get it

From HW point of view there is one pin available to connect a DHT22 sensor but it is not yet documented I think

iz3man, 2019-05-14:
Yes. FW upgrade. So it runs your printer control software, correct?
DHT22, hmmmm. I always wanted to know my printer's temperature ;)

luc-github, 2019-05-14:
Yes it is customized version of ESP3D - but sources are not public yet

A13xund3r, 2019-05-14:
>  Yes it is customized version of ESP3D - but sources are not public yet

Ah... here we go, I was curious what is your relation with panucatt so you was resisting SD upload idea so strong. Now I understand. Don't get me wrong, most likely will get one for printer I use at work, especially for this price, but at home, I have esp module already and skills to design, etch and solder PCB. I see on the photos it use multiplexer IC to isolate SD card from printer board, completely unnecessary for me. Will wait for source of this custom firmware.

luc-github, 2019-05-14:
@A13xund3r relation with panucatt is not new neither hidden, it is in readme ^_^- same code is already used in Azteeg Wifi for a year, these daugther boards are just the way to extend feature to other printers boards.

Why I resist to release ESP3D direct SD access sources?
* Everyone is ok to pay even a little for hardware, almost no one for software...It is not new, it has always been the case...
* I spend a lot of time on this projects: coding and giving support, I also spend money because I need to buy new  boards/components to support/test new features  so getting some support is nice as I do not have income, and I cannot rely on donation : 7 people donate in 3 years (big thanks to them). for an average of 3$ to 10$ which does not fit the needs for components.

So I think it is fair that I share 100% of code (fully fonctional and supported by community)  and delay for a specific part (only supported by me and related to specific hardware solution) a while to give some advantage to the people who supported me the most. 

And as you said there are several ways to share the SD,  ESP3D is not the 'only solution' ^_^
And do not get me wrong : I am not complaining - I just play the game as it is. 😸

A13xund3r, 2019-05-15:
Fully understand, I was curious but not enough to read readme in details (nobody read it), just section needed for installation. 
In terms of hardware this is where community comes, we are here to support you in making code better. It is impossible for you to have every possible hardware.
Is there date you plan to release code with SD support? Approximate?

luc-github, 2019-05-16:
> I was curious but not enough to read readme in details (nobody read it)   

I know, even installation section is not often read - that is why I do not spend so much time on it 😸 

>In terms of hardware this is where community comes,   

Well it is hard to add support for DHT if you do not have one, same for ESP32 ethernet module, oled lcd, tft etc... for new for new boards with same feature, I agree I cannot test all and  there is no point for this as it is either a bug in code or a hardware design issue, in last case it may also be hard to analyse the issue without the board if no one else can work on it. 

>Is there date you plan to release code with SD support? 

No idea sorry 🤔 but will do for sure

iz3man, 2019-05-27:
> * That said, I am currently working on new SD library implementation and will review this part to see if I can remove the limitation, thank you for reminding me

Hi Luc
received my panucatt board today. What a nice piece of hardware and software. Easy to install in 2min, up and running in one more minute. Working GREAT!!! THANKS!
Now the 8.3 limitation becomes more evident. It's a PITA to use with Marlin - and i tried switching to MK4Duo, but it doesn't work for me, and support community is Italian only.
Any plans on when/if you're gonna implement it?
And did i mention that it's a great product? :)

luc-github, 2019-05-27:
I am currently working on ESP3D 2.2 FW for Panucatt and I will add that - I can fix the upload easily on WebUI - but the listing will still be in Marlin output - for this I need just little bit more time in FW - and 2.2 may need an intermediate FW with notifications feature in 2.2 the FW is too big - need new partitionning - so I need to test update with a small FW changing partition from 1M/3M to 2M/2M then update with 2.2 using another update - and I need to discuss with panucatt if it is ok

iz3man, 2019-05-27:
Great to hear. Output like it is right now is Ok'ish as a work around, but very limiting in normal operation.

[image]

luc-github, 2019-05-27:
ping me on gitter - I will give you a test version of WebUI so you can try

luc-github, 2019-05-27:
https://gitter.im/luc-github/ESP3D?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge&utm_content=badge

luc-github, 2019-06-03:
https://github.com/sahlberg/libsmb2 could be an alternative to Webdav

A13xund3r, 2019-06-12:
> https://github.com/sahlberg/libsmb2 could be an alternative to Webdav

Not really, this is client, I think we are after server.

luc-github, 2019-06-12:
You are correct I have read too fast...

luc-github, 2019-11-01:
FTP server is no implemented and under windows is can be accessible in explorer
[image]
also FTP transfer speed are 3X higher than web speed so no meaning to go for a webdav server
Issue can be closed as implemented with FTP server

closed 2019-11-01 by luc-github