luc-github/ESP3D #321: SD upload not working

mat2k, 2019-04-27:
I've been seeing the related issues, but none are exactly the same as my problem. I'm using a homemade ESP3D (ESP12E module, an LD1117 3.3 for power, RX and TX connected directly to the Ramps1.4 Arduino Mega [since supposedly the esp is gpio 5v tolerant, and a resistive divider messed up my rx line anyways]. Running Marlin 1.1.9).
I have no garbage on the line, and am able to send any command to the printer just fine. But when I want to upload a file, this happens: (Copied the 4 lines of example gcode from another issue)

luc-github, 2019-04-27:
What version of ESP3D ?
What version of Marlin ?
Never enable Debug when doing upload. Disable it. It generate output if on Serial and overload if

luc-github, 2019-04-27:
The number of bed and extuders are defined in preferences of webUI

luc-github, 2019-04-27:
What is M20 or M115 command output in webui ?

mat2k, 2019-04-27:
> What version of ESP3D ?
2.1, happened on 2.0 too
> What version of Marlin ?
it's right there, 1.1.9

> Never enable Debug when doing upload. Disable it. It generate output if on Serial and overload if tcp or SPIFFS
Okay, I'll try that right now

mat2k, 2019-04-27:
`M20
Begin file list
XYZCAL.GCO 0
End file list
M115
FIRMWARE_NAME:Marlin 1.1.9 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:1
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:0`

luc-github, 2019-04-27:
ok if webUI see these M20 and M115 commands it means communication between ESP and printer is OK

it is Marlin 1.1.9 release or the latest 1.1X git ?

mat2k, 2019-04-27:
1.1.9 release

luc-github, 2019-04-27:
I cannot reproduce on my side with same configuration

can you try to do a manual upload:
enter in webui terminal the following lines one by one (just copy and paste to avoid typo difference)
```
M110 N0
N1 M28 hello.g*19
N2 M117 hi there*73
N3 M300 P400 S500*33
N4 M29*28
```
it should give you in output :
```
M110 N0
N1 M28 hello.g*19
N2 M117 hi there*73
N3 M300 P400 S500*33
N4 M29*28
Done saving file.
```

mat2k, 2019-04-27:
As soon as I sent the M117 the printer restarted...

```
M110 N0
N1 M28 hello.g*19
echo:Now fresh file: hello.g
Writing to file: hello.g
N2 M117 hi there*73
start
echo:Marlin 1.1.9
echo: Last Updated: 2018-08-01 | Author: (fla)
echo:Compiled: Apr 26 2019
echo: Free Memory: 2661  PlannerBufferBytes: 1664
echo:SD card ok
N3 M300 P400 S500*33
Error:Line Number is not Last Line Number+1, Last Line: 0
Resend: 1
```

luc-github, 2019-04-27:
well there is something wrong - but it does not come from ESP3D, are you using the webUI terminal or some telnet terminal ?

mat2k, 2019-04-27:
WebUI terminal. Do you want me to update to Marlin 2.0, see if it does anything? I've been planning to do it for quite some time anyways

luc-github, 2019-04-27:
I do not think it is Marlin issue - I did test on 1.1.9 before sending you the commands
these commands do not make Marlin to restart - so I suspect some hardware issue instead of FW - issue with SD card access or ESP board power issue - I cannot say 

what version of WebUI do you use

But if you plan to update Marlin take oportunity to also test the commands I sent you by serial 
and about no info in configuration be sure M503 has output, I can be disabled in Configuration.h

mat2k, 2019-04-27:
I don't think it's the ESP at all since it never reboots, in fact it hasn't rebooted since i turned it on about a couple hours ago. I'll see if I can get any debug info from Marlin

I'll also try to upload a file via USB (disconnecting the esp of course) since it should be exactly the same than the esp.

luc-github, 2019-04-27:
If it is not ESP which restart Marlin when restarting then it is definitly on printer board / SD / Marlin
Good luck - keep updated as it is really weird

mat2k, 2019-04-27:
Lol, it seems like I fixed it already... It was the SD card, even though it works flawlessly everywhere else. 8GB full-size SD card, class 4, formatted to fat32. I have no idea why Marlin rebooted with it. I changed to a 8gb microSD with an adapter, sent your test commands as you said, and the file appears perfectly and does what it should (displays hello there and beeps).

What's even weirder is that this SD card also didn't work before, because I have already tried this. Maybe my SD card slot is slightly borked, who knows. Anyhow awesome work on the ESP3D project, it seems awesome to me that you only need an ESP and a 3.3v LDO to add wifi to a printer...

closed 2019-04-27 by mat2k

luc-github, 2019-04-27:
Good to hear you solved 

SD card for printer work better when formated using https://www.sdcard.org/downloads/formatter/
not PC format

mat2k, 2019-04-27:
Well, this last one was formatted from my Android phone actually, so I guess that also works

Now I have a couple of questions/suggestions (largely unrelated) but i believe it would be better to ask them here than open a new issue:

On the WebUI the absolute maximum feedrate for XY seems to be 9999 mm/min and 999 for Z. My printer is a common Prusa clone and it does literally double the feedrates - I can get abt 20000 mm/min on X and Y, and 2500 on Z.  I think I've been able to use higher values from the preferences, but wouldn't it be better not to have such low limits?

Also, is there any update on the shared SD card for fast transfers? I've seen that and it looks very promising, but I haven't found any mention of it since 2 years ago.

And lastly, maybe it'd be even better to ask on the initial setup how many extruders and bed you have.

luc-github, 2019-04-27:
well these limits are originaly for my printer and to fit UI,  20000 mm/min = 333mm/sec I would not be surprised for Laser, but for 3D printer !!! what filament support this speed ?
for Z yes this can be increased I think,But actually I will add  min / max in preferences in future to not having  to handle it for all feedrate in coming versions as now it is used by GRBL_ESP32 which need higher boundaries for laser

Yes wizard could be completed with these questions 

About Shared SD card  for faster transfer, it is not under my control, but Roy from http://www.panucatt.com told me some boards should be delivered to him this week so I think they will be available soon

mat2k, 2019-04-27:
I meant it as jog speeds - What I mean is that you can't set those speeds on the jog buttons' feedrate boxes, of course I don't print at those speeds lol

also nice to know there's been progress on the sd card system. i guess you need to use something like multiplexers to select the esp or arduino?

luc-github, 2019-04-27:
>I meant it as jog speeds 

sorry my mistake 😊 

Sd card support is done already since 1 year for http://www.panucatt.com/azteeg_X5_mini_reprap_3d_printer_controller_p/ax5mini.htm, but code is not public yet