luc-github/ESP3D #323: ESP3D hangs at file upload

TheMM94, 2019-05-08:
We are currently using ESP3D 2.1 on the ESP32-Wroom together with Marlin 2.0.

We have the issue that ESP3D is hanging if we upload a G-Code file.
The upload starts and suddenly in middle of the File the ESP hangs and we need to do a hard reset.

The printer is connected to Serial 2 of the ESP32. Attached is the Debug output on Serial 0 (to get this working we edited the config.cpp Line 174 to `#if defined(USE_SERIAL_0) || (defined( DEBUG_ESP3D) && defined(DEBUG_OUTPUT_SERIAL)).` 

Any idea what the Problem is?
[ESP32.log](https://github.com/luc-github/ESP3D/files/3157917/ESP32.log)

luc-github, 2019-05-08:
Did. You tried with Serial to see if issue is not linked to serial 2 ? Nothing special popup from log

TheMM94, 2019-05-09:
We are planning to connect Serial 0 permanently to a USB/UART Chip and Serial 2 would be the connection to the Printer uC. 

I turned off debugging and tried to use Serial 0, same issue. What I forget to mention before, we use the asyncwebserver. I now also tried to use the syncwebserver and everything works (with Serial 0 or Serial 2). I also found the closed issue #308 . 

Could this basically be a Problem with the asyncwebserver? As I understood your comment in #308 you would suggest to use the syncwebserver, if we have stability issues?

luc-github, 2019-05-09:
Ho yes. This is key point
async is really not stable please switch to sync and should be ok
Async has currently several issues.Main dev is really busy and does not has time to maintain it. When async could be ok in most project. for esp3d it is not reliable enough.

luc-github, 2019-05-09:
In 3.0 I have put in hold async because of these issues

luc-github, 2019-05-09:
May I know what is your project ? I am curious ^_^

TheMM94, 2019-05-09:
Okay, Thanks for your help!

Of course, you can 😉. We make a 3D Printer as a school project. Development is currently still in progress. The current schematic is here: https://github.com/mnemocron/FHNW-Pro4E-FS19T8-3DPrinterBoard-STM32

Bay the way, an update of the README.md would be nice. “suggested for ESP32” sounds for me not like that it has several issues.

luc-github, 2019-05-09:
Nice. 
Ok will add some recommendations in readme

luc-github, 2019-05-09:
Readme updated. So I guess we can close issue

closed 2019-05-09 by luc-github