luc-github/ESP3D #326: Why 8.3 file name limitation ?

luc-github, 2019-05-09:
Please do not complain to me about this.
# This is not ESP3D limitation but current Marlin limitation
which does not support long filenames in commands, and it is not up to ESP3D to work around this limitation.

I will be more than happy to remove the  check when this limitation disapear. Until then please do not ask me to handle it.

closed 2019-05-09 by luc-github

Fighter777, 2020-01-03:
Bonjour,

Des nouvelles de ce problème ?
Je commence à réfléchir pour modifier Marlin afin de virer cette restriction codé en dur dans le code ^^.

J'ai une carte 32bits et je trouve cela dommage de ne pas pouvoir utiliser pleinement ESP3D :S

luc-github, 2020-01-03:
bha le probleme c'est Marlin pas ESP3D c'est ecrit en gras pourtant ^_^

Fighter777, 2020-01-03:
j'ai jamais dis que ESP3D était le problème ^^, j'ai trifouillé un peu les fichiers SD de marlin pour dégager cette limitation, va falloir que je refasse un peu de langage C, je suis un poil rouillé lol

ça serais quoi le nombre de caractères appréciable ?

luc-github, 2020-01-04:
Pur moi le mieux serait d faire comme MarlinKimbra, d'utilise SDFAT qui support les noms longs
https://github.com/MKFirmware/MK4duo/tree/master/MK4duo/src/sdcard

SDFAT V2 : https://github.com/greiman/SdFat-beta

Fighter777, 2020-01-04:
c'est quoi cette version Kimbra ? je l'ai vu dans l'interface de ESP3D mais j'ai pas vraiment cherché ce que c'était ^^

luc-github, 2020-01-04:
C'est une version italienne orientee `utilisateur` elle est la version de Marlin qui supportait le 32bit (DUE) il y a des annees,  supporte les noms long, les ecrans Nextions, etc.., et elle est stable ;). 
Elle etait ma reference justement a cause du support des noms et de la stabilitee

Je n'utilise pas vraiment Marlin donc je n'ai pas forcement beaucoup de recule, la j'y suis un peu plus dedans car je travaille sur l'integration de ESP3D dans la version ESP32 de Marlin (mais il y a toujours la limitation 32 char..., meme si ESP32 a une lib qui supporte les noms long, et meme supporte SDFat https://github.com/luc-github/ESP3D/tree/3.0/esp3d/src/modules/filesystem/sd) 

Deplus perso je prefere la stabilitee de Repetier meme si la c'est un peu a la traine, car c'est vrai que Marlin a plein de fonctionnalitees interressantes et surtout supporte plus de platformes, mais du coup il y a un soucis de stabilite et des problemes de regression entre les releases du au `succes` e l'implementation des nouvelles plateformes et fonctionnalitees.
Bref pas de monde parfait ^_^

Fighter777, 2020-01-04:
ben pour la version ESP32, ça a l'air de marcher correctement, le flash et l'intégration sur le port série d'une SKR PRO fonctionne (avec une version de marlin 2.0 modifié pour gérer la carte SD)

j'ai aussi branché l'ESP32 sur l'écran tactile TFT35 de bigtreetech, ça marche également, par contre j'ai pas les fichiers de la SD de l'écran, la commande M20 renvoi que ceux de la carte mère.

Donc si tu a besoin de tester ça sur un ESP32, y'a moyen de s'arranger :)

luc-github, 2020-01-04:
La version ESP32 de Marlin - c'est Marlin installe sur une ESP32 
La version ESP32 de ESP3D c'est en principe la meme que pour ESP8266 mais avec plus de fonctionnalitees surtout dans la version 3.0 en devt

Oui c'est sur ma liste :  support des TFT et les cartes BTT qui deviennent tres populaires.
 la en l'etat c'est basic  pas de support du flash disk ni de la sd embarquee - il faut que je regarde le code du tft mais ca doit etre jouable j'ai une petite idee  de comment faire
J'ai vu dans leur videos qu'il y a le support de ESP3D sur SKR et TFT ca m'a fait sourire de voir mon petit projet etre un claim marketing ^_^

Je pense qu'une carte comme https://www.panucatt.com/ProductDetails.asp?ProductCode=WB8266 serait plus adapte qu'une carte ESP de base, l'upload serial c'est quand meme la misere 😭
Il faut que je regarde ca aussi 

Merci pour la proposition je te contacterai pour tester le support UDISK et SC carte embarquee avant de pousser une PR chez BTT

luc-github, 2020-08-28:
additionally usinc M33 is just cosmetic and not functional - all print commands need short names, upload need short name, so long name cannot be used even less when doing upload.

M33 need to be done for each file name gave by M20, which means that increase the communication exchange and slow down the response time to display files which is IMHO not necessary - because after the complain will be: why display is so slow ?