luc-github/ESP3D #327: Marlin 2.0 Binary File Transfer?

mat2k, 2019-05-11:
I just stumbled upon this while updating:

// Add an optimized binary file transfer mode, initiated with 'M28 B1'
//#define BINARY_FILE_TRANSFER

I honestly don't know anything about this, except that it may be much better than what we currently have. worth a look

luc-github, 2019-05-11:
per https://github.com/MarlinFirmware/Marlin/pull/12249
it is still experimental, only for Marlin 2.X, still not documented and still not stable 
It is designed to transfer FW not GCODE so not sure about performances for GCODE.

Sorry I have currently no bandwidth to spend time on experimental  specific Firmware feature.

But feel free to have a look and report

mat2k, 2019-05-11:
> ...and due to less overhead and constant maximum packet size it improves transfer rate of gcode files.

it's fine, i'm just saying that this might be an alternative to having a shared SDcard if it improves speed significantly. mostly just a heads-up.

luc-github, 2019-05-11:
>...and due to less overhead and constant maximum packet size it improves transfer rate of gcode files.

yes I saw this, but it does not give any figures so it is hard to confirm unless to do benchmark
and for the moment it looks not stable so again hard to confirm

luc-github, 2019-05-12:
I think an easy way to test would be to modify the file transfer.py to use telnet instead of serial 
That would give I rought idea of the speed. 
ESP3D doing telnet /serial bridge it will work out of the box in theory

Andrei-Pozolotin, 2019-05-14:
@p3p Chris:
I am curious if apart from this zip there is  an "official protocol documentation"?
https://github.com/MarlinFirmware/Marlin/files/2532378/transfer.py.zip

p3p, 2019-05-14:
There is no official documentation, it was a POC PR that got merged really, I would like to tweak the protocol a bit make error recovery more reliable, and a few other things, before it becomes an official feature.

Currently its main use case is transferring firmware files to the sd cards on 32bit boards, although because of the fixed packet size (always the maximum your Marlin build supports), rather than transferring gcode line by line it can perform a fair bit better than a normal ascii transfer, especially with large command buffer (determines the packet size).

luc-github, 2019-05-22:
I close issue as mentioned https://github.com/luc-github/ESP3D/issues/327#issuecomment-491550877 I do not have bandwidth for this with current status - feel free to re open when there are some update

closed 2019-05-22 by luc-github

ModMike, 2020-06-05:
Has there been any developments on this? The ESP3D-WebUI is fantastic but the file transfer is basically unusable.

luc-github, 2020-06-05:
https://github.com/MarlinFirmware/Marlin/pull/14817