luc-github/ESP3D #329: FR: Add delay at startup before connecting to WiFI

axl12ap, 2019-05-16:
Hello,
I got an SKR 1.3 board (LPC1768 based with bootloader) from BigTreeTech and I attached an ESP wifi adapter with ESP3D firmware.

On the SKR there's the Marlin 2.0.x firmware and it works properly, but before firmware start bootloader takes about 5seconds to check firmware update from onboard SD card.

There's a side effect on the display, because ESP3D connect on WiFI before 5 seconds, so the messages: "Connecting...." and "Connected xxx.xxx.xxx.xxx" doesn't appears correctly on it.

If you can put inside ESP3D a delay (5 seconds would be ok) should be fine.... may be parametrizable trough the configuration panel 

Thank so much for you work

luc-github, 2019-05-16:
there is already a delay modifiable in code, in 3.0 it will be in settings - what version of ESP3D do you use ?

axl12ap, 2019-05-16:
Currently I'm using the version 2.1 of ESP3D... does still available the 3.0 version?? 

Can I flash it on my ESP-12E module??

luc-github, 2019-05-16:
3.0 is still in devt and does not have webUI yet
in 2.1 : https://github.com/luc-github/ESP3D/blob/2.1/esp3d/esp3d.ino#L29
the boot delay is already set to 8s
https://github.com/luc-github/ESP3D/blob/2.1/esp3d/esp3d.h#L33

try to add more delay like 15s : `myesp3d.begin(15000);`

axl12ap, 2019-05-16:
> 3.0 is still in devt and does not have webUI yet
> in 2.1 : https://github.com/luc-github/ESP3D/blob/2.1/esp3d/esp3d.ino#L29
> the boot delay is already set to 8s
> https://github.com/luc-github/ESP3D/blob/2.1/esp3d/esp3d.h#L33
> 
> try to add more delay like 15s : `myesp3d.begin(15000);`

Yes, I see your source code in the "begin" member there're 2 optional parameters.

I can try to enable the "recoverytimedelayms" also to add some time over normal "startdelayms"...

luc-github, 2019-05-16:
why?

axl12ap, 2019-05-16:
> why?
> startdelayms allows 65536 ms = more than 65s delay, more than 1 min - is it not enough ??

That's true... because in both cases I need to reflash the ESP module (I think RECOVERY can be enabled in the UI configuration tab).

O.T.: About baud rates, why do you put inside fw the value 921000 and not 1000000 as in Marlin firmware? I'd like to test the communication at maximum speed, should be compatible both values?? Or I need to slow down 500000??

luc-github, 2019-05-16:
I forget where I got this originaly, but this value is the max supported by ESP8266, I just found an old comment talking about it : https://github.com/luc-github/ESP3D/issues/93#issuecomment-407935405

axl12ap, 2019-05-16:
Thanks, I'll close FR issue

closed 2019-05-16 by axl12ap

spock64, 2019-08-09:
@axl12ap I'm trying to diagnose the core problem here - could you tell me what ESP8266 hardware module you are using? Also, how is the module connected to the bigtreetech board?

In my case I'm using an ESP-01 and am hitting problems that I think might be power related.

dp250f, 2019-08-16:
I'm using an ESP-12E, running the latest ESP3D (2.1 b33) on an SKR v1.3 (32-bit Marlin 2.0) and I've been playing with **startdelayms**.  I noticed it causes the **printer** to delay its startup by however long I set it, and on cold boot, it always sticks at **Connecting..** until the next M117 is sent from ESP. Not sure if this helps or not, but it's something I noticed.