luc-github/ESP3D #330: Sbase Smothie 3 problems

fedonkin, 2019-05-16:
Smoothie firmware, ESP3D 2.1 .You can not see the temperature of the extruder table. No config visible.  Very long delay. Press move x axis 10 mm, after 3 seconds the printer moves x axis 10 mm. Something need to change in the firmware smoothie? or how to fix it all?
[image]
[image]
[image]
[image]
[image]

luc-github, 2019-05-17:
looks like serial communication problem as information from ESP seems Ok.
Can you share  the ESP3D info tab?
Also can you share the wiring you use ?
which esp board you use ?

luc-github, 2019-05-22:
I close issue if no update

closed 2019-05-22 by luc-github