luc-github/ESP3D #332: Cheap esp12f based serial2wifi module

fape, 2019-06-01:
I managed to flash ESP3D to cheap (~$2)  ESP-12F based serial wifi module. [Here you can find my tutorial](https://gist.github.com/fape/17dd5add66b635d7f29f1d63ce246c46).

I know this is not a "real issue". Basically I do not know how and where to share my tutorial(is). I did not want to edit the wiki with out permission. But feel free to get my markdown and use it as wiki page ;)

luc-github, 2019-06-01:
Please feel free, Wiki is open to everyone - every contribution is welcome - you can even do small page which link to yours if you want.
I prefer the ones who use the proposal, do the wiki as they know better than me what they did to make it work ^_^

fape, 2019-06-08:
Thanks. I added to wiki: https://github.com/luc-github/ESP3D/wiki/Cheap-ESP-12F-based-serial-wifi-module-with-ESP3d

closed 2019-06-08 by fape

luc-github, 2019-06-08:
@fape Thanks  a lot 
I have changed page title to  https://github.com/luc-github/ESP3D/wiki/Cheap-ESP-12F-based-serial-wifi-module to make it shorter and moved the menu to root, as you describe there are already levelshifters, it could be confusing to link with other page suggestiong adding resistors or levelconverters

Thank you again for your contribution ^_^

fape, 2019-06-08:
It was my pleasure. Thanks.