luc-github/ESP3D #333: SoftwareSerial

LL1r1k, 2019-06-02:
Can i use SoftwareSerial for my NodeMCU? When i tried to connect my power off printer to NodeMCU and power on ESP, it doesn't start(Led not flashing), but when i press reset button all fine. Maybe if i change my GPIO0 and GPIO3 to other pins something will change.

luc-github, 2019-06-02:
Sorry - Software serial is not supported in ESP3D and for correct performance, better use hardware serial
You may have power supply issue if such behavior, how do you power your nodeMCU ?

LL1r1k, 2019-06-02:
I connected the NodeMCU through the power supply and i have relay connected to i which turn on printer. First, I turn on the board, and then switch the relay through the web interface and turn on the printer. In teory) But when i turn on NodeMCU, it doesn't turn on(LED does not flash), if press RESET button everything works fine.

LL1r1k, 2019-06-02:
NodeMCU connected to printer: TX Node -> RX MKS gen L ; RX Node -> TX MKS gen L ; GND ->GND

luc-github, 2019-06-02:
does the NodeMCU start properly if you connect by USB ?

LL1r1k, 2019-06-02:
doesn't matter

LL1r1k, 2019-06-02:
now, I tried to connect the board to the Arduino Mega turned off and it's the same.

LL1r1k, 2019-06-02:
Only starts working normally after pressing the restart button.

LL1r1k, 2019-06-02:
And of course, when the printer is already turned on, the NodeMCU may start up normally.

luc-github, 2019-06-02:
Serial is GPIO1 /GPIO3 not GPIO0/GPIO3

LL1r1k, 2019-06-02:
RX TX on NodeMCU,I have written it wrong.

luc-github, 2019-06-02:
Why do you think is serial pins the problem ?
Does your MCU start if not serial connected ?

LL1r1k, 2019-06-02:
Yes, MSU starts if nothing is connected and starts when connected to a working printer.

luc-github, 2019-06-02:
you power 5v or 3.3v ?

LL1r1k, 2019-06-02:
NodeMCU can be powered by 5v.

luc-github, 2019-06-02:
well it depend how you connect the 5V - you put in VCC, right ?

LL1r1k, 2019-06-02:
I connect VIN and GND to power supply

luc-github, 2019-06-02:
Well I am not great on electronic - so I have no idea - may be 5V tolerance issue on rx/tx ?
There is no difference from FW point of view between hard start and reset so issue is hardware for sure,  and FW even did not boot per your feedback, I doubt changing serial pins will do something, really looks like power issue, may be adding some capacitor ? 
I guess you have same issue what ever the FW - better ask on electronician gurus of ESP8266

LL1r1k, 2019-06-02:
Ok, thank you anyway.

closed 2019-06-06 by luc-github