luc-github/ESP3D #334: can i ad wifi to Anet A8

CsepregiArtur, 2019-06-04:
Hello

I want to ask if it is possible to add this wifi module to Anet A1284-Base V1.7 Control Board Mother Board Mainboard for Anet A8.
I found a good description here http://lokspace.eu/anet-a8-wifi-mod/
but I want to use esp3d with a Node MCU or D1 mini. 
Do you think it is necessary to unsolder R52 and R53 resistors or?

Thank you

luc-github, 2019-06-04:
I cannot answer as I do not have any anet printer/board - but links you mention explain that the addon card does not need to unsolder component with some risks. so may be ? 
Better ask to electrocian gurus or on linked article like you did.
 
I will let question open if anyone can answer you - feel free to feedback if you got update

rhialto56, 2019-06-04:
I do it without pb.  Juste soldier rx and tx wire on the board anet v1.7 just near the chip atmega

luc-github, 2019-06-05:
@CsepregiArtur @rhia lto56 feel free to add a page in wiki with your board and wiring ^_^

rhialto56, 2019-06-05:
Ok. I just sold two wires from number 9 and number 11 to connector j3. 
then I connected to the j3 connector as in the tutorials.
do not forget in the console esp3d to disable the M117 code otherwise retrieves error messages on the LCD . That's all
[image]

CsepregiArtur, 2019-06-06:
super thank you I will try also and I will come back with photos am will
enter it in wiki

rhialto56, 2019-06-06:
with a better picture
Good luck
[image]

CsepregiArtur, 2019-06-18:
I'm trying to flas marlin to my printer,  and with after that I will try to
connect it. Sorry for it taking so long,  i'm  stuckt to burn the
bootloader

CsepregiArtur, 2019-06-19:
I'm trying to flas marlin to my printer,  and with after that I will try to
connect it. Sorry for it taking so long,  i'm  stuck to burn the
bootloader. Maybe today I will find something.

CsepregiArtur, 2019-06-20:
hello

Please close this topic. my anet board is officially blown.

luc-github, 2019-06-24:
Sorry to read this😐

closed 2019-06-24 by luc-github

gello0, 2020-04-16:
I managed to solder wires on the 1.7 board. Would be nice if this is added in anet a8 wiki :)

luc-github, 2020-04-16:
@gello0 wiki is open to everyone , addition are always welcome, let me know if you have issue to add things

luc-github, 2020-04-16:
@gello0 thank you for your contribution - you can use this https://github.com/luc-github/ESP3D/issues/427 to store pictures