luc-github/ESP3D #335: "Enable lock interface" option

axl12ap, 2019-06-05:
Hello,
in the preferences tab there's an option called as in thread title "Enable lock interface".

Can someone how does it works when I enabled it?

Thanks

luc-github, 2019-06-05:
when enable no command to printer can be sent until you disable it using check box on top of window, interface is locked.

axl12ap, 2019-06-05:
> when enable no command to printer can be sent until you disable it using check box on top of window, interface is locked.
> 
> Only ESP commands are allowed but update and restart because on some board restart serial reset the printer board.

Thanks for explaining.. so can't set any password for webserver access security?

luc-github, 2019-06-05:
Did you enabled authentication ?

axl12ap, 2019-06-05:
> Did you enabled authentication ?

I don't know what you mean... please explain to me

luc-github, 2019-06-05:
you don't understand `authentication` ? to have password enabled security

axl12ap, 2019-06-05:
> you don't understand `authentication` ? to have password enabled security

Yes I understand what it mean, but I don't know where to set the password and username.

My goal should be to protect the webserver access with username and password...

luc-github, 2019-06-05:
In readme there is
 >Authentication for sensitive pages, here to enable/disable AUTHENTICATION_FEATURE  

 the flag to be modified in config.h

axl12ap, 2019-06-06:
> In readme there is
> 
> > Authentication for sensitive pages, here to enable/disable AUTHENTICATION_FEATURE
> 
> the flag to be modified in config.h
> 
> Which means : You must have authentication enabled to be able to modify the passwords: admin and user
> 
> FYR login is showed in this video : https://youtu.be/7vtWNn9jyDs?t=110

Thanks for your information, now it's clear :) 

I find also below in config.h the code about user/pass default values:

`const char DEFAULT_ADMIN_PWD []  PROGMEM =	"admin";`
`const char DEFAULT_USER_PWD []  PROGMEM =	"user";`
`const char DEFAULT_ADMIN_LOGIN []  PROGMEM =	"admin";`
`const char DEFAULT_USER_LOGIN []  PROGMEM =	"user";`

closed 2019-06-06 by luc-github