luc-github/ESP3D #337: Connection to RAMPS 1.4[BUG]

walterbux, 2019-06-12:
Hello,
I have followed your wiki to connect an AZ Delivery ESP-32 Dev Kit C
I have successfully flashed the ESP3D-2.1, configured the web server by AP and then connected the device to my network with a fixed address.

 **ESP3D Firmware:**
- Version:[2.1]
- Wifi mode:[STA]

**Target Firmware:**
 - Name: [Repetier]
 - Version [1.0.X]

**Board used (please complete the following information):**
 - MCU: [ESP32] 
 - Name:[AZ Delivery ESP-32 Dev Kit C] 
 - Flash size: [4M]

**Additional context**
I connect the ESP 32 to RAMPS 1.4 AUX 1 pins (+5, GND, TX, RX) respectively to pins V5, GND, TX0, RX0).

No communication with the printer.

By connecting the ESP32 to pc via USB only I see the web server interface corrrectly and through the arduino ide serial monitor I see the commands flowing to the U0 serial interface.

Do I need some 5 to 3,3V level adaptation circuit for the serial interface ?

Thanks a lot.

Best regards.

Walter Bux

luc-github, 2019-06-12:
So far communication is not 5v tolerant - on some board it may work but it is not recommended

https://lastminuteengineers.com/esp32-arduino-ide-tutorial/
>The ESP32 requires a 3.3V power supply and 3.3V logic levels for communication. The GPIO pins are >not 5V-tolerant! If you want to interface the board with 5V (or higher) components, you’ll need to do >some level shifting.

luc-github, 2019-06-12:
additionally be sure your Mega/ramps is not connect by USB

walterbux, 2019-06-12:
I have done the same connection type with a AZ-Delivery NodeMCU Lolin V3 and everything works flawlessly..
ok no connection to RAMPS via USB.
So what type of adaptation I need to connect the AZ Delivery ESP-32 Dev Kit C to RAMPS ?
Is there a circuit you advice to use ?
Have I to think I have already destroyed the module ?
Thanks.

Walter

luc-github, 2019-06-12:
As I said is depending of board.
I suggest you to connect serial to USB adapter (3.3v of course)  and then to arduino serial monitor to check serial is working as expected.

For possible circuit you can use the ESP12F wiki connection proposal : https://github.com/luc-github/ESP3D/wiki/ESP8266-12E-F

walterbux, 2019-06-16:
Yes, thanks, everything solved by adding a 680 to 1K ohm partition on the TX side of Ramps (D1 pin of AUX 1) as advised here: https://reprap.org/wiki/RAMPS_1.4#Connect_via_Bluetooth.
Unfortunately this has required to build a bad partitioning cable ... but it helps :).
I hate 5 to 3,3 V adaptation as much as I can :) :)

Walter

luc-github, 2019-06-16:
This is almost same as described  in wiki I linked above.
Happy you solved your issue

closed 2019-06-16 by luc-github