luc-github/ESP3D #338: wemos d1 mini wiring question

damienj210, 2019-06-16:
what's the purpose of the zener between rx and ground? if you dont have it will you have issues?

luc-github, 2019-06-16:
This wiki was done by @mauro - I let him answer

damienj210, 2019-06-16:
that's why I was confused, the wemos is based on 5 volts. is the TX and RX still 3.3?

luc-github, 2019-06-16:
ESP is 3.3v - wemos has voltage regulator for power but all GPIO are 3.3v and not 5V tolerant I think

closed 2019-06-17 by damienj210

firepower9966, 2019-08-30:
ESP8266 is 5V tolerant I/O I using D1 mini TX and RX directly connected to atmega TX  and RX.

luc-github, 2019-08-30:
https://www.espressif.com/en/support/explore/faq?keys=Are%20the%20GPIO%20pins%205V%20compatible%3F

[image]
ESP8266  is not 5v tolerant - for the boards it depend

firepower9966, 2019-09-01:
https://www.ba0sh1.com/blog/2016/08/03/is-esp8266-io-really-5v-tolerant/

luc-github, 2019-09-01:
this is a third party blog - I shared what espressif  published - now everyone is free to do what he wants