luc-github/ESP3D #340: [BUG] Typo in README.md: points ArduinoWebSockets, but is WebSockets

sl1pkn07, 2019-06-20:
**Describe the bug**
Typo in the README.md

points to ArduinoWebSockets, but is WebSockets

because with ArduinoWebSockets installed:

~~~
In file included from /tmp/arduino_build_196990/sketch/esp3d.cpp:85:0:
syncwebserver.h:25:30: error: WebSocketsServer.h: No such file or directory
 #include <WebSocketsServer.h>
                              ^
compilation terminated.
Usando librería ESP8266WiFi con versión 1.0 en la carpeta: /home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi 
Usando librería ESP8266mDNS con versión 1.2 en la carpeta: /home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266mDNS 
Usando librería ESP8266WebServer con versión 1.0 en la carpeta: /home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer 
Usando librería EEPROM con versión 1.0 en la carpeta: /home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/EEPROM 
Usando librería DNSServer con versión 1.1.1 en la carpeta: /home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/DNSServer 
exit status 1
WebSocketsServer.h: No such file or directory
~~~

commit used: 1d7b3ff83f72d5bc5cc8bc0a525ceda4f2333c02

sl1pkn07, 2019-06-20:
Seems Arduino IDE library manager "ArduinoWebsockers" @ links2004 (Markus Sattler) is named as "WebSockets" and "ArduinoWebSockets" @ gilmaimon (Gil Maimon) is named as "ArduinoWebSockets"

luc-github, 2019-06-20:
you did not copied https://github.com/luc-github/ESP3D/tree/2.1/libraries/arduinoWebSockets ?

sl1pkn07, 2019-06-20:
seems my Arduino IDE (1.8.9) linux not used that library and try to find external ones

luc-github, 2019-06-20:
all libraries are in my repo in directory libraries because version are frozen
I need to emphasis that and update readme

luc-github, 2019-06-20:
[image]

sl1pkn07, 2019-06-20:
yep. i see it in my local clone repository, but my IDE can't take it

luc-github, 2019-06-20:
yes - I will emphasis in readme

sl1pkn07, 2019-06-20:
ok, i copied it, but the IDE say me need update some libraries

[image]

i need keep it?

luc-github, 2019-06-20:
yes

luc-github, 2019-06-20:
I cannot test all update for all libraries - so I freeze version

closed 2019-06-20 by luc-github