luc-github/ESP3D #342: [BUG] invalid when set station password to empty

sl1pkn07, 2019-06-20:
**Describe the bug**
I try to add my ESP-01 dongle (with serial-ESP breakout pcb) to my newwork

my net is open, without auth, but with mac filter

set the ESP-01 to STA, without auth (open), and set my network SSID, refuse to connect

type [ESP410] to get the MAC address, and [ESP400]/[ESP401] to get/set the options

~~~
{"chip_id":"10766373",
"cpu":"160",
"freemem":"39.48 KB",
"SDK":"2.2.1(cfd48f3)",
"flash_size":"1.00 MB",
"update_size":"423.45 KB",
"spiffs_size":"110.55 KB",
"baud_rate":"115200",
"sleep_mode":"Modem",
"channel":"11",
"phy_mode":"11g",
"web_port":"28261",
"data_port":"8888",
"active_mode":"AP (brawbrawbraw)",
"ap_ssid":"ESP3D",
"ssid_visible":"Yes",
"ssid_authentication":"None",
"ssid_max_connections":"4",
"ssid_dhcp":"Started",
"ip":"192.168.0.1",
"gw":"192.168.0.1",
"msk":"255.255.255.0",
"connected_clients":[],
"disabled_mode":"STA (brawbrawbraw) is disabled",
"captive_portal":"Enabled",
"ssdp":"Enabled",
"netbios":"Enabled",
"mdns":"Enabled",
"web_update":"Enabled",
"pin recovery":"Disabled",
"autentication":"Disabled",
"target_fw":"???",
"M117_output":"Enabled",
"Notifications":"Disabled",
"Serial_output":"Enabled",
"Websocket_output":"Enabled",
"TCP_output":"Enabled",
"fw":"2.1.0.b28 ESP8266/8586"}

{"F":"network","P":"112","T":"I","V":"115200","H":"Baud Rate","O":[{"9600":"9600"},{"19200":"19200"},{"38400":"38400"},{"57600":"57600"},{"115200":"115200"},{"230400":"230400"},{"250000":"250000"},{"500000":"500000"},{"921600 ":"921600 "}]},
{"F":"network","P":"117","T":"B","V":"2","H":"Sleep Mode","O":[{"None":"0"},{"Light":"1"},{"Modem":"2"}]},
{"F":"network","P":"121","T":"I","V":"28261","H":"Web Port","S":"65001","M":"1"},
{"F":"network","P":"125","T":"I","V":"8888","H":"Data Port","S":"65001","M":"1"},
{"F":"network","P":"130","T":"S","V":"MYESP","H":"Hostname" ,"S":"32", "M":"1"},
{"F":"network","P":"0","T":"B","V":"2","H":"Wifi mode","O":[{"AP":"1"},{"STA":"2"}]},
{"F":"network","P":"1","T":"S","V":"GheTto-WiFi","S":"32","H":"Station SSID","M":"1"},
{"F":"network","P":"34","T":"S","V":"********","S":"64","H":"Station Password","M":"0"},
{"F":"network","P":"116","T":"B","V":"3","H":"Station Network Mode","O":[{"11b":"1"},{"11g":"2"},{"11n":"3"}]},
{"F":"network","P":"99","T":"B","V":"1","H":"Station IP Mode","O":[{"DHCP":"1"},{"Static":"2"}]},
{"F":"network","P":"100","T":"A","V":"192.168.0.1","H":"Station Static IP"},
{"F":"network","P":"104","T":"A","V":"255.255.255.0","H":"Station Static Mask"},
{"F":"network","P":"108","T":"A","V":"192.168.0.1","H":"Station Static Gateway"},
{"F":"network","P":"218","T":"S","V":"ESP3D","S":"32","H":"AP SSID","M":"1"},
{"F":"network","P":"251","T":"S","V":"********","S":"64","H":"AP Password","M":"0"},
{"F":"network","P":"330","T":"B","V":"2","H":"AP Network Mode","O":[{"11b":"1"},{"11g":"2"}]},
{"F":"network","P":"120","T":"B","V":"112","H":"SSID Visible","O":[{"No":"0"},{"Yes":"1"}]},
{"F":"network","P":"118","T":"B","V":"11","H":"AP Channel","O":[{"1":"1"},{"2":"2"},{"3":"3"},{"4":"4"},{"5":"5"},{"6":"6"},{"7":"7"},{"8":"8"},{"9":"9"},{"10":"10"},{"11":"11"}]},
{"F":"network","P":"119","T":"B","V":"0","H":"Authentication","O":[{"Open":"0"},{"WPA":"2"},{"WPA2":"3"},{"WPA/WPA2":"4"}]},
{"F":"network","P":"329","T":"B","V":"2","H":"AP IP Mode","O":[{"DHCP":"1"},{"Static":"2"}]},
{"F":"network","P":"316","T":"A","V":"192.168.0.1","H":"AP Static IP"},
{"F":"network","P":"320","T":"A","V":"255.255.255.0","H":"AP Static Mask"},
{"F":"network","P":"324","T":"A","V":"192.168.0.1","H":"AP Static Gateway"},
{"F":"network","P":"168","T":"B","V":"0","H":"Notification","O":[{"None":"0"},{"Pushover":"1"},{"Email":"2"},{"Line":"3"}]},
{"F":"network","P":"332","T":"S","V":"********","S":"63","H":"Token 1","M":"0"},
{"F":"network","P":"396","T":"S","V":"********","S":"63","H":"Token 2","M":"0"},
{"F":"network","P":"855","T":"S","V":"⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮","S":"127","H":"Notifications Settings","M":"0"},
{"F":"printer","P":"461","T":"B","V":"0","H":"Target FW","O":[{"Repetier":"5"},{"Repetier for Davinci":"1"},{"Marlin":"2"},{"Marlin Kimbra":"3"},{"Smoothieware":"4"},{"Grbl":"6"},{"Unknown":"0"}]},
{"F":"printer","P":"129","T":"F","V":"0","H":"Output msg","O":[{"M117":"1"},{"Serial":"4"},{"Web Socket":"8"},{"TCP":"16"}]}
~~~
 
reset the device and always refuse to connect and back to safemode AP. i think because the options, the station password is set, then when set

`[ESP401]P=34 T=S V=` get incorrrect cmd

if try with `[ESP401]P=34 T=S V=""`, seems set `""` as password instead of empty

**Expected behavior**
connect to my network without problem

**ESP3D Firmware:**
- Version:[e.g 2.1] (1d7b3ff83f72d5bc5cc8bc0a525ceda4f2333c02)
- Wifi mode:[e.g. STA]

**Target Firmware:**
 - Name: [e.g. Marlin] Marlin if all OK)
 - Version [e.g. 2.0] GIT

**Board used (please complete the following information):**
 - MCU: [e.g. ESP32] ESP8266
 - Name:[e.g. NodeMCU2s]  -
 - Flash size: [e.g. 4M: 2M/2M] 1M

**Additional context**
Add any other context about the problem here.
Wiring, decoded stack, etc...

connected througth Arduino UNO as "FTDI" serial

luc-github, 2019-06-20:
I never tested such case sorry,
what ever have password or not - you will always get :
{"F":"network","P":"34","T":"S","V":"********","S":"64","H":"Station Password","M":"0"},
the stars are to avoid to send the info

This was in my todo list and I forget it : to connect AP 
it use `WiFi.begin (sbuf, pwd);` where pwd is empty - but for open network it should be `WiFi.begin (sbuf, NULL);`
I will fix this tomorrow, but if you want to check now you can modify the code to check if it solve your issue:

replace :
https://github.com/luc-github/ESP3D/blob/2.1/esp3d/wificonf.cpp#L470
`WiFi.begin (sbuf, pwd);`
by
```
if (pwd.length() > 0){
      WiFi.begin (sbuf, pwd);
} else {
      WiFi.begin (sbuf, NULL);
}
```

sl1pkn07, 2019-06-20:
~~~
/home/sl1pkn07/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/include -I/home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip2/include -I/home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino_build_913170/core -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DNONOSDK221=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DESP8266 -I/home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266 -I/home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/variants/generic -I/home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src -I/home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266mDNS/src -I/home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src -I/home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/EEPROM -I/home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/DNSServer/src -I/home/sl1pkn07/Arduino/libraries/arduinoWebSockets/src -I/home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266SSDP -I/home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266NetBIOS -I/home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/Hash/src /tmp/arduino_build_913170/sketch/espcom.cpp -o /tmp/arduino_build_913170/sketch/espcom.cpp.o
/tmp/arduino_build_913170/sketch/config.cpp: In static member function 'static bool CONFIG::isHostnameValid(const char*)':
/tmp/arduino_build_913170/sketch/config.cpp:323:41: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int i = 0; i < strlen (hostname); i++) {
                                         ^
/tmp/arduino_build_913170/sketch/config.cpp: In static member function 'static bool CONFIG::isSSIDValid(const char*)':
/tmp/arduino_build_913170/sketch/config.cpp:343:37: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int i = 0; i < strlen (ssid); i++) {
                                     ^
/tmp/arduino_build_913170/sketch/config.cpp: In static member function 'static bool CONFIG::isPasswordValid(const char*)':
/tmp/arduino_build_913170/sketch/config.cpp:363:41: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int i = 0; i < strlen (password); i++)
                                         ^
/tmp/arduino_build_913170/sketch/config.cpp: In static member function 'static bool CONFIG::isLocalPasswordValid(const char*)':
/tmp/arduino_build_913170/sketch/config.cpp:378:41: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int i = 0; i < strlen (password); i++) {
                                         ^
/tmp/arduino_build_913170/sketch/config.cpp: In static member function 'static bool CONFIG::isIPValid(const char*)':
/tmp/arduino_build_913170/sketch/config.cpp:403:35: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int i = 0; i < strlen (IP); i++) {
                                   ^
/tmp/arduino_build_913170/sketch/espcom.cpp:45:6: warning: unused parameter 'async' [-Wunused-parameter]
 void ESPCOM::bridge(bool async)
      ^
/tmp/arduino_build_913170/sketch/notifications_service.cpp: In member function 'bool NotificationsService::sendPushoverMSG(const char*, const char*)':
/tmp/arduino_build_913170/sketch/notifications_service.cpp:157:19: warning: 'axTLS::WiFiClientSecure::WiFiClientSecure()' is deprecated (declared at /home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureAxTLS.h:35): Upgrade to BearSSL is advised, check https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecure.h#L25-L99 [-Wdeprecated-declarations]
     TSecureClient Notificationclient;
                   ^
/tmp/arduino_build_913170/sketch/notifications_service.cpp: In member function 'bool NotificationsService::sendEmailMSG(const char*, const char*)':
/tmp/arduino_build_913170/sketch/notifications_service.cpp:190:19: warning: 'axTLS::WiFiClientSecure::WiFiClientSecure()' is deprecated (declared at /home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureAxTLS.h:35): Upgrade to BearSSL is advised, check https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecure.h#L25-L99 [-Wdeprecated-declarations]
     TSecureClient Notificationclient;
                   ^
/tmp/arduino_build_913170/sketch/notifications_service.cpp: In member function 'bool NotificationsService::sendLineMSG(const char*, const char*)':
/tmp/arduino_build_913170/sketch/notifications_service.cpp:283:19: warning: 'axTLS::WiFiClientSecure::WiFiClientSecure()' is deprecated (declared at /home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiCli [... cut ...]
Usando librería ESP8266SSDP con versión 1.0 en la carpeta: /home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266SSDP 
Usando librería ESP8266NetBIOS con versión 1.0 en la carpeta: /home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266NetBIOS 
Usando librería Hash con versión 1.0 en la carpeta: /home/sl1pkn07/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/Hash 
exit status 1
request for member 'length' in 'pwd', which is of non-class type 'char [65]'
~~~

luc-github, 2019-06-20:
I have read wrong definition sorry, in setup it is not a string:
```
if (strlen(pwd) > 0){
      WiFi.begin (sbuf, pwd);
} else {
      WiFi.begin (sbuf, NULL);
}
```

luc-github, 2019-06-21:
I have reproduced issue and solution is correct but not complete
on simple code it can connect but not on ESP3D - I check what is causing the problem as commands are the same...

luc-github, 2019-06-21:
Ok issue was due also to that an empty  password cannot be saved and so use a previous password on connection - 
checked on WebUI and using [ESP101]

please confirm it is ok on your side

luc-github, 2019-06-21:
fixed by https://github.com/luc-github/ESP3D/commit/4d6bac193a1464f3159f2c45b2309c538a6078a6

sl1pkn07, 2019-06-21:
yep. fixed and working

great work!

closing :)

closed 2019-06-21 by sl1pkn07