luc-github/ESP3D #343: [BUG] no printer config when using marlin 2.0

TheFlash12314, 2019-06-21:
**Describe the bug**
no printer configuration when using marlin 2.0

**To Reproduce**
Steps to reproduce the behavior:
1. Go to printers tab
2. Click on refresh
3. nothing except have label, value, help box

**Expected behavior**
list out full printer configuration 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ESP3D Firmware:**
- Version: 2.1
- Wifi mode:[e.g. STA]

**Target Firmware:**
 - Name: [ Marlin]
 - Version [2.0]

**Board used (please complete the following information):**
 - MCU: [esp01s and ESP32] 
 - Name:[Doit devkit v1] 
 - Flash size: [4M i think]

[image]

luc-github, 2019-06-21:
What is the output of M503  when you send to Marlin ?

TheFlash12314, 2019-06-21:
[image]

TheFlash12314, 2019-06-21:
[image]

luc-github, 2019-06-21:
ok so issue is in webui, what is your webUI version? check what display on menu

TheFlash12314, 2019-06-21:
v2.1

luc-github, 2019-06-21:
[image]

TheFlash12314, 2019-06-21:
UI: 2.1b20 / FW: 2.0.0.c23

luc-github, 2019-06-21:
this was fixed in 2.1b24 so update with latest webUI and should be Ok

TheFlash12314, 2019-06-21:
ok... will try

luc-github, 2019-06-21:
just upload the latest index.html.gz from here https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz

TheFlash12314, 2019-06-21:
i tried upload that code ... it is still UI: 2.1b20 / FW: 2.0.0.c23

luc-github, 2019-06-21:
[image]

TheFlash12314, 2019-06-21:
ok, It's working now, thanks!

closed 2019-06-21 by TheFlash12314