luc-github/ESP3D #344: [QUESTION] where i can configure the delay between power on and start the connection

sl1pkn07, 2019-06-24:
Hi

where i can add a delay between when power on the ESP unit and start the adquire the network connection?

my machine take like 10 seconds between power on the machine and finish the loading marlin. and show it in the LCD.
 
the ESP unit it takes like 7 seconds from power on the unit and finish the connection.

i want start the connection in the ESP after the machine is enterely up

is possible? where i can add this delay?

greetings

luc-github, 2019-06-24:
in begin you can set the delay https://github.com/luc-github/ESP3D/blob/2.1/esp3d/esp3d.ino#L29

by default is 8000  ms https://github.com/luc-github/ESP3D/blob/2.1/esp3d/esp3d.h#L33

sl1pkn07, 2019-06-24:
Hi

seems not works. because that delay make the main board also delayed to start. 

seems ESP send "data" over serial port making block the main board until the finish that delay, then the board start again as normal, but with the same behavoir i want avoid

luc-github, 2019-06-24:
may be your board reset when serial start - it happen on some boards
try to move serial init at the begining

on https://github.com/luc-github/ESP3D/blob/2.1/esp3d/esp3d.cpp#L101
add ` CONFIG::InitBaudrate();`
on https://github.com/luc-github/ESP3D/blob/2.1/esp3d/esp3d.cpp#L151
change to 
`if (  !CONFIG::InitExternalPorts() ) {`

sl1pkn07, 2019-06-24:
if help, the board is a ReARM

sl1pkn07, 2019-06-24:
like this?

~~~
diff --git a/esp3d/esp3d.cpp b/esp3d/esp3d.cpp
index 6281c63..cc842d5 100644
--- a/esp3d/esp3d.cpp
+++ b/esp3d/esp3d.cpp
@@ -98,6 +98,7 @@ void Esp3D::begin(uint16_t startdelayms, uint16_t recoverydelayms)
     WiFi.mode (WIFI_OFF);
     wifi_config.WiFi_on = false;
     //check EEPROM Version
+    CONFIG::InitBaudrate();
 #if defined( DEBUG_ESP3D) && defined(DEBUG_OUTPUT_SERIAL)
     CONFIG::InitBaudrate(DEFAULT_BAUD_RATE);
     delay (2000);
@@ -147,7 +148,7 @@ void Esp3D::begin(uint16_t startdelayms, uint16_t recoverydelayms)
     }
 #endif
     //check if EEPROM has value
-    if (  !CONFIG::InitBaudrate() || !CONFIG::InitExternalPorts() ) {
+    if ( !CONFIG::InitExternalPorts() ) {
         breset_config = true;  //cannot access to config settings=> reset settings
         LOG ("Error no EEPROM access\r\n")
     }

~~~

luc-github, 2019-06-24:
yes but it is just a guess

sl1pkn07, 2019-06-24:
nope, not works

only work if unpin the TX pin, power on the machine, and pin again when the mainboard is ready

luc-github, 2019-06-24:
what is your OS?

sl1pkn07, 2019-06-24:
the printer OS or my PC OS?

printer Marlin bugfix-2.0x (git) and the PC is Linux (Archlinux)

sl1pkn07, 2019-06-24:
>only work if unpin the TX pin, power on the machine, and pin again when the mainboard is ready

the machine and mainboard i mean the printer and the ReARM

luc-github, 2019-06-24:
Ok so revert the changes
according https://github.com/luc-github/ESP3D/issues/329
who use same chip as ARM you must set 15s = 15000ms to make it work

sl1pkn07, 2019-06-24:
https://github.com/luc-github/ESP3D/issues/329#issuecomment-493007386

oh, i modified the line https://github.com/luc-github/ESP3D/blob/2.1/esp3d/esp3d.h#L33 instead like say in that post

sl1pkn07, 2019-06-24:
~~~patch
diff --git a/esp3d/esp3d.cpp b/esp3d/esp3d.cpp
index 4ea5b6f..9aa1c3f 100644
--- a/esp3d/esp3d.cpp
+++ b/esp3d/esp3d.cpp
@@ -99,6 +99,7 @@ void Esp3D::begin(uint16_t startdelayms, uint16_t recoverydelayms)
     WiFi.mode (WIFI_OFF);
     wifi_config.WiFi_on = false;
     //check EEPROM Version
+    CONFIG::InitBaudrate();
 #if defined( DEBUG_ESP3D) && defined(DEBUG_OUTPUT_SERIAL)
     CONFIG::InitBaudrate(DEFAULT_BAUD_RATE);
     delay (2000);
@@ -148,7 +149,7 @@ void Esp3D::begin(uint16_t startdelayms, uint16_t recoverydelayms)
     }
 #endif
     //check if EEPROM has value
-    if (  !CONFIG::InitBaudrate() || !CONFIG::InitExternalPorts() ) {
+    if ( !CONFIG::InitExternalPorts() ) {
         breset_config = true;  //cannot access to config settings=> reset settings
         LOG ("Error no EEPROM access\r\n")
     }
diff --git a/esp3d/esp3d.ino b/esp3d/esp3d.ino
index aba3c02..c0124f1 100644
--- a/esp3d/esp3d.ino
+++ b/esp3d/esp3d.ino
@@ -26,7 +26,7 @@ Esp3D myesp3d;
 //Setup
 void setup()
 {
-    myesp3d.begin();
+    myesp3d.begin(15000);
 }
 
 //main loop
~~~

no sucess :(

luc-github, 2019-06-24:
But when boot is done can you communicate with ESP3D and does ESP3D can communicate with printer ?

sl1pkn07, 2019-06-24:
yes (i guess), but when the machine and the ESP is loaded,the LCD only show "connecting." from the ESP

luc-github, 2019-06-24:
So if it show 'connecting' why adding a delay? The problem is showing IP not delaying display

luc-github, 2019-06-24:
What baud rate is set on ESP? What baud rate on Marlin ?

sl1pkn07, 2019-06-24:
2500000 in both

sl1pkn07, 2019-06-24:
[image]

sl1pkn07, 2019-06-24:
the problem i can see, when power on the rearm and the ESP, if the ESP delay is added, makes the REARM freeze until the delay is passed, then the rearm start boot again and finish boot, but the ESP is already booted and connected to wifi, and the mailin only show the last message captured from the ESP, which is "connecting."

if i unplug the TX pin in the ESP,and power up both, (with the delay added), the the rearm boot without problem, and if plug the TX pin before the ESP delay is over, then the printer show the message commands from ESP without problem and in time

i'm not sure how fix this. maybe a hardware timer "relay to connection" (power up and wait X time before close se output) connected to pin TX?

luc-github, 2019-06-24:
try to add : ESPCOM::println (WiFi.localIP().toString().c_str(), PRINTER_PIPE);
before line https://github.com/luc-github/ESP3D/blob/2.1/esp3d/wificonf.cpp#L544

sl1pkn07, 2019-06-24:
> ESPCOM::println (WiFi.localIP().toString().c_str(), PRINTER_PIPE);

with the latest changes (delay added) or clean code?

luc-github, 2019-06-24:
I do not have REARM with Marlin but it was reported Ok in the past 
well if REARM restart when ESP is starting it should also reset lcd and so nothing should appear on LCD but cleam Marlin menu
on clean code would be better - too many change is not good

sl1pkn07, 2019-06-24:
with only 

~~~patch
diff --git a/esp3d/esp3d.ino b/esp3d/esp3d.ino
index aba3c02..c0124f1 100644
--- a/esp3d/esp3d.ino
+++ b/esp3d/esp3d.ino
@@ -26,7 +26,7 @@ Esp3D myesp3d;
 //Setup
 void setup()
 {
-    myesp3d.begin();
+    myesp3d.begin(15000);
 }
 
 //main loop
diff --git a/esp3d/wificonf.cpp b/esp3d/wificonf.cpp
index c54b971..fc8166c 100644
--- a/esp3d/wificonf.cpp
+++ b/esp3d/wificonf.cpp
@@ -541,6 +541,7 @@ bool WIFI_CONFIG::Setup (bool force_ap)
 #endif
             return false;
         }
+ESPCOM::println (WiFi.localIP().toString().c_str(), PRINTER_PIPE);
 #ifdef ARDUINO_ARCH_ESP8266
         WiFi.hostname (hostname);
 #else
~~~

https://www.youtube.com/watch?v=VUONYQOtrn4
(sorry the koality)

sl1pkn07, 2019-06-24:
same patch with plug/unplug the TX pin

https://www.youtube.com/watch?v=a3HgWvxrc_U&feature=youtu.be

you can notice the ReARM is boot more fater than with the TX pin plug, and when plux the TX pin the message start appears with normaly

luc-github, 2019-06-24:
I think there is something in Marlin 2.0 which do that but I am not sure what, sorry :( I do not know this ESP board neither.
according https://github.com/MarlinFirmware/Marlin/issues/14345 Re-ARM and Marlin 2.0 are still in beta, this problem is may be a regression issue

luc-github, 2019-06-25:
I have an ARM but it use Smoothieware and no issue - I will try to install latest Marlin 2.0 when I have time to check behavior

sl1pkn07, 2019-06-25:
this is my marlin config

https://github.com/sl1pkn07/Marlin/commit/cf696e0f1df8e356780a85e1dde16c0826912e21

the ESP is connected to J4 in the Rearm (the the breakout i used have a little 5v to 3.3v LDO for no need worry about voltage (powered directly by 5v)

greetings

luc-github, 2019-06-25:
I can reproduce issue on my RE-ARM and Viki2 Screen - it is super weird - if restart ESP after using web UI - I do not have the issue, means if Re_ARM is alreadu power up

Also when have issue if it stuck to `Connecting.`
if I send M115 using command I get `ConnectingM115` I think there is an issue in Marlin FW and serial init I will see if I can work around...

sl1pkn07, 2019-06-25:
if software not works, i start thinking in a piece of hardware with the cheap delay circuit C005 and this component https://docs-emea.rs-online.com/webdocs/15f5/0900766b815f55a7.pdf (for use as relay), for temporize the connexion of the TX pin (see my last video, uses manual (badum tss) emulation)

luc-github, 2019-06-25:
Hi @sl1pkn07 can you try this on clean FW - no need any previous changes
just comment https://github.com/luc-github/ESP3D/blob/2.1/esp3d/wificonf.cpp#L532

you get the IP at the end just no `Connecting...` message
let me know if it works for you as same as for me

sl1pkn07, 2019-06-25:
yep, now the ESP and rearm is booted and the IP is printed in the LCD

luc-github, 2019-06-25:
I have tested unchanged FW on same configuration but with Smothieware instead of Marlin and no issue.
I suspect some issue in Marlin Rx at cold start because if restart ESP3D when Marlin is already started there is no issue.

I will add a define in config.h to disable the `Connecting...` output as a workaround - I hope issue will be solved later on Marlin FW

sl1pkn07, 2019-06-25:
oki

luc-github, 2019-06-25:
workaround available https://github.com/luc-github/ESP3D/commit/abe75464930b3fafd8cbce26fe81fad245e069e9

closed 2019-06-25 by luc-github