luc-github/ESP3D #345: SD upload failed larg size file[BUG]

Varshil18, 2019-06-28:
I cant upload file over wifi in esp3d. below details of my esp3d.

M115
FIRMWARE_NAME:Marlin 1.1.9 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1

i cant upload via repetier or via web interface. i can upload small size file like 300kb but i cant upload 8000kb file.

Please help.

luc-github, 2019-06-28:
please file the informations requested in bug report I need them to understand your issue

Varshil18, 2019-06-28:
**ESP3D Firmware:**
- Version: 2.1.0.b27
- Web UI Version: 2.1b20
- Wifi mode: Client Station
- Any change done in FW - No
UI: 2.1b20 / FW: 2.1.0.b27

**Target Firmware:**
 - Name: Marlin
 - Version 1.1.9

**Board used:**
 - MCU: ESP8266
 - Name:[e.g. NodeMCU2s] 
 - Flash size: [e.g. 4M: 2M/2M]
https://robotdyn.com/mega-wifi-r3-atmega2560-esp8266-flash-32mb-usb-ttl-ch340g-micro-usb.html

Varshil18, 2019-06-28:
I loved your work and project. thanks for making. cant wait for 3.0. kindly guide me for this issu. thank you.

luc-github, 2019-06-28:
This configuration works and you confirm it works with small files.
be sure you do not use baudrate too fast, what baud rate do you use ?
be sure you only upload and do not do any other activity when doing upload
I would say check your connection between ESP and Board but they are on same PCB ? right ?  

also what is the error message ? happen when in upload ?

Varshil18, 2019-06-28:
baudrate 115200
yes esp and bord on same pcb all things works well escap upload take long time

Varshil18, 2019-06-28:
no error occured during upload but 8000kb file take 1hour to upload

luc-github, 2019-06-28:
no error but you wrote : SD upload failed larg size file
well 1h I am not surprised upload is slow like hell

Varshil18, 2019-06-28:
yes it ocurred at the end of uplode. after 100% complete.

Varshil18, 2019-06-28:
normally how mutch time 10000kb file take to upload?

luc-github, 2019-06-28:
about speed : https://github.com/luc-github/ESP3D/issues/292

did you check on SD card what size was uploaded ?

Varshil18, 2019-06-28:
yes i can check file size after upload.

luc-github, 2019-06-28:
>normally how mutch time 10000kb file take to upload?

too long for me - this feature is implemented for compatibility but IMHO it is useless for big files
and really reliable, that is why I worked with panucatt for a solution and also thinking about a ESP3D panel acting as host in 3.0

Varshil18, 2019-06-28:
ok no problem i am waiting for 3.0.
Thank You.

closed 2019-06-28 by Varshil18