luc-github/ESP3D #346: Compile error PAGE_NOFILES esp8266

huexpub, 2019-06-30:
arduino 1.8.9 with 8266 2.52 

i get code renew for mks tft ,so down code 2.1 Compile，but i get error 

```
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_634839\sketch\syncwebserver.cpp: In function 'void handle_web_interface_root()':

syncwebserver.cpp:131:60: error: 'PAGE_NOFILES' was not declared in this scope

     web_interface->web_server.send_P(200,CONTENT_TYPE_HTML,PAGE_NOFILES,PAGE_NOFILES_SIZE);

                                                            ^

syncwebserver.cpp:131:73: error: 'PAGE_NOFILES_SIZE' was not declared in this scope

     web_interface->web_server.send_P(200,CONTENT_TYPE_HTML,PAGE_NOFILES,PAGE_NOFILES_SIZE);
```

luc-github, 2019-06-30:
You may delete something in code - compilation is working fine
[image]

luc-github, 2019-06-30:
I suggest you clean your current esp3d directory and to download repository again

huexpub, 2019-07-01:
emm  nothing  i redownload code and well Compile

closed 2019-07-01 by huexpub