luc-github/ESP3D #347: Latest version of firware cauing problem with Smoothieware

mvadu, 2019-07-07:
Today I built new version of firmware and upgraded the ESP8266 embedded in TinyFab CPU for Cetus printer. Its a Smoothiware board. I am using fairly new version of Smoothie

Old firmware: UI:2.0c9 / FW: 2.0.0.c19
New: UI: 2.1b41 / FW: 2.1.0.b30
smoothie version: Build version: edge-9f3a9cc, Build date: May 27 2019 20:53:27, MCU: LPC1768, System Clock: 100MHz

After this upgrade I am facing ew problems:
1. Uploading files is taking forever. Same 3MB gcode that took less than a minute  now takes forever. 
2. Deleting files with long names is failing. It was working in the old version. Also its issuing rm command, where as previous version was issuing M30 I believe.

I built the firmware using the Platform IO config in the repo using visual studio code.

luc-github, 2019-07-07:
2.0.0c is a test firmware for  Azteeg wifi from Panucatt device which allows direct SD access unlike tne public code on my repository. So what you describe with 2.1 is normal behavior as it use serial upload and smoothieware delete commands, unlike the azteeg wifi firmware.

Sorry, I think you can only stick to what tinyfab gave you as I do not give support for it.

closed 2019-07-07 by luc-github

mvadu, 2019-07-07:
@luc-github ah!! I did not know that. Is there any easy way to get that build back? I don't see separate branch for Azteeg.

luc-github, 2019-07-07:
No sorry code is not public. I suggest you to contact tinyfab

mvadu, 2019-07-07:
Okay sure, thank you! Looking at the Tinyfab schematic, they don't even mention anything about the Azteeg or Panucatt. It gives direct access to SD card via FST3257 bus multiplexer so both Smoothie CPU and ESP8266 gets to use SD card. If you are not under NDA, can you open source the code removing any specific implementation details for the Azteeg?

luc-github, 2019-07-08:
I guess they studied the Azteeg wif board and copied it.