luc-github/ESP3D #348: [FEATURE REQUEST]

MP7, 2019-07-09:
I need to add button for switch On & Off, eg. If use Wifi can switch On or Off via WEB UI, but if I need to print directly from SD Card need to switch On by button. (my printer is couple floor down), so I haven't every time lap with me. 
EG. if I connect push button to GPIO 5(pull Up), need somehow to logically conect to macro function..
have you some suggestion ?

luc-github, 2019-07-09:
use macro feature of webUI - add a button and add a gcode file to be executed which use 
direct pin control command https://github.com/luc-github/ESP3D/blob/2.1/docs/Commands.txt#L53-L59

MP7, 2019-07-09:
it's work from webUi, but need read pin from ESP automatically then trigger macro.

luc-github, 2019-07-09:
you need to add an `attachInterrupt ` function linked to the pin you target and the state you want to catch there is an example for the recovery pin in 3.X https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/modules/recovery/recovery_service.cpp#L61-L62
which trigger a reset

luc-github, 2019-07-11:
I think allowing such thing need to be hard coded and so defined like recovery pin is done, also to keep it simple, limit to 1 pin

MP7, 2019-07-12:
Thx Luc. I understood, Will try next week.

MP7, 2019-08-02:
Hi Luc

sorry for delay, I've been on the trip for a while now, figured out how to
solve my problem I just didn't have time to deal with it.
I will but when at the moment is big Q..
thank you for help, will send my modification via email asap..

best regards,

closed 2019-08-26 by stale[bot]

royfocker19, 2020-02-06:
> Hi Luc sorry for delay, I've been on the trip for a while now, figured out how to solve my problem I just didn't have time to deal with it. I will but when at the moment is big Q.. thank you for [...]

Hey, were you able to solve this finally? 
Regards
Fede

MP7, 2020-02-08:
sorry , not even to start solving.
a bad year is still around me, I need time that I don't have..

Regards,

Mario
 even start.  :(..

čet, 6. velj 2020. u 23:52 royfocker19 <notifications@github.com> napisao
je: