luc-github/ESP3D #350: [FEATURE REQUEST] fastest baud rate, like 921600, 1000000, 2000000 bps

rafaljot, 2019-07-13:
Pls add fastest baud rate, like 921600, 1000000, 2000000 bps
There is option 921600 in drop down but I can't chose it. It turns to red after selection.

luc-github, 2019-07-13:
Hi about 921600 it is a space that mess up the list - I fixed it in FW and worked around in webUI so you just need to update latest webUI and it works 

About other speed - sorry I won't do that because these speeds are not well supported and corrupt the serial transfert - as mentionned here : https://github.com/luc-github/ESP3D/issues/93#issuecomment-407935405 and confirmed here : https://github.com/luc-github/ESP3D/issues/296#issuecomment-460080769

rafaljot, 2019-07-13:
hahaha. I copied "921600 " and add "2000000 ", etc with same sapce, with the same issue ;)

About other speed. it probably depends on uC an other factors. SKR-PRO is quite fast.
I will test it.

luc-github, 2019-07-13:
yes space can be naughty ^_^
Sure let us know your finding

rafaljot, 2019-07-13:
4000000 works for me. It means it can read file list, folders from SD, commands like home, move, M117 works etc. Unfortunately file upload doesn't work at all speeds but as I understand it is old issue with Marlin https://github.com/luc-github/ESP3D/issues/321

luc-github, 2019-07-13:
SD Upload should work what ever the speed, as mentionned in links I shared ealier but some data may be corrupted and need some retry which is  implemented. 
Be sure you are not in debug mode, and you are not doing anything when doing upload, ESP8266 is very limited for that.
 
I do not have STM32f4 board, I guess you use the 2.0 version of marlin ?
I tested with Marlin 2.0 with RE-ARM and it was ok.

rafaljot, 2019-07-13:
Yes, it is last Marlin 2.0-bugfix

luc-github, 2019-07-13:
Well this branch is still under heavy development  code worked before  on 2.0 and work with 1.1.9
So issue is not on my side.
May be trying serial fileupload with pronterface may help to see where issue is?

rafaljot, 2019-07-13:
It starts to write file. create file size 0 on SD then esp3d freezes. File size does not matter. nothing in console.

luc-github, 2019-07-14:
I just tested this file on Marlin 1.1.9 and current ESP3D 2.1 git code at 250000 baud rate
[happy.zip](https://github.com/luc-github/ESP3D/files/3389603/happy.zip)

And it upload is working as well as printing after upload - same code was working on 2.0 before - 
I do not have a system with Marlin 2.0 right now but if ESP freeze it may be a power supply issue - when uploading esp use a lot of power and power need to be stable.

rafaljot, 2019-07-22:
Uploading by ESP3D:
[image]

by OctoPrint 
works :)

by RepetierHost (code with one line M117 ):
[image]

luc-github, 2019-07-22:
Error 0 means webui cannot connect to webserver

luc-github, 2019-07-22:
Also how did you solved your previous issue ?

rafaljot, 2019-07-22:
I made 8 tests wit 250000:
3 - "Error 0"
5 - freezes

luc-github, 2019-07-22:
What is your signal strengh ?

rafaljot, 2019-07-22:
58%

rafaljot, 2019-07-22:
One more try:
[image]

luc-github, 2019-07-22:
obviously there is some communication issue as no checksum is detected by printer
58% is not great but should be Ok
what is `[ESP420]plain ` output ?

rafaljot, 2019-07-22:
[ESP420]plain
Chip ID: 4639724
CPU Frequency: 80Mhz
Free memory: 35.91 KB
SDK: 2.2.1(cfd48f3)
Flash Size: 1.00 MB
Available Size for update: 305.26 KB(Not enough)
Available Size for SPIFFS: 228.20 KB
Baud rate: 250000
Sleep mode: None
Channel: 1
Phy Mode: 11g
Web port: 80
Data port: 8888
Hostname: MYESP1
Active Mode: STA (BC:DD:C2:46:CB:EC)
Connected to: ProkocimA
Signal: 62%
IP Mode: DHCP
IP: 192.168.100.103
Gateway: 192.168.100.1
Mask: 255.255.255.0
DNS: 192.168.100.1
Disabled Mode: AP (BE:DD:C2:46:CB:EC)
Captive portal: Enabled
SSDP: Enabled
NetBios: Enabled
mDNS: Enabled
Web Update: Enabled
Pin Recovery: Disabled
Authentication: Disabled
Target Firmware: Marlin
M117 output: Enabled
Notifications: Disabled
Serial output: Enabled
Web socket  output: Enabled
TCP output: Enabled
FW version: 2.1.0.b30 ESP8266/8586

luc-github, 2019-07-22:
you must flash to CPU Frequency: 160Mhz not 80Mhz -  that may explain why Webserver is not always responding

rafaljot, 2019-07-22:
I've tried both frequencies  with the same result.

luc-github, 2019-07-22:
It may not be the complete solution but  it is part of it 160 Mhz is a must

How your esp is powered?

rafaljot, 2019-07-22:
OK. It is 160Mhz
I've tried ESP-01S connected and powered from dedicated port on SKR-PRO
and LoLin NodeMcu v3 powered from additional USB PSU (2Amp)

rafaljot, 2019-07-22:
Upload to SD with OctoPrint works @ 4000000

luc-github, 2019-07-22:
Stm32 is 3,3v ? sorry I am not familiar with this board.
Yes your pi and stm32 can handle 4000000,   but esp8266 cannot do well as I mentioned earlier
Also error 0 means no http response, so not related to serial  speed, and freeze is not normal  neither so  may esp crash and restart,

rafaljot, 2019-07-23:
Previous test with esp8266 I made at 250000 and 115200. I wanted to show that SKR-PRO could handle mush higher speed. 
Frankly speaking esp8266 works stable with 4000000 as well but not the upload.
Yes, SKR-PRO  is 3.3 V

I think it is something similar to:
https://github.com/MarlinFirmware/Marlin/issues/11871
and 
https://github.com/foosel/OctoPrint/issues/2285

luc-github, 2019-07-23:
It is because of Marlin issue I introduced the checksum, I am in thread your linked.
But even with checksum the upload was ok but data was corrupted at high speed due to weakness of checksum https://github.com/luc-github/ESP3D/issues/296#issuecomment-460872648

I have no clue why ESP webserver does not respond during upload- neither freeze - I cannot reproduce with my hardware

Upload is tricky only for medium / big files because ESP cannot do anything else when upload is ongoing, any action during upload will overload the ESP MCU and so data will be lost/corrupted

but in your case ESP does not respond so it is not this issue, sorry I am dry on this

rafaljot, 2019-07-23:
Could you give me an advice how to setup environment to debug ESP in this case?

luc-github, 2019-07-23:
You are not in async webserver, right ?

to setup debug use nodemcu
enable debug in config to SPIFFS but use only small file or it will fail due to SPIFFS slow down everything, clean often the log file
uncomment lines : https://github.com/luc-github/ESP3D/blob/2.1/esp3d/config.h#L151-L152

the SD file upload is here : https://github.com/luc-github/ESP3D/blob/2.1/esp3d/syncwebserver.cpp#L1172
Serial command is here :
https://github.com/luc-github/ESP3D/blob/2.1/esp3d/webinterface.cpp#L76-L306

but if ESP crash you won't be able to see anything as cannot log this output from ESP, need to duplicate GND and TX lines from ESP and connect to PC using Serial/USB adpter to see the output in serial terminal / monitor

luc-github, 2019-07-25:
@rafaljot did you made some progress ? same behavior was reported with esp32 and custom Marlin Firmware

rafaljot, 2019-07-25:
I tried to upload 1KB file with hundred M117 commands inside ( br=115200 )
It seems that ESP sent 42 only , then showed an error 0 message.

[log.zip](https://github.com/luc-github/ESP3D/files/3433626/log.zip)

luc-github, 2019-07-26:
please use not async - the log show you are using it and it show also the transfer is ok , which means async loose data, that is why it is not recommended to use async with serial, async do not raise error when not answering or ignoring packet.

rafaljot, 2019-07-26:
I misunderstood you. I thought you suggested that I should turn on the asynchronous mode before debugging. I turned it on yesterday first time.
BTW: async needs 
`compiler.c.extra_flags=-fpermissive `
due to mess with char* and uint8_t*

So I repeated the test without async.
I started to upload file (2000 lines, 26KB) and reset after 10 minutes.
It uploaded only 639 of 2000 lines.

Maybe it is hardware problem. Now I realized I tested with Octoprint and it works perfect but I used serial0 and USB cable , with ESP I use Serial6 and direct uC-uC tx/rx  connection  

(...)
************
Got ok
Purge Serial
Purge done
Send line M117 0000635  NB:636
Purge Serial
Purge done
Send line N636 M117 0000635*55
Response:
************
ok

************
Got ok
Purge Serial
Purge done
Send line M117 0000636  NB:637
Purge Serial
Purge done
Send line N637 M117 0000636*53
Response:
************
ok

************
Got ok
Purge Serial
Purge done
Send line M117 0000637  NB:638
Purge Serial
Purge done
Send line N638 M117 0000637*59
Response:
************
ok

************
Got ok
Purge Serial
Purge done
**Send line M117 0000638  NB:639
Set STA mode**
SSID XXXXXXXXXXXXXX
Setup Done
2 Web command
plain:[ESP800]
PAGEID:
Web Command:[ESP800]
Execute Command
2 Web command
plain:[ESP800]
PAGEID:
Web Command:[ESP800]
Execute Command
2 Web command
plain:[ESP400]
PAGEID:
Web Command:[ESP400]
Execute Command
request:/macrocfg.json
1564139535656:
PAGEID:
type:application/octet-stream
request:/preferences.json
1564139535656:
PAGEID:0
type:application/octet-stream
request:/log.txt
type:text/plain

luc-github, 2019-07-26:
Yes I see a reset in log - the only way to know what do the reset it is to duplicate GND line and esp tx line and connect to PC and you can disable log, when esp RESEST/Crash  it will show threason and the error stack - dfecoding stack should explain what is the root cause / hardware / software or at least point out when it happen

rafaljot, 2019-07-26:
I get it. It does not freeze. It is terrible slow. 
100 lines, 1KB gcode file takes him apx two minutes to upload. That's why I thought he was hanging out on bigger files.
 
I added some extra logging and it looks like it works but terrible slow.
```
LOG(millis())
    LOG(" Purge Serial\r\n")

```
2291353-2292816= 1.5s per line 

```
************
Got ok
2291241 Purge Serial
Purge done
2291353 Send line M117 0001382  NB:1383
#2000#2010#2020#2030#2040#20502291801 Purge Serial
Purge done
#2060Send line N1383 M117 0001382*5
#2070flush
if wait_for_data
true
memory:26488

#1000
#1010
2292627 Response:
************
ok

************
Got ok
2292816 Purge Serial
Purge done
2293149 Send line M117 0001383  NB:1384
#2000#2010#2020#2030#2040#20502293596 Purge Serial
Purge done
#2060Send line N1384 M117 0001383*3
#2070flush
if wait_for_data
true
```

rafaljot, 2019-07-26:
It is possible that at the beginning I also had other problems with power supply or something.

luc-github, 2019-07-26:
yes slow like hell - I know it is disapointing, I just support this feature by courtesy - I do not use it, it is useless for big files

rafaljot, 2019-07-27:
it should transfer many lines at once.
DuetWifi works great with ESP but it use ISP instead UART

luc-github, 2019-07-28:
>DuetWifi (RepRapFirmware) works great with ESP but it use ISP instead UART

It does not use M28/M29 protocol but write packets of 2048B to SD, same size of packet received, so it is faster

>Next observation is upload speed nearly doesn't depends on bout rate.

Yes baudrate it is not the bottleneck, the M28/M29 protocol is (said here https://github.com/luc-github/ESP3D/issues/290, and https://github.com/luc-github/ESP3D/issues/216)

>removed purge_serial() and it is faster 

Yes but I put it for safety to not catch wrong ack, you can also remove every checks, not wait for any ok  to consider it is ok and hope no error happen, it will be faster just not reliable ^_^, slow for slow I prefered to make it reliable.

Anyway what ever the hack it will still be super slow and almost useless due to protocol

luc-github, 2019-07-29:
I guess issue can be closed now

closed 2019-07-29 by luc-github