luc-github/ESP3D #352: Compiling error

Fliens, 2019-07-26:
I keep getting this error:
In file included from C:\Users\flien\Documents\Arduino\libraries\ESP3D-2.0\src\esp3d.cpp:85:0:

C:\Users\flien\Documents\Arduino\libraries\ESP3D-2.0\src\syncwebserver.h:25:30: fatal error: WebSocketsServer.h: No such file or directory

 #include <WebSocketsServer.h>

                              ^

compilation terminated.

exit status 1

I've installed the board with git so there should be no problem but i keep getting this error :/

luc-github, 2019-07-26:
please use 2.1 instead of 2.0 it has several bug fixes 
please read the readme installation part for libraries installation and it is not recommended to use async due to stability issue

Fliens, 2019-07-26:
thats the error im getting now:

In file included from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:30:0,

                 from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecure.h:41,

                 from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiServerSecure.h:20,

                 from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:41,

                 from sketch\wificonf.h:27,

                 from sketch\config.h:250,

                 from sketch\esp3d.cpp:35:

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/CertStoreBearSSL.h:40:23: error: 'FS' has not been declared

     int initCertStore(FS &fs, const char *indexFileName, const char *dataFileName);

                       ^

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/CertStoreBearSSL.h:46:5: error: 'FS' does not name a type

     FS *_fs = nullptr;

     ^

sketch\wificonf.cpp: In member function 'bool WIFI_CONFIG::Setup(bool)':

sketch\wificonf.cpp:242:45: warning: 'void ESP8266WiFiGenericClass::onEvent(WiFiEventCb, WiFiEvent_t)' is deprecated (declared at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:54) [-Wdeprecated-declarations]

     WiFi.onEvent(onWiFiEvent, WIFI_EVENT_ANY);

                                             ^

sketch\notifications_service.cpp: In member function 'bool NotificationsService::sendPushoverMSG(const char*, const char*)':

sketch\notifications_service.cpp:157:19: warning: 'axTLS::WiFiClientSecure::WiFiClientSecure()' is deprecated (declared at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecureAxTLS.h:35): Upgrade to BearSSL is advised, check https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecure.h#L25-L99 [-Wdeprecated-declarations]

     TSecureClient Notificationclient;

                   ^

sketch\notifications_service.cpp: In member function 'bool NotificationsService::sendEmailMSG(const char*, const char*)':

sketch\notifications_service.cpp:190:19: warning: 'axTLS::WiFiClientSecure::WiFiClientSecure()' is deprecated (declared at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecureAxTLS.h:35): Upgrade to BearSSL is advised, check https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecure.h#L25-L99 [-Wdeprecated-declarations]

     TSecureClient Notificationclient;

                   ^

sketch\notifications_service.cpp: In member function 'bool NotificationsService::sendLineMSG(const char*, const char*)':

sketch\notifications_service.cpp:283:19: warning: 'axTLS::WiFiClientSecure::WiFiClientSecure()' is deprecated (declared at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecureAxTLS.h:35): Upgrade to BearSSL is advised, check https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecure.h#L25-L99 [-Wdeprecated-declarations]

     TSecureClient Notificationclient;

                   ^

In file included from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:30:0,

                 from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:214,

                 from sketch\webinterface.h:37,

                 from sketch\esp3d.cpp:41:

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src/detail/RequestHandlersImpl.h:67:28: error: expected ')' before '&' token

     StaticRequestHandler(FS& fs, const char* path, const char* uri, const char* cache_header)

                            ^

esp3d.cpp:340:1: error: expected '}' at end of input

 }

 ^

esp3d.cpp:340:1: error: expected unqualified-id at end of input

esp3d.cpp:340:1: error: expected '}' at end of input

exit status 1
expected '}' at end of input

luc-github, 2019-07-26:
please use the version as recommended in read me esp core 2.5.2 not git

luc-github, 2019-07-29:
I guess issue is closed

closed 2019-07-29 by luc-github