luc-github/ESP3D #353: [BUG] ESP-01 with PUYA Flash

AFUman, 2019-07-27:
**Describe the bug**
I am aware that there are problems with the ESP-01 with a PUYA flash. I tried a lot because I get SPIFFS up and running the ESP3D too. Unfortunately, I have found no tips on the Internet, which work. Unfortunately, after the ESP-01 flashes with ESP3D 2.1, the ESP-01 does not start, no WLAN identifier is transmitted. The ESP-01 is alright, I flashed Tasmota as a test and that starts everything and it works. Does anyone have a hint or maybe a finished bin file, which I can flash. Sit at the problem for a few days and do not get it to work. I tested it on a WEMOS, because ESP3D runs without problems. But I would like to use the ESP-01, because the hardware has a socket for where I can plug it directly.

**ESP3D Firmware:**
- Version:2.1
  

**Board used (please complete the following information):**
 - MCU: ESP8266
 - Name:ESP-01
 - Flash size: PUYA 1M
**Additional context**
Add any other context about the problem here.
Wiring, decoded stack, etc...

luc-github, 2019-07-27:
1 how did you applied the define ?

AFUman, 2019-07-30:
Unfortunately I did not get any further, I just get ESP3D running on the ESP-01 with Puya Flash.

luc-github, 2019-07-30:
sorry before you wrote cannot - now can ? so what is the issue ?

AFUman, 2019-07-30:
Unfortunately there was a mistake in the translation and unhappily I have not got any further. I haven’t found a working solution because of the SPIFFS file system on an ESP-01 with a puya flash. Unfortunately none of the many tips works.

luc-github, 2019-07-30:
So back to original question how did you applied the patch

AFUman, 2019-07-30:
I found different esp.ccp files in different forums. I have tried this, I have also found patches but unfortunately no info how to install this.

d-a-v, 2019-07-30:
When using arduino ide to compile esp3d and if you use latest 2.5.2 esp8266 arduino core, you can create this file `platform.local.txt` next to `platform.txt` in your directories containing:
```
compiler.c.extra_flags=-DPUYA_SUPPORT=1
compiler.cpp.extra_flags={compiler.c.extra_flags}
```

Note: it is ***not*** sufficient to add `#define PUYA_SUPPORT 1` anywhere in your code.

What would be better for everybody is to go to esp8266 arduino github repository and open a new issue asking for default support for these flash chips.

luc-github, 2019-07-30:
The issue and solution is actually pinned on top of issues

luc-github, 2019-07-31:
@AFUman thanks to @d-a-v the puya settings is now applied by default
You need to take latest git from https://github.com/luc-github/ESP3D/tree/2.1
and latest git from https://github.com/esp8266/Arduino

and puya is applied by default - hope it solve your issue

AFUman, 2019-07-31:
Thank you for your help. I can write SPIFFS now. But after flashing of ESP3D will not start unfortunately no output on'dem terminal and also no wifi with ESP3D SSID. I flash Easypesy with the tool on the ESP-01 starts the Smarthome solution without problems. Could you maybe make me a BIN for the ESP-01 I think I'm doing something wrong.

luc-github, 2019-07-31:
flash using `Erase Flash :  All Flash Contents`
To be in clean situation

AFUman, 2019-07-31:
OK, thank you I'll give it a try.

luc-github, 2019-08-04:
@AFUman any update ?

AFUman, 2019-08-05:
Yes, i have ESP3D on my ESP-01 and all is OK. Many thanks for your help

closed 2019-08-05 by AFUman