luc-github/ESP3D #354: ESP3d Wont upload

TurtleDude29, 2019-07-30:
Hey i got this error when trying to upload. I got a different one earlier but i fixed that and got this one. Thank you to anyone trying to help!

esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM15
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\Users\yonat\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "C:/Users/yonat/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2653, in main
    esp.connect(args.before)
  File "C:/Users/yonat/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 468, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

luc-github, 2019-07-30:
Hi this one is related to esp8266 core. Better to open issue on esp8266 github

luc-github, 2019-07-30:
I would suggest to remove esp8266 core completly and do a new fresh install as under windows 10.  2.5.2 works on my side

TurtleDude29, 2019-07-30:
Okay. Thank you ill try that and if it doesnt work ill ask on the esp github

luc-github, 2019-07-30:
be sure nothing left in C:\Users\yonat\Documents\ArduinoData\packages\esp8266 after removal

TurtleDude29, 2019-07-30:
okay

TurtleDude29, 2019-07-31:
i put a jumper on Ground and GPIO0 because i have a programmer and it worked but i screwed it up.
i connected to the esp wifi and messed up the setup. i misspelled the password or something where the esp3d wifi does not show up in my router or my wifi networks. Will i need a new esp?

luc-github, 2019-07-31:
Just use serial terminal and ESP commands to setup your wifi  or reset settings

TurtleDude29, 2019-07-31:
How do i do that? im a noob

TurtleDude29, 2019-07-31:
Found it. Thank you

TurtleDude29, 2019-07-31:
Nothing is showing up when i type commands

luc-github, 2019-08-01:
be sure you use correct baud rate and NL/CR is selected
[image]
then type
`[ESP444]RESET`
and so reboot with `[ESP444]RESTART`

luc-github, 2019-08-04:
@TurtleDude29 Any update ?

TurtleDude29, 2019-08-04:
Oh yeah i forgot to say. Yes it worked perfectly. Thank you!!!!!!!!!

luc-github, 2019-08-04:
-Great

closed 2019-08-04 by luc-github