luc-github/ESP3D #355: connecting esp-01 to anet A8

drunkscientist, 2019-08-01:
it looks like [someone got close](https://github.com/luc-github/ESP3D/issues/334) #334 before having a bad day, i would like to hopefully finish what they started and update the wiki. i dont have the same board model (not sure if the green connectors for power/bed/extruder are newer or older) but is it safe to assume that the connectors for j3 are the same? if im reading this board correctly, it has some sort of 1117 regulator, and j3 has a 3.3v pin which should have enough to power the esp. the rx/tx of the esp can then connect to tx/rx of j3, probably through a level shifter/voltage divider. or do i need to bypass the onboard ch340 chip? i was trying to understand why there were extra wires being soldered. if that is the case would it not be easier to use the middle pair of pins from the USB BLE header, according to[ this diagram ](https://github.com/ralf-e/ANET-3D-Board-V1.0/blob/master/ANET3D_Board_Schematic.png) those middle two pins are AIFO and AOFI, which connect directly to the atmegas rx/tx pins, and they become the usb_tx/rx at the ch340. i might have confused myself...

luc-github, 2019-08-01:
@rhialto56 got success https://github.com/luc-github/ESP3D/issues/334#issuecomment-499260771
He may answer your question 

For me you need to use serial pins not ch340 usb pins to connect to ESP

drunkscientist, 2019-08-01:
ok after much staring at this schematic and tutorials pictures, im pretty confident the AOFI and AIFO pins are connected directly to the atmega. assuming they are, i should be able flash the esp and connect rx/tx lines to there and be good? just need to make sure to disconnect before using usb

luc-github, 2019-08-11:
Any update on this ?

drunkscientist, 2019-08-11:
had other unrelated hardware failure before i could test. still plan to try again once replacements arrive

closed 2019-08-11 by drunkscientist