luc-github/ESP3D #356: Panucatt Wifi Backpack firmware source.

gerleimarci, 2019-08-12:
Hi,
Is it possible to share the firmware source? It would be even better with schematic. I only want to experiment with it on proto board before buying the actual product.
Also being open source is a nice thing🙂

luc-github, 2019-08-12:
Sources are 99% identical as this repo, the only difference is the direct access to SD, for this feature sources are not available yet - will be one day, but not yet. 

Here the videos so you can have a look to have an idea.
https://www.youtube.com/watch?v=pJGBRriNc9I
https://www.youtube.com/watch?v=br7BryjWPJU&

I do not have any schematics, devices are done by Panucatt devices so better check there.

gerleimarci, 2019-08-13:
The interesting part for me is the SD access. The serial upload is slow, buggy and doesn't compatible with all 3D printer firmware. So please contact with Panucatt and release the sources (at least the firmware, but it would be nice to release the hardware, too).

luc-github, 2019-08-13:
Yes I agree serial upload is a outdated and slow like hell protocol - It is why I looked for a solution, I spent a lot of time on it and , so for firmware sources I already answered, as I am the owner of sources, Roy from Panucatt is the owner of hardware if you have request for it, please feel free to contact him, I am sure he will answered to you if you explain you want to clone his hardware to do a prototype ^_^.

Also Panucatt board + ESP3D is not the only solution for SD access, and some are fully open already:
Marlin 2.0 has some POC to change this protocol for a faster one, github has also some hardware +FW solution, so you have already several choice for your protoboard if you cannot wait for the ESP3D sources release.

luc-github, 2019-08-14:
I guess issue can be closed

closed 2019-08-14 by luc-github

edilcurvelo, 2020-08-02:
The source for direct SD access already available? If not, are there any predictions?

luc-github, 2020-08-02:
https://github.com/luc-github/ESP3D/issues/487#issuecomment-666560477