luc-github/ESP3D #357: SD mount on 2.1.0.b33

SanShell, 2019-08-14:
Hi everyone, I just updated my ESP8266 from firmware 1.0 to 2.1.0.b33, the printer is a coreXY (Hypercube) with Smoothieware.
With the 1.0 I had no problem managing the files on the SD using the integrated file browser, now I can no longer use it and I receive the warning that there is no SD mounted, the only way to manage the SD is with the M commands using the integrated serial terminal .
Obviously ESP3D is configured and set to Smoothieware, in fact I can access the firmware configuration via the appropriate tab.
I haven't touched or updated smoothie firmware in a couple of months.
 
Did I miss something on the way? I'm not kidding, but I think I spent at least 4 hours reading wikis, config.h and trying to do various uploads to figure out where's the problem.

I ask you if you know a solution or let me know if I miss to read an issue
[image]
 already closed with the same problem.

Thanks.

luc-github, 2019-08-14:
what is the UI version ?

SanShell, 2019-08-14:
> what is the UI version ?

2.1b20

> what printer board ?

Azsmz mini

> For smoothiware I use console command ls /sd instead of Gcode
Yep, i almost forgot that using multiple firmware

luc-github, 2019-08-14:
can you show the esp setting in web ui for SD 
[image]

SanShell, 2019-08-14:
OFC

[image]

luc-github, 2019-08-14:
I think upgrade did not reset the SD information
try this :
in your webbrowser paste :
`http://192.168.1.36/command?plain=%5BESP401%5DP%3D851%20T%3DB%20V%3D1&PAGEID=0`
change IP with your esp IP then enter
the page should say ok
then go back to you webUI and refresh - and try refresh SD files

luc-github, 2019-08-14:
let me know if it is ok after this

SanShell, 2019-08-14:
Super, now it works! :)

Thanks you thousand for the help and the super fast response.

closed 2019-08-14 by SanShell

luc-github, 2019-08-14:
you are welcome ^_^