luc-github/ESP3D #359: #Issue - esp3d sd upload failed

renanvellone, 2019-08-17:
Hello guys,

I have some problems to upload the files.
I'm using the Arduino Mega with 1.1.x Merlin
and ESP8266 - with ESP3D 1.0 
and RAMPS

Everything works fine, I can send commands and move XYZ normally...

But When I'm going to upload a fille:
in the LCD says: SD upload done.
in the browser: 
Command tab: OK
SD files / : does not show anything.....

could you guys help me?

luc-github, 2019-08-17:
use the 2.1, the 1.0 have not great support of Marlin and also since 1.0 the SD upload process changed in Marlin, they make the checksum mandatory which was not implemented in 1.0

renanvellone, 2019-08-18:
Hello luc,

I’ll test it until tomorrow afternoon. I’ll let you know the status until tomorrow!

Thanks a lot.

renanvellone, 2019-08-18:
Hi luc, I'm seing this error

sketch\notifications_service.cpp:38:35: fatal error: WiFiClientSecureAxTLS.h: No such file or directory

 #include "WiFiClientSecureAxTLS.h"

                                   ^

compilation terminated.

exit status 1

luc-github, 2019-08-18:
what version of esp core are you using ?

renanvellone, 2019-08-18:
I'm using ESP8266

luc-github, 2019-08-18:
version of core

renanvellone, 2019-08-18:
2.4.0

luc-github, 2019-08-18:
As readme explain :
Development version for 2.1 & ESP-WEBUI (2.1 branch):
Arduino ide 1.8.9 with 2.5.2 version of ESP8266 or git version ESP32 for 100% support of ESP32 :

renanvellone, 2019-08-18:
What  version of ESP8266FS i need to use? 0.4.4?

luc-github, 2019-08-18:
You do not need any, the web page at start allow to upload

renanvellone, 2019-08-18:
Ohh the ESP connect and I got the IP, what i need to upload in there?

luc-github, 2019-08-18:
Upload the latest index.html.gz from https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz

renanvellone, 2019-08-18:
Now it's saying: 
  Error 0 :

renanvellone, 2019-08-18:
When I try to upload file.gco it say:   Error 0 :

luc-github, 2019-08-18:
What is the output of[ESP800] and M115 ?

renanvellone, 2019-08-18:
Where I can see this information?
The commands tab is blank.

luc-github, 2019-08-18:
Did you do any change in fw ？
You should be able to send from webui

renanvellone, 2019-08-18:
no, i didnt change anything...

"You should be able to send from webui" mean commands tab?

luc-github, 2019-08-18:
Yes from command panel

renanvellone, 2019-08-18:
[ESP800]
FW version:2.1.0.b33 # FW target:marlin # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81 # hostname:MYESP

M115: Nothing appears (command or SERIAL)

Trying to upload: nothing appears in commands but in serial appears this:

M115
N1 M29*25
N2 M29*26
N2 M29*26
M30 
M117 SD upload failed
N3 M29*27
N3 M29*27
M30 
M117 SD upload failed

luc-github, 2019-08-18:
There is no communication with printer
Check your baud rate and wires, 
Any output for M20 command ?

luc-github, 2019-08-24:
@renanvellone  any update ?

renanvellone, 2019-08-25:
Hello Luc,

take a look:

**M115**
FIRMWARE_NAME:Marlin 1.1.8 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:EEPROM:0
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0

**[ESP800]**
FW version:2.1.0.b33 # FW target:marlin # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81 # hostname:MYNEWESP

and

**M20**
Begin file list
End file list

luc-github, 2019-08-25:
you do not have any gcode file on SD card ?

I see you use Marlin 1.1.8 - the stable version is 1.1.9 - I suggest you to update to latest if you can

I have just  commited some update on ESP3D and WebUI  to catch better the upload errors -  you may update them also

renanvellone, 2019-08-25:
I'm using the esp3d, that i downloaded in the last we, when we talk.
For merlin, do you have the link of the new one?

luc-github, 2019-08-25:
I have updated the esp3d yesterday and webUI today
for marlin : https://github.com/MarlinFirmware/Marlin/releases

renanvellone, 2019-08-25:
Okay, 

One more questio: Is there a easy way to upload the esp3d?

luc-github, 2019-08-25:
compile the firmware will all correct settings using `Export compiled Binary`
[image]

the binary will appear in your esp3d directory 
the in webUI go to FW update , select the compiled binary you just did and update the FW
[image]

to update webUI 
just take the index.htm.gz from webUI repo and in current webUI go to file system manager
[image]
then update index.html.gz
[image]

and refresh page to load the new webUI

luc-github, 2019-08-25:
but if you have gcode file on SD and Marlin do not see then : M20 is empty - you may need to format your SD card using https://www.sdcard.org/downloads/formatter/
M20 is Marlin command - ESP3D can do nothing if M20 do not see files when they are present on SD, file with .g, or .gco extensions

renanvellone, 2019-08-26:
Hey @luc-github,

now When I try to upload it's returning:

Error (9): Mounting SD failed.

follow the new info:

**M115**
FIRMWARE_NAME:Marlin 1.1.8 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:EEPROM:0
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0

**[ESP800]**
FW version:2.1.0.b34 # FW target:marlin # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81 # hostname:MYNEWESP

M20
Begin file list
End file list

renanvellone, 2019-08-26:
Hey @luc-github 

now When I try to upload it's returning:

Error (12 or 3): File write failed:

follow the new info:   

**M115**
FIRMWARE_NAME:Marlin 1.1.9 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:EEPROM:0
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1

**[ESP800]**
FW version:2.1.0.b34 # FW target:marlin # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81 # hostname:MYNEWESP

**M20**
echo:Unknown command: "M20"

luc-github, 2019-08-26:
>M20
echo:Unknown command: "M20"

Looks like you did not enabled SD support when you upgraded Marlin - you need to enable it in your configuration file according your printer

Roman-RV78, 2019-08-28:
Hello. I ran into a similar problem. I can see the files on the SD card, I can delete them, but copying the files to the SD card fails, error. Arduino DUE Marlin 2.0, NODE MCU

M115
FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:CHAMBER_TEMPERATURE:0
[ESP800]
FW version:2.1.0.b35 # FW target:marlin # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81 # hostname:MYESP
M20
Begin file list
CUBE.G 0
End file list

luc-github, 2019-08-28:
@Roman-RV78  I moved your issue to https://github.com/luc-github/ESP3D/issues/365

renanvellone, 2019-08-29:
Hello @luc-github 

now When I try to upload it's returning:

Error (9):  Mounting SD failed

follow the new info:

**M115**
FIRMWARE_NAME:Marlin 1.1.9 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:EEPROM:0
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1

**[ESP800]**
FW version:2.1.0.b34 # FW target:marlin # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81 # hostname:MYNEWESP

**M20**
Begin file list
End file list

luc-github, 2019-08-29:
Ok what the output of  `M21` command ?

renanvellone, 2019-08-30:
@luc-github 

M21
echo:SD init fail
ok

luc-github, 2019-08-30:
OK this is confirm the issue meet by ESP3D, Marlin cannot mount the SD card
be sure your SD card configuration is correct 
be sure your SD card is well formated , use https://www.sdcard.org/downloads/formatter/

renanvellone, 2019-08-30:
Ohh I thought that i didnt neeed an external SD card. The flash memory from esp3d is not enough?

Do I need to use a normal sd card in my LDC (my lcd has a slot to insert a sd card)?

luc-github, 2019-08-30:
ESP3D is just a bridge - not a host and flash is limited
ESP3D help to bridge to your Printer SDCard but does not store on itself
So yes you need to use the SD card on your LCD

luc-github, 2019-09-02:
I think issue can be closed then

closed 2019-09-02 by luc-github

renanvellone, 2019-09-05:
@luc-github ,

when I plug my sdcard in there in the displays Says "card inserted" but in the menu says "No SD card"
When I try to put: 
M21
Error:volume.init failed

Im using full graphic smart controller

I already format my SD card too

renanvellone, 2019-09-05:
@luc-github ,

I insert it:

/* power on init. sd card */
#if ENABLED(SDSUPPORT)
  SERIAL_ECHOLNPAIR("At startup the card is ", card.cardOK ? "OK" : "not OK");
  card.initsd();
#endif
}
and shows:

At startup the card is not OK
Error:volume.init failed

renanvellone, 2019-09-05:
@luc-github 

How can I see if I'm using the correct SD card configurations?

luc-github, 2019-09-05:
You may have a pin config issue, a defect SD reader, a bad SPI speed, 

The correct SD is when you have SD inserted 
and when you do `M21` - it does not failed
then when you do `M20` it can list GCODE files on SD

you have the correct detect pin because it see when card is inserted - so check if SPI speed is not too high or if others pins are properly set

renanvellone, 2019-09-05:
Hi @luc-github 

I tried to change the SPI speed, I tried the 3 option, still not working.

How can I see the others pin? (what pin?)

thanks

renanvellone, 2019-09-05:
My SD has 60GB, is that a problem?

luc-github, 2019-09-06:
this is may be be a Marlin config issue - you will get better support on MArlin forum as I do not really use Marlin

If you use standard Hardware with a standard board - pins should be automaticaly set when you defined  your hardware in Configuration.h

For this capacity I cannot say - I always use 4Gb SD cards sorry

reynolds99i5, 2020-09-18:
Hi @luc-github, 
You make great work!

"**Error 0 :**"

Board SKR 1.4Turbo on LPC1769 Chip
Today i compile ESP3D 2.1 and ESP3D 3.0 and both have this same error while uploading file.
Board software Marlin Marlin-bugfix-2.0.x
I have serial comunication, i see SD card files but this same problem "**Error 0 :**".
After this exception my bed and nozzle temperature is set to printing temperatures (60 and 200).
After error when i refresh sd files (from WEB-GUI) i see this:

[image]

I have definied in marlin configuration.h: 

#define SERIAL_PORT 3
#define LPC_PINCFG_UART3_P4_28

This means:

  For UART3:                     
        define LPC_PINCFG_UART3_P4_28 to use pin 82 and 85, PORT 4-28/29, FUNC 3

M20 code output:
M20
Begin file list
CAT-35~1.GCO 34432620
TEST_B~1.GCO 32784
DOG-25~1.GCO 12449184
PIG-4H~1.GCO 19126652
SPIRAL~1.GCO 241643307
CE3PRO~2.GCO 58462550
11_BAZ~1.GCO 8411667
12_MOC~1.GCO 0
13_CHL~1.GCO 7516263
14_BLT~1.GCO 4559645
15_WYP~1.GCO 3570906
TEST_A~1.GCO 5605723
PLA12_~1.GCO 3406518
PLA11_~1.GCO 7844806
PLA13_~1.GCO 7356504
PLA14_~1.GCO 3986674
CE3PRO~1.GCO 508092
AGA~1.GCO 6170297
PLA14_~2.GCO 863132
OSA~1.GCO 13528471
End file list

M115
FIRMWARE_NAME:Marlin bugfix-2.0.x (Sep 18 2020 17:56:06) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 Pro EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:RUNOUT:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0

luc-github, 2020-09-18:
this is an old thread, if you have issue open new one and fill the template