luc-github/ESP3D #360: [BUG] VERY SLOW PRINT (Temperature Messages too much)

ajaybnl, 2019-08-17:
When i print from repetier over wifi with esp3d on esp8266, the print start ok, but after some time the tempoerature update messages comes every 2 seconds.

Please help me. my print speed becoming too low.

luc-github, 2019-08-17:
wifi is not real time unlike serial, and esp just do bridge, I am not sure I can do anything, espcially if you do not get good bandwidth between esp and your router

ajaybnl, 2019-08-17:
why is the Temperature is coming ?
14:07:09.595 : ok
14:07:09.595 : N1592 G1 F1800 X36.589 Y108.810 E2718.07905*52
14:07:09.610 : ok
14:07:09.610 : N1593 G0 X36.589 Y108.27 F6000*123
14:07:10.268 : T:200.09 /200.00 B:36.81 /0.00 @:58 B@:0
14:07:11.274 : T:199.93 /200.00 B:36.77 /0.00 @:63 B@:0
14:07:12.284 : T:200.18 /200.00 B:36.77 /0.00 @:56 B@:0
14:07:13.292 : T:200.09 /200.00 B:36.75 /0.00 @:59 B@:0
14:07:13.672 : ok
14:07:13.672 : N1594 M117 ETE 4h 09m 17s*4
14:07:13.681 : ok
14:07:13.681 : N1595 G1 F1800 X156.769 Y108.270 E2725.61040*2
14:07:13.693 : ok
14:07:13.693 : N1596 G0 X156.769 Y107.73 F6000*75
14:07:14.303 : T:200.04 /200.00 B:36.77 /0.00 @:60 B@:0
14:07:15.308 : T:199.93 /200.00 B:36.81 /0.00 @:63 B@:0
14:07:16.315 : T:199.93 /200.00 B:36.83 /0.00 @:62 B@:0
14:07:17.324 : T:200.09 /200.00 B:36.83 /0.00 @:58 B@:0
14:07:17.754 : ok
14:07:17.754 : N1597 G1 F1800 X36.589 Y107.730 E2733.14175*55
14:07:17.775 : ok
14:07:17.775 : N1598 G0 X36.589 Y107.19 F6000*114
14:07:18.349 : T:199.97 /200.00 B:36.81 /0.00 @:62 B@:0
14:07:19.357 : T:199.97 /200.00 B:36.85 /0.00 @:62 B@:0
14:07:20.364 : T:200.00 /200.00 B:36.81 /0.00 @:61 B@:0
14:07:21.371 : T:199.93 /200.00 B:36.83 /0.00 @:62 B@:0
14:07:21.839 : ok
14:07:21.839 : N1599 M117 ETE 4h 09m 09s*6
14:07:21.845 : ok
14:07:21.845 : N1600 G1 F1800 X156.769 Y107.190 E2740.67310*12
14:07:21.860 : ok
14:07:21.860 : N1601 G0 X156.769 Y106.65 F6000*64
14:07:22.382 : T:200.00 /200.00 B:36.90 /0.00 @:60 B@:0

see?

luc-github, 2019-08-17:
1- what version of ESP3D do you use ? - there is a template when you open
issue it help to get proper idea of situation
2 - if you do not ask WebUI to send temperature update then temperature
come from repetier host, ESP3D does not send anything to printer by it self

ajaybnl, 2019-08-17:
ESP3d 1.0 , i have not used Temperature update from web gui.

luc-github, 2019-08-17:
then M105 command come from repetier host

ajaybnl, 2019-08-17:
I have shown all commands from log. there's no extra command going. If i print from sdcard, nothing happens.

luc-github, 2019-08-17:
if you print from SD, repetier host is open ?
what happen if you log from terminal like putty ? does it send temperature ?

AnHardt, 2019-08-17:
Probably it's
```
#define AUTO_REPORT_TEMPERATURES
```
from `Configuration_adv.h`.

ajaybnl, 2019-08-19:
Yes, but it does not pause frequently like when printing thru wifi.
SDcard Print:

Yes it gives temperature

09:16:13.066 : T:185.43 /185.00 B:31.08 /0.00 @:46 B@:0
09:16:13.097 : N38 M27*45
09:16:13.353 : SD printing byte 39289/2500282
09:16:13.353 : ok
09:16:14.077 : T:185.57 /185.00 B:31.27 /0.00 @:42 B@:0
09:16:15.088 : T:185.37 /185.00 B:31.19 /0.00 @:48 B@:0
09:16:15.391 : N39 M27*44
09:16:15.480 : SD printing byte 41578/2500282
09:16:15.480 : ok
09:16:16.097 : T:185.43 /185.00 B:31.14 /0.00 @:46 B@:0
09:16:17.105 : T:185.34 /185.00 B:31.16 /0.00 @:48 B@:0
09:16:17.687 : N40 M27*34
09:16:17.729 : SD printing byte 44067/2500282
09:16:17.729 : ok
09:16:18.108 : T:185.34 /185.00 B:31.16 /0.00 @:48 B@:0
09:16:19.120 : T:185.40 /185.00 B:31.23 /0.00 @:46 B@:0
09:16:19.986 : N41 M27*35
09:16:20.127 : T:185.20 /185.00 B:31.25 /0.00 @:51 B@:0
09:16:21.139 : T:185.00 /185.00 B:31.21 /0.00 @:56 B@:0
09:16:21.142 : SD printing byte 44508/2500282
09:16:21.142 : ok
09:16:22.144 : T:184.89 /185.00 B:31.19 /0.00 @:58 B@:0
09:16:22.286 : N42 M27*32
09:16:23.123 : SD printing byte 44688/2500282
09:16:23.123 : ok
09:16:23.150 : T:184.74 /185.00 B:31.19 /0.00 @:61 B@:0
09:16:24.161 : T:184.69 /185.00 B:31.12 /0.00 @:60 B@:0
09:16:24.577 : N43 M27*33
09:16:25.164 : T:184.72 /185.00 B:31.08 /0.00 @:57 B@:0
09:16:25.232 : SD printing byte 44904/2500282
09:16:25.232 : ok

luc-github, 2019-08-19:
as I said ESP does not send M105 so temperature queries come from repetier or from an auto report
if you print from SD and do not open web UI and do not enable auto check - ESP3D does not send any query 
About speed yes SD printing is 1000% better than WiFi printing for me, I never print over wifi. WiFi is not real time transfer so some packet may take more time then others and so slow dow the transfer

Try a serial-wifi bridge firmware to check : https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiTelnetToSerial/WiFiTelnetToSerial.ino

ajaybnl, 2019-08-19:
I have disabled the PING PONG ok communication and reduce "Received Cache Size" to 60 in repetier, now it is doing ok. No pauses, and no skipping.

I thing you should mention Repetier Settings (above written) in your Readme.md .

My issues are over.

luc-github, 2019-08-19:
these are repetier settings and not related to temperatures 
also the pause looks like : https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/38
which is not related to ESP WIFI as they happen also by USB

ajaybnl, 2019-08-19:
but that was a serial problem. consider writing repetier settings bro.

closed 2019-08-19 by ajaybnl

luc-github, 2019-08-19:
People do not read/fill issue template when opening issue - why do you think they will read the docs ? ^_^