luc-github/ESP3D #361: Error EEprom

Wuckel01, 2019-08-18:
Hello I have a problem with the access to the flash memory is not the web interface is but when it comes to save nothing.

I am using a NODEMCU V3 4MB
on the home page, I can not upload anything and he does not show a memory if I but with telnet memory query this is there.

I have already used several modules ESP8266-01 1MB and also NODEMCU 0.9 everywhere the same problem no memory.
IN IDE I have adjusted the SPiFFS jedenlam suitable.

Maybe you have an idea.

luc-github, 2019-08-18:
I have no idea with current information, please fill the issue template

Wuckel01, 2019-08-18:
Sorry was not quite finished yet.

luc-github, 2019-08-18:
what esp core version ?

Wuckel01, 2019-08-18:
esp verson 2.5.2      IDE 1.8.5

luc-github, 2019-08-18:
looks like your SPIFFS is not writable or corrupted
try `[ESP710]FORMAT pwd=admin` as it looks like you enabled authentication
then  upload index.html.gz with webpage

Wuckel01, 2019-08-18:
[ESP710]FORMAT
Formating...Done
I had already done the formatting

echo:Unknown command: "[ESP710]FORMAT"
                                      ok
                                        echo:Unknown command: "Incorrect Cmd"
                                                                             ok
when uploading the index.html.gz file man can wait forever because nothing happens.

luc-github, 2019-08-18:
your esp is connected to printer ?
please debug it disconnected - your commands seems going to printer not ESP 
Do formating when serial is connected to ESP, use arduino monitor to send commands

Wuckel01, 2019-08-18:
I made the password function but. I am always checked out for password.
witty ip, he shows in the LCD but esp sends nothing but an error.
G28
Error 401 :Authentication failed!

luc-github, 2019-08-18:
can you try with authentication disabled ?

Wuckel01, 2019-08-18:
//AUTHENTICATION_FEATURE: protect pages by login password
//#define AUTHENTICATION_FEATURE

I did it anyway he asks

luc-github, 2019-08-18:
It won't if disabled -  close arduino editor and open again to be sure you start on clean base

luc-github, 2019-08-18:
then refresh webUI

Wuckel01, 2019-08-18:
I have made the same thing he fargt after password and I can not choose printer type and synonymous nix save settings. I have another module taken the same problem igentwas that makes me nit in the flash can write

luc-github, 2019-08-18:
you enabled authentication what others changes did you do in FW ?

Wuckel01, 2019-08-18:
hi I read in a forum post that in the config.h
DEFAULT_STA_SSID was the problem I have set it so maybe that's the error

//position in EEPROM
//AP mode = 1; Station client mode = 2
#define EP_WIFI_MODE			2    //1 byte = flag

luc-github, 2019-08-18:
what forum post ?

do not change anything but within this section : https://github.com/luc-github/ESP3D/blob/2.1/esp3d/config.h#L40-L93

others part do not touch

Wuckel01, 2019-08-18:
I wanted to have wlan sta mode because there were only those password queries via wlan, that's why I set the values

post
https://github.com/luc-github/ESP3D/issues/218

luc-github, 2019-08-18:
I this post the guy mess up the values, do not change them, change only the ones I gave before

Wuckel01, 2019-08-18:
thanks jop that was it I've done it all goes now oh man

luc-github, 2019-08-19:
Ok great - I close issue then

closed 2019-08-19 by luc-github