luc-github/ESP3D #363: [FEATURE REQUEST] Support for Creality 1.1.4 board

baxinabox, 2019-08-25:
I'm looking to add ESP3d support to my Ender 3 Pro 
I have a ESP8266 D1 mini board flashed and working and am at the step where I will interface it with my printer. I have a newer Ender 3 Pro with a Version 1.1.4 board

Can I ask for some suggestions on where to hack onto the board.
I have with the help of the wiki here Identified the TXRX pins on the main chip, the 1.1.4 uses a different usb chip, (CH340G) and from it's datasheet I've ID'd the TXRX pins.
I have also seen reference to hacking into the LCD cable.

Can anyone speak to the best place to attach the D1 mini to this PCB ?
Is hacking into the RXTX at the main chip more sensitive than going onto the USB chip or the LCD cable. 

Has anyone seen info on this LCD hack? I can't seem to find any info on it or a link to the pinouts for the Ender 3 LCD cable. 

Thanks any help is appreciated.
[image]

luc-github, 2019-08-25:
I doubt there is a (RX/TX) serial on such lcd cable
LCD cable is 10 wires from what I can see on google
1 for buzzer
1 for GND
1 for VCC
3 for Rotary encoder (A, B, C)
3 at least for lcd (RS/EN/Data)
so only one pin left in best case

nionio6915, 2019-08-27:
is the LCD a conventional graphical one or is the TFT touchscreen kind?

luc-github, 2019-08-27:
From what I see on google Ender-3 Pro screen has a standard screen with rotary encoder
[image]

The Touch upgrade come after a board upgrade if I read well the forums

baxinabox, 2019-08-27:
I have an ender 3 pro with a regular lcd screen. It has the three “Exp” ports on the back. Of which I am using #3 to comm with the main board.
I intend on accessing the tx/rx on the main board but wasn’t sure if there was a better reason to use the lcd cable for this mod.
I can’t find any info on modding the lcd for rx/tx, just break out boards for attaching bed level sensors.

[image]<https://user-images.githubusercontent.com/8822552/63779535-fe89fc80-c8e6-11e9-8252-fb4161fd9cdf.png>

baxinabox, 2019-08-27:
> I'm pretty sure there's no pins free on the LCD cable. The various leveller additions use a breakout board that reuse the buzzer pin; I'm not sure how many pins the BLTouch uses, but there sure can't be much spare. I don't know if you could piggyback onto the LCD communications?

Ok thanks, it was suggested by a fellow maker in YouTube but he didn’t provide proof of concept, just said it was possible a few times. But I couldn’t find anything about it online.
I’ll just go directly into the main board Tx

wilbur4321, 2019-09-04:
This might help?  https://reprap.org/wiki/Melzi#Melzi_with_Bluetooth

luc-github, 2019-09-05:
this creality board seems same as others creality available in wiki, so wiki should cover it
@baxinabox do you confirm ?

baxinabox, 2019-09-05:
I am not rushing into this mod right away because I need my printer to be ready to print over the next couple weeks and don’t want to risk a mistake and damage my
Mainboard
I will look at the costs of buying a wifi ready main board upgrade vs getting a backup stock mainboard before I start to solder directly to my MB.

I’ll update this when I move ahead

this creality board seems same as others creality available in wiki, so wiki should cover it
@baxinabox<https://github.com/baxinabox> do you confirm ?

closed 2019-10-15 by stale[bot]

Lisheng2016, 2020-01-02:
I've soldered Rx to onboard-Tx0 / Rx to onboard-Rx from esp01s to ender3pro motherboard.And it seems to giving commands but machine is not responding to the commands, didn't see hack message on startup as well..

luc-github, 2020-01-02:
@lisheng2016 this issue is closed. please open new one
also as it is esp01 provide all wiring you used for it