luc-github/ESP3D #364: [FEATURE REQUEST] Reset a pin's state after using [ESP201]

josedpedroso, 2019-08-26:
I connected GPIO 2 on my ESP01 to my printer's reset pin. To reset the printer, I use a macro in spiffs that contains:

[ESP201]P2 V1
[ESP201]P2

This is too fast to actually reset the printer, but I have no way to delay the second command that resets the pin. Issuing these commands manually from the dashboard works fine. Using just the first line in the macro also works, ie, the printer reboots, but I'm not comfortable leaving the printer's reset pin high all the time.

I see 2 possibilities here: I either need a command to delay the macro's execution or the [ESP201] command should have an option to reset a pin to its original state after some time.

As a workaround, I tried adding an [ESP410] command between the 2 [ESP201] lines, but that did nothing at all.

UI: 2.0c9 / FW: 2.0.0.c25

Thanks.

josedpedroso, 2019-08-26:
My printer doesn't like it when I set the reset pin low (0). Would it be possible to use R parameter with no value to set the pin's mode to input?

josedpedroso, 2019-08-26:
In input mode, an arduino's pin is in a high impedance state, ie, floating from my printer's point of view. Reference: https://www.arduino.cc/en/Tutorial/DigitalPins

> equivalent to a series resistor of 100 megohm in front of the pin

josedpedroso, 2019-08-26:
I tried that, it locks the printer up until I send [ESP201]P2.

josedpedroso, 2019-08-26:
In any case, just having the delay in the first [ESP201] command would solve my issue.

josedpedroso, 2019-08-26:
That'd work for me. And I bet it could be useful in other situations. Thanks.

josedpedroso, 2019-08-26:
Just by curiosity, do you know why [ESP410] didn't work to delay the macro? Is that command only available interactively?

luc-github, 2019-08-26:
we could add a 2 new parameters 
[ESP201]P2 V1 D2000 R0
D for delay in ms
R for revert to the value after delay

luc-github, 2019-08-26:
I am not sure of pin state when in input mode, should be low

luc-github, 2019-08-26:
but should be able to see 
doing manually the test 
send 
[ESP201]P2 V1
then wait few sec, then send
[ESP201]P2

if acting sam as 
[ESP201]P2 V1
wait few sec
send [ESP201]P2 V0
then send [ESP201]P2

then using [ESP201]P2 V1 D2000 R0 then [ESP201]P2 should be ok

luc-github, 2019-08-26:
the delay without changing state is confusing IMHO
so better to create a delay command in that case
e.g. [ESP250]1000  for delay 1000ms

luc-github, 2019-08-26:
Too fast I guess

luc-github, 2019-08-27:
I have implemented the command in 3.0, but there is not webUI for it yet
* Delay command
[ESP290]<delay in ms>[pwd=<user password>]

closed 2019-08-27 by luc-github

luc-github, 2019-08-27:
Command also added in 2.1

commit d39ea079 (2019-08-27): Add [ESP290] command per @josedpedroso

fix #364