luc-github/ESP3D #365: [BUG]SD Upload

luc-github, 2019-08-28:
@Roman-RV78  please use this issue 
Hello. I ran into a similar problem. I can see the files on the SD card, I can delete them, but copying the files to the SD card fails, error. Arduino DUE Marlin 2.0, NODE MCU

M115
FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:CHAMBER_TEMPERATURE:0
[ESP800]
FW version:2.1.0.b35 # FW target:marlin # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81 # hostname:MYESP
M20
Begin file list
CUBE.G 0
End file list

luc-github, 2019-08-28:
what is the error ? when happen in upload ?

luc-github, 2019-08-28:
what error appear?

luc-github, 2019-08-28:
what is the size of upload file ?

Roman-RV78, 2019-08-28:
The process of downloading files to the SD card has been in progress for a very long time, and when it reaches 100%, there is an error loading the file, and a file of zero size appears on the SD card.
Translated using Google

Roman-RV78, 2019-08-28:
[image]
[image]

Roman-RV78, 2019-08-28:
I sent a file of 1.53 megabytes in size.
  File from Home.zip archive loaded successfully

luc-github, 2019-08-28:
If upload of home.g is working then upload is working in normal way
can you share the gcode file you use ?

Roman-RV78, 2019-08-28:
[cube.zip](https://github.com/luc-github/ESP3D/files/3551562/cube.zip)

luc-github, 2019-08-28:
ok nothing pop up my eyes, I will add more info when issue raise in websocket message to get some clue

luc-github, 2019-09-02:
I tried with your configuration with baud rate 250000 and cannot duplicate
the upload happen with long time but at the end the file is there

luc-github, 2019-09-04:
@Roman-RV78 what is your baud rate ?

Roman-RV78, 2019-09-04:
250000

luc-github, 2019-09-05:
I cannot reproduce on my side - and you said it is working with test file - did you tried another file ?
also how is your wiring ? you may have noise on it

DiegoGiovany, 2019-09-05:
I Cant upload too, but I think its caused by marlin ADVANCED_OK feature...

luc-github, 2019-09-08:
@DiegoGiovany why do you think isue could come ADVANCED_OK  ?

DiegoGiovany, 2019-09-08:
Cause when ADVANCED_OK is disabled, upload works....

luc-github, 2019-09-08:
@DiegoGiovany thank you for this feedback - I will enable ADVANCE_OK to see what is happening then

luc-github, 2019-09-17:
@Roman-RV78 any update ? do you have ADVANCED_OK  enabled ?

Roman-RV78, 2019-09-17:
How to check if ADVANCED_OK is enabled or not

AnHardt, 2019-09-17:
Two ways.
a) Look it up in Configuration_adv.h
b) Look it up in the log. If 'ok' is not followed by Nlinenumber an buffer states `ADVANCED_OK` is not enabled.

Roman-RV78, 2019-09-18:
In Configuration_adv.h, the line ADVANCED_OK is always commented out by default, I didn’t even touch it

closed 2019-10-12 by stale[bot]