luc-github/ESP3D #366: Arduino Uno - No voltage regulator, possible? MKS Wifi Module

ETE-Design, 2019-09-01:
Is it possible to connect the ESP8266 to an Arduino Uno, without any voltage regulator? Or can I just supply the 3.3v from an other device? And is there an diagram to connect the MKS Wifi Module to the Arduino?

luc-github, 2019-09-01:
according https://www.farnell.com/datasheets/1682209.pdf uno board is 5V - there are many tutorial on how to connect esp to uno board - I shared the pinout here : https://github.com/luc-github/ESP3D/issues/249#issuecomment-475898023

ETE-Design, 2019-09-01:
Do you know if I can use an Raspberry Pi then? Cause it have 3.3v on the gpio... And how to set it up in Arduino then?

luc-github, 2019-09-01:
No sorry I only used mks wifi module with serial 2 usb  device

luc-github, 2019-09-01:
what is not working is not in FW - is in web UI - you can do a customized webUI with features that MKS wifi module do not allows, I did not tested all actually

ETE-Design, 2019-09-01:
Like this?
https://www.ebay.com/itm/USB-RS232-TTL-UART-PL2303HX-Converter-Serial-Cable-Adapter-Arduino-/253844303313?ef_id=EAIaIQobChMI1rz_qriw5AIVWOJ3Ch2J5AJfEAQYBiABEgKv1_D_BwE:G:s

ETE-Design, 2019-09-01:
I have a cable like this and when I open it up there is a pin for 3.3v :-)

luc-github, 2019-09-01:
I use this one https://www.aliexpress.com/item/1897044115.html it has 5v and 3.3v jumper

ETE-Design, 2019-09-01:
Thanks :-) Is there a "How To" to customize the Web UI?

luc-github, 2019-09-01:
Most ui is already customizable using preferences, for more specific changes you need to modify html.

ETE-Design, 2019-09-01:
Nice... Will look at it :-) Last questions:
1. Have you tryed the ESP3D on an MKS Robin Lite 1.1? Have one here, and have just ordered the module for it from Makerbase...
2. Do you know if there is other WiFi modules made that will fit in the socket? Makerbase's price seems a bit high...

luc-github, 2019-09-01:
I do not  know this board, what wifi module fit to it seems not mks wifi 1.0

ETE-Design, 2019-09-01:
This one:
https://www.aliexpress.com/item/32816882694.html?spm=a2g0s.9042311.0.0.51d14c4dGWIv3j

luc-github, 2019-09-01:
Interresting now they use 4MB flash

ETE-Design, 2019-09-01:
MKS Robin Lite is not that new, but yes 4MB Flash, and will try to make it work direct with Marlin, so not closed source like MKS Wifi on the MKS TFT...
Not that we are talking what kind of Camera will work best with this module???

luc-github, 2019-09-02:
any IP cam that allow to be visualised in browser will fit

luc-github, 2019-09-02:
May be FW is not same as MKS WIFI 1.0 as it is different board and use more pins to connect

ETE-Design, 2019-09-02:
But it use Marlin, and I have Full control over the board... And will flash your ESP3D to the ESP8266 from MKS so Hope to open it up for full support...

ETE-Design, 2019-09-02:
Will it be possible to fit a camera direct to the ESP8266? Like Arducam?

ETE-Design, 2019-09-02:
Schematic for pins used on the module:
https://github.com/makerbase-mks/MKS-Robin/blob/master/MKS%20Robin%20Lite/hardware/MKS%20Robin%20Lite%20V1.1_002/MKS%20Robin%20Lite%20V1.1_002%20SCH.pdf

luc-github, 2019-09-02:
Performance will be very poor. If use esp3d + cam on esp8266. There is a plan for special UI with esp32 as even esp32 has hardware support for cam , better to limit feature when running

ETE-Design, 2019-09-02:
Okey, then I will start without cam, and later make the ESP32 Work in IT when the UI is ready... Can also help testing if you need :-) But for now I Will Focus on the ESP8266 to get something running... I'm closing this cause need to wait till I get my ESP8266 from Ali...

luc-github, 2019-09-02:
Well if you use Marlin you actually do not need to customize webUI then all should be Ok

ETE-Design, 2019-09-02:
I know, for the MKS Robin Lite, All should be fine og I can make it talk with the ESP8266... But with MKS WiFi on MKS TFT is another story...

luc-github, 2019-09-02:
if you like challenge there is an opensource FW for MKS TFT but I never tested

ETE-Design, 2019-09-02:
Seems a bit old, so there will be a lot to do I think... So for now I stay with the MKS TFT FW, and your ESP3D part ;-)

ETE-Design, 2019-09-03:
BYW, is it possible to add a Logo in the web UI?

luc-github, 2019-09-03:
What logo ?
It is html so it is possible

ETE-Design, 2019-09-03:
Would like to add the Logo of the machine I use on the page...

ETE-Design, 2019-09-05:
So I got my 5v to 3.3... Using an Arduino Uno, but seems like it won't connect when I try to flash :-( Can you help? My pins goes like this:
Esp8266 -> Arduino
IOO ->3.3V
GND -> GND
VCC -> GND
RXD -> TX
TXD -> RX
NC -> 3.3v

HOw to connect it right?

luc-github, 2019-09-05:
flash mode GPIO0 is to GND
normal mode GPIO0 is HIGH

ETE-Design, 2019-09-05:
Think I got it...
It said:
Leaving...
Hard resetting via RTS pin...

How do I add it to my network then? Just flash the MKS_Config.txt again? It dosen't seem to make an AP either :-( Do I have to erase it first, if then, how to?

ETE-Design, 2019-09-05:
And I have enabled the MKS Wifi in the Config.h

luc-github, 2019-09-06:
Test it first when connected with your UNO the AP mode should start automatically
you can check also if it start by sending `[ESP111]` to get IP

`leaving` is not in my FW

closed 2019-09-06 by ETE-Design