luc-github/ESP3D #368: ESP32 uploads slow fixes?

dboon1978, 2019-09-11:
Uploads are extremely slow, it would make more sense to use an external SD card for uploads instead of going through AUX or TFT RX/TX.  Or add support for FTP or just a URL to the gcode (dropbox, google drive etc)

A flag in the config to make uploads default to an external SD card?
A "print from url" text box

luc-github, 2019-09-11:
how printer access to external SD card ?
how printer access to ftp / url ?
printing from url : how do you handle communication errors/retry  ?

dboon1978, 2019-09-11:
1) that was actually my question.  I had an external sd card connected to my ESP32 and my own code was accessing the SD card fine, but I couldn't figure how to tell the ESP3d code to look at the SD instead of the one connected via the printer.
2) I was looking at something like this https://github.com/lucasromeiro/DropboxManager
3) That's a good question, I am not sure how i would handle that.
4) I know, I'm trying to find a solution so I can use one of my ESP32s instead of buying a raspi

luc-github, 2019-09-11:
1 - Upload to any SD is fast - once your gcode is to the external SD - how printer access to this file ?

the only solutions I found is the ones I suggest in the pinned issue : https://github.com/luc-github/ESP3D/issues/292

luc-github, 2019-09-11:
I think you have the wrong idea of the process -  you think ESP3D is the blocking point, the problem is actually where or how to copy the gcode file so the printer has access to it, ESP3D can copy any place you want if direct access to them

dboon1978, 2019-09-11:
1 - So if I have an external SD with gcode on it, it's impossible for the ESP3d to print directly to the printer?  The printer needs to see the SD card's data or it won't print the gcode?
2 - i see, so downloading can't go straight to print, I was hoping it would work like the wifi-to-serial protocol

Ok so if I want to have fast uploads the only way is to connect my ESP32 directly to the printer board's SD card pins?  Have you seen anyone do this on an SKR1.3?

Thanks so much for your quick replies!  I really love this project, I just want it to work better for me.

luc-github, 2019-09-11:
1 - No ESP3D do not 'print' anything - the printer need to have access to GCODE in some way, should be on printer SD Card or get each command one by one sent by a host 
2 - you can but without error management it is risky, sending file without any control of the flow  - I won't do it 

yes but it is not possible `just like this` 
1 - you need to have special spi driver on printer FW allowing ESP3D to access the shared SD - this is done on Reprap firmware but other FW do not have such driver
2 - ESP3D has first access to SD and share its SD with printer when he has finished upload
this is how the panucat wifi back pack works - I guess should work with SKR1.3 whic has an external display
3 - use wifi SD card - the wifi SD card allows wifi and printer to access board in same time

4 -https://github.com/ardyesp/ESPWebDAV
 but I won't support it as not easy to be reliable IMHO and depend on hardware that need to be done

luc-github, 2019-09-11:
Panucatt addon boards
https://www.youtube.com/watch?v=br7BryjWPJU&feature=youtu.be

Wifi SD card
https://www.youtube.com/watch?v=enYwh7UY-Oo I did a file manager for it : https://github.com/luc-github/Flashair-FM

luc-github, 2019-09-11:
Panucatt addon boards
https://www.youtube.com/watch?v=br7BryjWPJU&feature=youtu.be

Wifi SD card
https://www.youtube.com/watch?v=enYwh7UY-Oo I did a file manager

dboon1978, 2019-09-11:
Okay so
1) reprap doesn't seem to support skr1.3 so that won't work
2) I'm not sure how this solution works, can you expand on that one?
3) that looks pricey.. $30 + sd->microsd adapter = same price as octoprint solution
4) using something you won't support seems foolish

Panucatt board seems reasonable..  $20 vs $45 for a raspi

Thanks!  You've given me a lot to think about.

luc-github, 2019-09-11:
1 - yes only for duet wifi
2 - watch the first video - the board allows to share the SD from display but only one device can access at once

3 - well in past I only have this one ^_^ - octoprint is nice but printing from SD is safer for me
4 - I give options ^_^ , there are a lot of more that I did not listed  ;)

Have a good brainstroming then ^_^
if `feature doesn't require too much work` someone already did it usually  ;)

dboon1978, 2019-09-11:
So with the SKR1.3 it has two EXP ports, one of mine is connected to the LCD screen.  If I can figure out the pins for the EXP port is it plausible this will connect directly to the SD card?

Maybe this is a better question for an SKR group.   But let me know if I'm even slightly close.

luc-github, 2019-09-11:
https://youtu.be/9yGLbuITNBg
Acording this glcd can be connected so yes panucat board should allow to share glcd  sdcard i

dboon1978, 2019-09-11:
panucat board can share glcd sdcard, but not an esp32 directly?

dboon1978, 2019-09-11:
k, thanks!

luc-github, 2019-09-11:
No, the sharing is done by hardware , esp3d just send command to select which hardware( esp or printer) has access to SD
It is not that easy ^_^

dboon1978, 2019-09-11:
gotcha, so for now I might as well just upload via PC then print from ESP3D.  Thanks!

luc-github, 2019-09-11:
Yes this is the simplest solution

dboon1978, 2019-09-13:
Question, if there was a way to connect the micro usb on a nodemcu dev board to the usb on the main board, would it be possible to get a direct connection to the usb?  My question is cause I can upload to usb from my PC extremely fast, so it must have some sort of faster connection to the sd card right?

dboon1978, 2019-09-13:
Like if I spliced a USB type b (the type my skr1.3 uses) and a micro USB (the type used for ESP32/nodemcu serial usb connection) and hooked it up, would that work as a serial connection as well as being able to transfer data faster?

luc-github, 2019-09-13:
No, you need one of them be an usb host , and none of them has this feature and again the issue is the  communication protocol

luc-github, 2019-09-17:
I guess question is answered - I close issue

closed 2019-09-17 by luc-github